chan-sccp/chan-sccp #252: Transfer on external originated call / Codec Mismatch -> Segfault

dkgroot, 2015-02-04:
Reported By: Dave Boyd
Date: 2015-02-03
Subject:  When using the transfer soft key asterisk core dumps with Chan_SCCP-4.2.0_RC2_r5893

Steps:
- Generate an outbound external call
- answer call
- transfer call to ext
- answer call
- Core

Prevented by adding:
Disallow =all
Allow= ulaw
Allow= alaw
Allow=gsm

Analysis: Codec mismatch during transfer causes segfault.
Required: Codec Negotaition during transfer + Potential Transcodind when necessary

pnbronaldng, 2017-02-27:
HI dkgroot,

May I know how was the feeling about chan-sccp you are using? 

I am helping my client to move from Cisco Call Manager to Asterisk and they have around 1,000 Cisco IP phone with SCCP, 7925 and 7975. All using SCCP protocol.  I need to investigate the stability about chan-sccp driver.  

Let's share your feeling and experience.

Thanks,

Ronald Ng

dkgroot, 2017-02-27:
Hi Ronald,

This question would have been better suited for the mailinglist (See: https://sourceforge.net/p/chan-sccp-b/mailman/), instead of an open issue report. But ok :-) Would be better to ask a user of the software, and not one of the maintainers :-)

The 'develop' branch has proven to be more stable, then anything before it. All current issues in the issues list are of a somewhat minor priority (not directly stability related). Some of our bigger users are reporting multiple months of uptime, with the current version. For example @voipAsteriskGuy: he has 900 phones connected to a single server, (uptime.4 weeks (after manual update/reload) with 224590 calls processed reported on the 15th of feb 2017).

My advise:
- Create a HA cluster of 3 back-end nodes (330 phones per instance). Group departments together.
- Have a SIP-trunk head node that does all call routing to the back-end cluster members
- Cluster Nodes can be running on VM(KVM, vmware etc) or Container(LXC, Docker etc), making it easy to update/upgrade/clone etc.
- Store your node configuration in a central repository (git, db, san, drbd), makeing it easier to keep them all the same.
- use DunDI or ENUM to route calls from the frontend to the backend. This will allow the frontend to find out dynamically on what node a phone is currently connected.
- Use directRTP between head / backend, this will allow phone calls to be completed even if a backend server falls over. Use sccp quick failover functionality in case of emergency, allowing a phone to switch servers in seconds.
- Create a clone to use for testing/staging changes (OS/asterisk/chan-sccp) upgrades. Maybe setup a little lab with a couple of phone, so you can do tests, before putting stuff in production.
- If possible use vanilla asterisk and your own dialplan (I personally am not a big fan of FreePBX etc).
- Make your dialplan as generic as possible, so it can be shared by all nodes. This makes them interchangeble.
- Setup and Test Qos on your network switches/routers
- In regard to the 7925, make sure you wifi coverage is spotless.

This might look like a very long list / overkill, but with a little planning, it should not be that hard to implement and test. It will make any future changes/updates much easier and your client will be happier, even in the even in the even that something does go wrong.

voipAsteriskGuy, 2017-02-27:
hi @pnbronaldng  I am using develop version commit 5ad78e4, with 950 sccp phones some phones with shared lines, also enabled functionality of group pickup, call transfer, conference, busy lamp field, speed dials and video.

This deploy has been in production running for 6 weeks 12 hours without issues and also processed 311361 calls with a peak on some days of 70 active calls simultaneity 

[image]

[image]

[image]

pnbronaldng, 2017-03-02:
Dear @voipAsteriskGuy<mailto:notifications@github.com>,

Thanks for your reply. Really treasure about it.  May I know any IVR you have implemented such as password authentication, multi-language pack that you have applied to Cisco SCCP phone?  Did Cisco SCCP has such auto random reboot, handset hang up event?
And also, can you share the configuration and architectural logical diagram to us?

If I would like to invite you to join the proof of concept test for Cisco SCCP and Asterisk configuration, are you available?

Please advise.

——— / Regards / Ronald Ng / 96000297 / ronald.ng@pnbhk.com

[mage]<https://cloud.githubusercontent.com/assets/22506296/23368245/d435dd06-fcd2-11e6-85ca-f561fe746fa3.png>

[mage]<https://cloud.githubusercontent.com/assets/22506296/23368267/e1be7d8e-fcd2-11e6-9d82-5e48f4f36bac.png>

[mage]<https://cloud.githubusercontent.com/assets/22506296/23368281/f0cf51ea-fcd2-11e6-8787-0802416740f3.png>

dkgroot, 2017-03-02:
@pnbronaldng : Please sign up to our mailinglist and restart/continue the discussion there. This is really not supposed to be in done in an open issue :-) Thanks

dkgroot, 2017-03-02:
@pnbronaldng :
Please sign up to our [mailinglist](https://sourceforge.net/p/chan-sccp-b/mailman/) and restart/continue the discussion there. Also there is a [chan-sccp chat](https://gitter.im/chan-sccp/chan-sccp?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge) that might be better suited as well. You will have a change to relay you questions to multiple people , who might be able to help.

I don't want to be annoying about this, but these types of discussion are really not supposed to be in done in an open issue :-) Normally only developers are checking (and informed about changes to) the issue database on a regular basis.

closed 2020-03-01 by dkgroot