chan-sccp/chan-sccp #258: note: expected 'const char *' but argument is of type 'struct ast_cli_entry *'

marcelloceschia, 2015-04-11:
```
pbx_impl/ast/ast113.h:134:5: note: expected 'const char *' but argument is of type 'struct ast_cli_entry *'
 int pbx_manager_register(const char *action, int authority, int (*func) (struct mansession * s, const struct message * m), const char *synopsis, const char *description);
     ^

```

dkgroot, 2015-04-11:
I cannot find this one. Was this a scan-build. Running clang with -Werror, does not produce this on my system.

marcelloceschia, 2015-04-11:
Ich habe nur ein svn update gemacht. building with gcc

marcelloceschia, 2015-04-11:
```
In file included from pbx_impl/ast/ast.h:106:0,
                 from chan_sccp.h:264,
                 from common.h:51,
                 from sccp_cli.c:50:
pbx_impl/ast/ast113.h:134:5: note: expected 'const char *' but argument is of type 'struct ast_cli_entry *'
 int pbx_manager_register(const char *action, int authority, int (*func) (struct mansession * s, const struct message * m), const char *synopsis, const char *description);
```

dkgroot, 2015-04-11:
Ok switching compilers ;-) Komisch

Diederik

dkgroot, 2015-04-11:
./configure  --enable-devdoc --enable-conference --enable-advanced-functions --enable-experimental-mode --with-asterisk=/usr/local/asterisk-13-branch CC=gcc
              CC : gcc
             CPP : gcc -E
         HOST_CC : gcc

make
....
make[2]: Entering directory `/data/development/sccp/sources/chan-sccp-b_release/src'
  CC       chan_sccp_la-sccp_enum.lo
  CC       chan_sccp_la-sccp_channel.lo
  CC       chan_sccp_la-sccp_device.lo
  CC       chan_sccp_la-sccp_indicate.lo
  CC       chan_sccp_la-sccp_pbx.lo
  CC       chan_sccp_la-pbx_impl.lo
  CC       chan_sccp_la-sccp_softkeys.lo
  CC       chan_sccp_la-sccp_featureButton.lo
  CC       chan_sccp_la-sccp_line.lo
  CC       chan_sccp_la-sccp_utils.lo
  CC       chan_sccp_la-sccp_actions.lo
  CC       chan_sccp_la-sccp_cli.lo
  CC       chan_sccp_la-sccp_features.lo
  CC       chan_sccp_la-sccp_config.lo
  CC       chan_sccp_la-sccp_hint.lo
  CC       chan_sccp_la-sccp_refcount.lo
  CC       chan_sccp_la-sccp_event.lo
  CC       chan_sccp_la-sccp_adv_features.lo
  CC       chan_sccp_la-sccp_management.lo
  CC       chan_sccp_la-sccp_mwi.lo
  CC       chan_sccp_la-sccp_conference.lo
  CC       chan_sccp_la-sccp_rtp.lo
  CC       chan_sccp_la-sccp_appfunctions.lo
  CC       chan_sccp_la-sccp_protocol.lo
  CC       chan_sccp_la-sccp_socket.lo
  CC       chan_sccp_la-sccp_threadpool.lo
  CC       chan_sccp_la-sccp_devstate.lo
  CC       chan_sccp_la-ast.lo
  CC       chan_sccp_la-ast113.lo
  CC       chan_sccp_la-chan_sccp.lo
  CCLD     chan_sccp.la
make[2]: Leaving directory `/data/development/sccp/sources/chan-sccp-b_release/src'
make[1]: Leaving directory `/data/development/sccp/sources/chan-sccp-b_release/src'
Making all in doc
...

Also lauft bei mir glatt. Brauchst du vielleicht ein make clean :-)

Allerletzte asterisk-13 und asterisk-13-branch gecheckt.

Diederik

commit e91bb5d1 (2015-04-11): Fix: cli / ifdef issue (fixes #258)

git-svn-id: https://svn.code.sf.net/p/chan-sccp-b/code/trunk@6016 43c1dc8c-776e-4df2-8ef1-6e829299be21
closed 2015-04-11 by dkgroot