chan-sccp/chan-sccp #262: sccp R6139 invalid memory reference in log

marcelloceschia, 2015-07-08:
Date: 7.7.2015
```
[Jul  6 14:12:06] VERBOSE[25559] sccp_hint.c:     -- SCCP: (sccp_hint_notifyPBX) Notify asterisk to set state to sccp channelstate 'ONHOOK' (1) on line 'SCCP/103'
[Jul  6 14:12:06] VERBOSE[25559] sccp_hint.c:     -- SCCP: (sccp_hint_notifyPBX) => asterisk: 'Device is in use' (2) => 'Device is not in use' (1) on line SCCP/103
[Jul  6 14:12:06] VERBOSE[25559] sccp_hint.c:     -- SCCP: (sccp_hint_notifyPBX) Notified asterisk to set state to sccp channelstate 'ONHOOK' (1) => asterisk: 'Device is not in use' (1) on channel SCCP/103
[Jul  6 14:12:06] VERBOSE[25533] sccp_hint.c:   == 103: (sccp_hint_devstate_cb) Got new hint event SCCP/103, state: 0 (Idle), cidname: , cidnum:
[Jul  6 14:12:06] VERBOSE[25533] sccp_hint.c:     -- 103: (sccp_hint_notifySubscribers) notify 5 subscriber(s) of SCCP/103's state ONHOOK
[Jul  6 14:12:06] NOTICE[25592][C-0000001e] sccp_actions.c: SEP00055E7CCC14: Device sent a Keypress, but there is no (active) channel! Exiting
[Jul  6 14:12:06] VERBOSE[25592][C-0000001e] sccp_refcount.c: SCCP (sccp_refcount.c:0532 (sccp_refcount_autorelease)) ALARM !! trying to release a UNREF ((nil)) with invalid memory reference! this should never happen !
[Jul  6 14:12:06] ERROR[25592][C-0000001e] sccp_refcount.c: SCCP: (release) Refcount Object 0xb7260f57 could not be found (Major Logic Error). Please report to developers
```

dkgroot, 2015-07-08:
No exactly sure why this unref is causing issues. I do have a potential patch, which creates a new tmp variable to walk the list, instead of reusing the one on which the reference is returned. Will send it via mail.

dkgroot, 2015-09-03:
DLList Find function is using a local tmp variable and traverses the list using ref/unref

closed 2015-09-03 by dkgroot