chan-sccp/chan-sccp #279: Invalid SCCP message! : ID :143: readByte() underrun

marcelloceschia, 2015-10-30:
```
1219: WRN 16:58:13.973153 JVM: Startup Module Loader|cip.sccp.co:? - Invalid SCCP message! : ID :143: readByte() underrun, length = 0 - close connection and alarm in future
1220: ERR 16:58:13.979592 JVM: 16:58:13|cip.sccp.bd: ID :143: readByte() underrun, length = 0
	at cip.sccp.ay.readByte()B(Unknown Source)
	at cip.sccp.ay.a(I)Ljava/lang/String;(Unknown Source)
	at cip.sccp.ay.a()Ljava/lang/String;(Unknown Source)
	at cip.sccp.ez.b(Lcip/sccp/ay;)V(Unknown Source)
	at cip.sccp.bv.a(Lcip/sccp/ay;)V(Unknown Source)
	at cip.sccp.ay.a()Lcip/sccp/bv;(Unknown Source)
	at cip.sccp.co.e()V(Unknown Source)
	at cip.sys.l.run()V(Unknown Source)
	at java.lang.Thread.startup(Z)V(Unknown Source)
```

dkgroot, 2016-02-20:
Can you check if this still happens with current git/master, after the rework in sccp_session.c ?

dkgroot, 2016-11-19:
Not reported anymore. Closing

closed 2016-11-19 by dkgroot