chan-sccp/chan-sccp #280: Invalid SCCP message! : ID :9a

marcelloceschia, 2015-10-30:
1417: CRT 18:12:36.537219 JVM: Startup Module Loader|cip.sccp.ah@131ba1:createMessage - No entry in hashtable for ID=0x9a
1418: WRN 18:12:36.538816 JVM: Startup Module Loader|cip.sccp.co:? - Invalid SCCP message! : ID :9a: MessageFactory.createMessage failed, length = 0 - close connection and alarm in future
1419: ERR 18:12:36.543793 JVM: 18:12:36|cip.sccp.bd: ID :9a: MessageFactory.createMessage failed, length = 0

dkgroot, 2015-10-31:
Do you know which two messages these are  (DynamicNotify and ClearDisplay ?)
ProtocolVersion ?

marcelloceschia, 2015-10-31:
0x9a

dkgroot, 2015-10-31:
Taking that as cleardisplay. This device does not seem to know this message, so that's why asked for the protocol version.  Maybe we should have used ClearNotifyMessage on this type of device instead.

marcelloceschia, 2015-10-31:
ist ein 7975 mit aktuellster firmware
daher aktuelle protokoll version

dkgroot, 2015-10-31:
Bin gespannt was du dort finden wirst.

dkgroot, 2016-02-01:
@marcelloceschia : Gab es ibz message "9a" hier noch neuigkeiten. Mussen wir da noch was machen / fixen ?

marcelloceschia, 2016-02-01:
da müssen wir einfach die phone logs noch mal prüfen

dkgroot, 2016-02-01:
Ok. War mir nicht ganz deutlich wie/wo wir da geblieben waren :-)

dkgroot, 2016-03-23:
Dropped usage of this message for all phones, so far no reports about missing functionality. Issue Closed

closed 2016-03-23 by dkgroot