chan-sccp/chan-sccp #283: Bad PBX_VARLIB on FreePBX distro

lynt-smitka, 2015-11-19:
Hi,

I noticed PBX_VARLIB points to non-existent directory on the FreePBX distro after ./configure

    Operating System : Linux
              Kernel : 2.6.32-431.el6.x86_64
        Architecture : x86_64
    Asterisk Version : 11300 / 113
      Asterisk Repos : TRUNK
          Build user : root
             Version : 4.3.0_TRUNK_rc44f500
                  CC : gcc
                 CPP : gcc -E
             HOST_CC : gcc
              CFLAGS : -O0 -Wall -I. -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -std=gnu99 -fgnu89-inline -Wno-ignored-qualifiers  -I/usr/include -DHAVE_ASTERISK -g3 -ggdb3 -pthread     -fstack-protector-all -Wmissing-declarations -Wnested-externs -Wno-long-long -Wnonnull -Wcast-align -Wno-unused-parameter -Wno-missing-field-initializers -Wstrict-aliasing -Wmissing-format-attribute -Wmissing-noreturn -Winit-self -Wmissing-include-dirs -Wunused-but-set-variable -Warray-bounds -Wimplicit-function-declaration -Wreturn-type -Wsign-compare -Wmissing-prototypes -Wshadow -ffunction-sections
            CPPFLAGS : -U_FORTIFY_SOURCE  -I/usr/include -DHAVE_ASTERISK -g3 -ggdb3
             LDFLAGS :   -Wl,-gc-sections -L/usr/lib64
            PBX_TYPE : Asterisk
          PBX_PREFIX :
             PBX_ETC :
             PBX_LIB : /usr/lib64
          PBX_VARLIB : /usr/var/lib/asterisk
         PBX_INCLUDE : /usr/include/asterisk
    Module Directory : /usr/lib64/asterisk/modules

Right direcory may be /var/lib/asterisk

    [build@ast2-22-26 chan-sccp-b]# ls /usr/var/lib/asterisk
    ls: cannot access /usr/var/lib/asterisk: No such file or directory
    [build@ast2-22-26 chan-sccp-b]# ls /var/lib/asterisk
    agi-bin  astdb.sqlite3  bin  documentation  images  keys  licenses  moh  phoneprov  rest-api      sounds  static-http  ucp

I don't know if it is problem.

dkgroot, 2015-11-19:
Hi lynt-smitka,
At the moment this should not present a real problem, but it might in the future. I will check why/where this happens and fix it. Thanks for reporting the issue.
Regards,
Diederik

closed 2015-11-19 by marcelloceschia

lynt-smitka, 2015-11-19:
Thanks, I appreciate your work!

dkgroot, 2015-11-19:
No Problem :-)

commit 79199da2 (2015-11-19): Fixes #283: setting PBX_VARLIB. Reported by lynt-smitka.
Fixes sccp_manager_action2pbx_str by adding GCC_NESTED or CLANG_BLOCK if found.

git-svn-id: https://svn.code.sf.net/p/chan-sccp-b/code/trunk@6407 43c1dc8c-776e-4df2-8ef1-6e829299be21