chan-sccp/chan-sccp #284: Mixed IPv6/IPv4 service

spape84, 2015-12-06:
Hi!

I started using IPv6. The first phone I tried to use with IPv6 unexpectedly loses registration sometimes and Asterisk crashes. This night I found an error message in the logs:

```
[Dec  6 00:40:25] ERROR[367] sccp_refcount.c: SCCP: (release) Refcount Object 0x7f7350002690 could not be found (Major Logic Error). Please report to developers
[Dec  6 00:40:25] WARNING[1100] sccp_actions.c: SCCP: Defined softkeyset: 'softkey' could not be found. Falling back to 'default' instead !
[Dec  6 00:40:25] WARNING[367] sccp_utils.c: SCCP: (backtrace)
================================================================================
OPERATING SYSTEM: Linux, ARCHITECTURE: x86_64, KERNEL: 3.19.0-32-generic
ASTERISK: 11.7.0~dfsg-1ubuntu1
CHAN-SCCP-b: Skinny Client Control Protocol (SCCP). Release: 4.2.1 STABLE - 6368M (built by 'root' on '2015-11-09 13:48:06 UTC')

To get a better backtrace you would need to install libbfd (package binutils devel package)
--------------------------------------------------------------------------(bt)--
 (bt) > /usr/lib/asterisk/modules/chan_sccp.so(sccp_refcount_release+0x3fb) [0x7f7347d025bb]
 (bt) > /usr/lib/asterisk/modules/chan_sccp.so(sccp_line_deleteLineButtonsArray+0x4f) [0x7f7347cae7cb]
 (bt) > /usr/lib/asterisk/modules/chan_sccp.so(sccp_dev_clean+0x83b) [0x7f7347c9999b]
 (bt) > /usr/lib/asterisk/modules/chan_sccp.so(destroy_session+0x258) [0x7f7347d1d6a8]
 (bt) > /usr/lib/asterisk/modules/chan_sccp.so(sccp_socket_device_thread_exit+0x16e) [0x7f7347d1da8e]
 (bt) > /usr/lib/asterisk/modules/chan_sccp.so(sccp_socket_devic
[Dec  6 00:40:25] ERROR[1100] sccp_socket.c: SEP002699EF0CF3: error 'Bad file descriptor (9)' while reading from socket.
[Dec  6 00:40:25] ERROR[1100] sccp_socket.c: SEP002699EF0CF3: stats: Register Available Lines Message (45), msgDataSegmentSize: 4, UnreadBytesAccordingToPacket: 4, bytesToRead: 4, bytesReadSoFar: 0
```

A second strange thing I found was: When setting "nat=off" for the test-phone using IPv6 Asterisk crashes as soon as the rtp stream starts. I'm also not sure, which value to set for "externip" (server is not behind a nat) and directrtp? All the SIP trunks I'm using are IPv4, my test-phone is using IPv6. Does directrtp work in this setup?

Maybe you can point me into the right direction for IPv6.

dkgroot, 2015-12-06:
Hi spape84:

Could you try out the latest revision from svn trunk/git master instead of 6368, to see if that makes a difference ?

The nat handling for ipv6 changed two days ago (in trunk) to make sure sccp reload works correctly. You don't have to set nat at all, it will be set to 'auto' by default and ipv6 handling should be ok in that case. ExternIP will be used only in case nat is detected and externip has been set. When using ipv6, we would normally not expect nat to be in use.

Only last week we figured out that directrtp was broken in the stable release. This has been corrected in the current trunk/master revision.

To use ipv6 you will also have to modify your SEP... file so that it includes both
<ipPreferenceModeControl>2</ipPreferenceModeControl> <! dual mode = 2>
<processNodeName>The ipv6 ip-address of the server</processNodeName>

spape84, 2015-12-06:
Hi Diederik!

Thanks for your help. These are all the IPv6 related options I set in my SEP file. That should be OK, right?

```
<device xsi:type='axl:XIPPhone'>
	<ipAddressMode>2</ipAddressMode>
	<ipPreferenceModeControl>1</ipPreferenceModeControl>
	<devicePool>
		<callManagerGroup>
			<members>
				<member priority='0'>
					<callManager>
						<ipv6Addr>2a01:238:434b:dead:beef:dead:beef:2f1</ipv6Addr>
						<ports>
							<ethernetPhonePort>2000</ethernetPhonePort>
						</ports>
						<processNodeName>2a01:238:434b:dead:beef:dead:beef:2f1</processNodeName>
						<processNodeName>domain.name.which-resolves.to.ipv6_and_ipv4</processNodeName>
					</callManager>
				</member>
			</members>
		</callManagerGroup>
</device>

```

I'll get back to you after testing the trunk version.

dkgroot, 2015-12-06:
<ipPreferenceModeControl>1</ipPreferenceModeControl>
Means you only want ipv6 mode (not dual mode), which is fine, just something you need to know.
Hopefully you have only one processNodeName per member entry (i guess you do). If it does resolve to both ipv4 and ipv6, you might want to use the dual mode or set the ip-address instead. 

I do hope this a stable/static ip-address. If it is actually a dynamic ip-address that would cause problems, the sccp protocol is tcp only and uses a steady ip stream between client and server, which cannot be interupted without major issues. It would potentially explain the issue descripted above where we loose the file descriptor if the ip-address changed while devices where still connected.

spape84, 2015-12-06:
Hey Diederik!

Thanks for your reply. As far as I understand the docs setting "ipPreferenceModeControl" = 1 means that the phone will first try to connect using IPv6 and if that doesn't work it tries IPv4?? That's why I set this setting to 1. Shall I set it to 2?

```
<ipPreferenceModeControl>1</ipPreferenceModeControl>
```

The ipAddressMode is still set to 2, which means dual mode according to the docs:

```
<ipAddressMode>2</ipAddressMode>
´´´

The IPv6 address of the server doesn't ever change - it's statically allocated. The IPv6 address of the phone might change when my office router reconnects the DSL line, but that happens every second month or so.

I'll have some more info about the trunk version by tomorrow. Thanks a lot for your help!

dkgroot, 2015-12-06:
Ooh you are absotely right about ipAddressMode, sorry about that one.
Ok so no issue on the ip-side then (maybe i misinterpretted your xml example post).

spape84, 2015-12-07:
Hey!

After installing the newest master/trunk version I set nat = auto in sccp.conf and tried to initiate a call. As soon as rtp is supposed to start Asterisk crashes. (see console output with "sccp set debug all")

[ast_capture.txt](https://github.com/marcelloceschia/chan-sccp-b/files/53714/ast_capture.txt)

When setting nat = on (although there is no IPv6 NAT - the phone's address is routable!) I can use the phone. Seems to be stable as long as nat if turned on.

dkgroot, 2015-12-07:
Good to know it works with NAT off (which makes most sense when using ipv6) with the latest version. I will need to see if i can reproduce the issue when nat is switched on.

Note: When setting the debug logging, it makes sense to pick carefully instead of setting 'all', which does produce a whole lot. Also in case of a crash a backtrace is more helpfull, because the actual cause of the crash will never make it to the log file. If you could, please send the gdb backtrace instead that would be very helpfull, to help fix this issue.

Thanks for reporting the issue and testing the latest revision.

dkgroot, 2015-12-07:
Ran a couple of tests with nat=on on my ipv6 phones (without a firewall actually being there) which did not segfault on my system. I would have to make quite a number of changes in my network to switch this to actually be natted. Also i don't know where your firewall is placed and what side the nat is on (Pbx / Phone). Would be easier if you would send me the backtrace instead, thanks.

spape84, 2015-12-07:
Hey Diederik!

Maybe we misunderstood... Sorry for that!

Asterisk crashes when nat = auto OR nat = off.
It doesn't crash when nat = on

nat = on - doesn't make sense on IPv6, I think?

Current setup is:

[image]

Could you please send me information on how to create the gdb backtrace? Don't have the experience doing that on Linux :-(

spape84, 2015-12-07:
Something new. Phone reconnects sometimes. Internet connection is stable...?! Strange...

```
[Dec  7 12:47:44] NOTICE[7887][C-0000000e] sccp_socket.c: SEP002699EF0CF3: Closing session because connection timed out after 33 seconds (timeout: 33000) (ip-address: 2003:48:dead:beef:dead:beef:feef:cf3).
[Dec  7 12:52:07] NOTICE[22079] sccp_socket.c: SEP002699EF0CF3: Closing session because connection timed out after 33 seconds (timeout: 33000) (ip-address: 2003:48:dead:beef:dead:beef:feef:cf3).
[Dec  7 13:15:34] NOTICE[23715] sccp_socket.c: SEP002699EF0CF3: Closing session because connection timed out after 33 seconds (timeout: 33000) (ip-address: 2003:48:dead:beef:dead:beef:feef:cf3).
[Dec  7 14:06:29] NOTICE[662][C-00000048] sccp_socket.c: SEP002699EF0CF3: Closing session because connection timed out after 33 seconds (timeout: 33000) (ip-address: 2003:48:dead:beef:dead:beef:feef:cf3).
[Dec  7 14:14:53] NOTICE[20754] sccp_socket.c: SEP002699EF0CF3: Closing session because connection timed out after 33 seconds (timeout: 33000) (ip-address: 2003:48:dead:beef:dead:beef:feef:cf3).
```

dkgroot, 2015-12-07:
gdb-process: (also see: https://sourceforge.net/p/chan-sccp-b/wiki/Reporting%20Bugs/)
1. have coredump enabled in /etc/asterisk/asterisk.conf
2. have debug symbols installed for asterisk (asterisk-dbg / asterisk-debuginfo)
3. find the 'core' file left behind by asterisk (/var/log/asterisk or /tmp most of the time), go to that directory
4. execte 'gdb asterisk core' and wait for it to load all the symbol files, you end up on a prompt
5. execute 'bt'
6. execute 'bf full'
7. send us the output (on of the last two commands)

I will recheck with 'nat=off' and 'nat=auto' to see if that makes a difference on my system. At the moment i don't have a firewall between my phones and the asteriskl, making this a bit bad simulation.

You might want to increase the keepalive time to 60 or 120, to allow a little more time. Could this possible indicate other network trouble you might be having when using ipv6 in general (routing loop or something ?)

What does auto nat detection tell you during registration of the device, does it switch to (Auto)Off after having registered. The segfault happens when you are trying to make the call not during registration (right).

dkgroot, 2015-12-07:
I just notice that the session timeout is happening during a call, that is highly unusual to be honest. Can you run a couple of network connectivity checks (latency / troughput) to see what might be going on with your ipv6 streams. Are you running ipv6-in-ipv4 tunnels or something ?

spape84, 2015-12-07:
I changed keepalive ... let's see what happens

```
keepalive=120
```

I'm going to do some network checks.
Currently the phone is connected to a managed switch, which connects to the router. Network traffic is low. There are no loops in the network.

I'll create a coredump later for you and post it.

dkgroot, 2015-12-07:
Ok, looking forward to the dump, to see what might have happened.

spape84, 2015-12-07:
bt:

```
#0  0x00007fe48c81bcc9 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007fe48c81f0d8 in __GI_abort () at abort.c:89
#2  0x00007fe48c814b86 in __assert_fail_base (fmt=0x7fe48c965830 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7fe4339ab558 "a->addr.sa_family == PJ_AF_INET || a->addr.sa_family == PJ_AF_INET6", 
    file=file@entry=0x7fe4339ab4e0 "../src/pj/sock_common.c", line=line@entry=388, 
    function=function@entry=0x7fe4339ab700 "pj_sockaddr_get_len") at assert.c:92
#3  0x00007fe48c814c32 in __GI___assert_fail (
    assertion=0x7fe4339ab558 "a->addr.sa_family == PJ_AF_INET || a->addr.sa_family == PJ_AF_INET6", 
    file=0x7fe4339ab4e0 "../src/pj/sock_common.c", line=388, function=0x7fe4339ab700 "pj_sockaddr_get_len") at assert.c:101
#4  0x00007fe4339a22ac in pj_sockaddr_get_len () from /usr/lib/libpj.so.2
#5  0x00007fe43400113f in __rtp_recvfrom (instance=instance@entry=0x7fe474004948, buf=buf@entry=0x7fe474009d08, 
    sa=sa@entry=0x7fe42dbacd30, rtcp=rtcp@entry=0, flags=0, size=8192) at res_rtp_asterisk.c:1483
#6  0x00007fe43400618c in rtp_recvfrom (sa=0x7fe42dbacd30, flags=0, size=8192, buf=0x7fe474009d08, instance=0x7fe474004948)
    at res_rtp_asterisk.c:1513
#7  ast_rtp_read (instance=0x7fe474004948, rtcp=<optimized out>) at res_rtp_asterisk.c:3529
#8  0x00007fe4412e5153 in sccp_wrapper_asterisk111_rtp_read (ast=0x7fe480008608) at pbx_impl/ast/ast111.c:398
#9  0x00007fe48ce51452 in __ast_read (chan=0x7fe480008608, dropaudio=0) at channel.c:4010
#10 0x00007fe48cef4140 in local_bridge_loop (pvt1=0x7fe480007c00, pvt0=0x7fe42000f778, rc=0x7fe42dbae6e8, fo=0x7fe42dbae6e0, 
    flags=<optimized out>, timeoutms=-1, instance1=0x7fe474004948, instance0=0x7fe420000968, c1=0x7fe480008608, c0=0x7fe420018598)
    at rtp_engine.c:1024
#11 ast_rtp_instance_bridge (c0=0x7fe420018598, c1=0x7fe480008608, flags=<optimized out>, fo=0x7fe42dbae6e0, rc=0x7fe42dbae6e8, 
    timeoutms=-1) at rtp_engine.c:1592
#12 0x00007fe48ce55ea7 in ast_channel_bridge (c0=c0@entry=0x7fe420018598, c1=c1@entry=0x7fe480008608, 
    config=config@entry=0x7fe42dbaec80, fo=fo@entry=0x7fe42dbae6e0, rc=rc@entry=0x7fe42dbae6e8) at channel.c:7990
#13 0x00007fe48ce91090 in ast_bridge_call (chan=chan@entry=0x7fe420018598, peer=peer@entry=0x7fe480008608, 
---Type <return> to continue, or q <return> to quit---
    config=config@entry=0x7fe42dbaec80) at features.c:4483
#14 0x00007fe4499c2b7e in dial_exec_full (chan=0x7fe420018598, data=<optimized out>, peerflags=peerflags@entry=0x7fe42dbaf6b0, 
    continue_exec=continue_exec@entry=0x0) at app_dial.c:3045
#15 0x00007fe4499c3b16 in dial_exec (chan=<optimized out>, data=<optimized out>) at app_dial.c:3128
#16 0x00007fe48cedb22e in pbx_exec (c=c@entry=0x7fe420018598, app=app@entry=0x7fe48ef5ade0, 
    data=data@entry=0x7fe42dbafc60 "SCCP/796535,20") at pbx.c:1622
#17 0x00007fe48cee8594 in pbx_extension_helper (c=c@entry=0x7fe420018598, con=con@entry=0x0, context=<optimized out>, 
    exten=exten@entry=0x7fe420019438 "12926535", priority=priority@entry=4, label=label@entry=0x0, 
    callerid=callerid@entry=0x7fe4800009d0 "0015154733105", action=action@entry=E_SPAWN, found=found@entry=0x7fe42dbb3d0c, 
    combined_find_spawn=combined_find_spawn@entry=1) at pbx.c:4922
#18 0x00007fe48ceea5ee in ast_spawn_extension (combined_find_spawn=1, found=0x7fe42dbb3d0c, 
    callerid=0x7fe4800009d0 "0015154733105", priority=4, exten=0x7fe420019438 "12926535", context=<optimized out>, 
    c=0x7fe420018598) at pbx.c:6038
#19 __ast_pbx_run (c=c@entry=0x7fe420018598, args=args@entry=0x0) at pbx.c:6513
#20 0x00007fe48ceebb6b in pbx_thread (data=data@entry=0x7fe420018598) at pbx.c:6843
#21 0x00007fe48cf2ab5e in dummy_start (data=<optimized out>) at utils.c:1162
#22 0x00007fe48b56d182 in start_thread (arg=0x7fe42dbb4700) at pthread_create.c:312
#23 0x00007fe48c8df47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

bt full:

```
#0  0x00007fe48c81bcc9 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
        resultvar = 0
        pid = 4333
        selftid = 4471
#1  0x00007fe48c81f0d8 in __GI_abort () at abort.c:89
        save_stage = 2
        act = {__sigaction_handler = {sa_handler = 0x7fffb9b5b8b5, sa_sigaction = 0x7fffb9b5b8b5}, sa_mask = {__val = {
              140619587919132, 140618095047904, 388, 0, 140619586561251, 4294967296, 140617996486272, 163975972736, 
              140617996486336, 140617996486432, 0, 0, 0, 21474836480, 140619592527872, 140619587934256}}, sa_flags = 865776984, 
          sa_restorer = 0x7fe4339ab700}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007fe48c814b86 in __assert_fail_base (fmt=0x7fe48c965830 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", 
    assertion=assertion@entry=0x7fe4339ab558 "a->addr.sa_family == PJ_AF_INET || a->addr.sa_family == PJ_AF_INET6", 
    file=file@entry=0x7fe4339ab4e0 "../src/pj/sock_common.c", line=line@entry=388, 
    function=function@entry=0x7fe4339ab700 "pj_sockaddr_get_len") at assert.c:92
        str = 0x7fe4800056a0 "\300K"
        total = 4096
#3  0x00007fe48c814c32 in __GI___assert_fail (
[... 233 lines cut ...]
#20 0x00007fe48ceebb6b in pbx_thread (data=data@entry=0x7fe420018598) at pbx.c:6843
        c = 0x7fe420018598
#21 0x00007fe48cf2ab5e in dummy_start (data=<optimized out>) at utils.c:1162
[... cut ...]
                -1433794484950763908, -1433430728137083268}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0, 0x0}, data = {
              prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = <optimized out>
        pagesize_m1 = <optimized out>
        sp = <optimized out>
        freesize = <optimized out>
        __PRETTY_FUNCTION__ = "start_thread"
#23 0x00007fe48c8df47d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
---Type <return> to continue, or q <return> to quit---
No locals.

dkgroot, 2015-12-07:
Hi S.Pape,

you are bridging ipv4 on one side and ipv6 on the other. When starting the rtp port on the sccp phone we don't know what the remote side will be using. On a ipv6 device we assume everyone to be using ipv6 and create an ipv6 port on the device. 

When nat is off, we also tell asterisk to use that ip-address directly. Asterisk should not allow a directrtp bridge between ipv4-ipv6 phone, so directrtp will be disabled. This means we will be talking to asterisk and not directly to the remote end. 

When nat is on, we would substitute the phones ipaddress to whatever the externip is set to, when that is a ipv4 address on your server, we would create and present ipv4 address to asterisk.

Now that being said:
* Segfault is happening inside pj_sip and not the chan-sccp-b side, meaning this is out of our control (not a 100% sure if we are not causing this somewhere though).
* the ss_family seems to change from 10 to 1 when crossing the border between asterisk and pj_sip which is perculiar (not 100% if dealing with the same ip-address here).

Note: We have mostly tested all of the nat material using asterisk-11 + chan_sip. I personally don't have a lot of experience with pj_sip (yet). Would like to know if the same issue happens on this server, when using chan_sip instead. And/Or what would happen if chan_pjsip was also running in ipv6 mode.

@marcelloceschia: Do you have anything to contribute ?

Not sure how to go forward from here.

spape84, 2015-12-07:
Hey Diederik!

I'm not using PJSIP although there are some prefixes named "PJ" in my dump.
I'm using chan_sip.so (I double checked a moment ago).

What I tried now was turning on IPv6 for SIP as well. It binds to IPv6 and IPv4 now.

I first tried calling with nat = yes

That still works. I captured the message from the console for this call (firstdump.txt)

```
core set verbose 10
core set debug 10
sccp set debug all
sip set debug peer qsc
rtp set debug on
```

I then tried using nat=auto. That let Asterisk crash again. (seconddump.txt and another core.txt)

Could you please share your email address with me as I don't want to post the whole info on this board. There is lots of information included, which should not be seen by too many people. You may contact me using c o n t a c t @ s e b a s t i a n p a p e . d e

I'll remove my address after I received your contact info...

Thank you.

dkgroot, 2015-12-07:
Not sure where those pj_ prefixes come from, i will look into that. Running chan_sip here as well, both with nat=on / off and auto. Would love to figure out what is going on in this instance.

dkgroot, 2016-02-20:
Could you confirm if the issues mentioned above still exist in the current version ? Or if they have been solved in the meantime. Some work was done on both session handling and refcount issues.

dkgroot, 2016-03-23:
No Response, closing issue for now.

closed 2016-03-23 by dkgroot