chan-sccp/chan-sccp #290: Update Wiki Pages

dkgroot, 2016-04-24:
After the migration from sourceforge some of the links need to be update.
Wiki content is a little outdated at some places.
New Wiki content is still required

dkgroot, 2016-04-25:
Work in progress. Could use some help !

marcelloceschia, 2016-04-25:
@dkgroot Hilfe, bei was bzw wie willst Du es aufteilen?

dkgroot, 2016-04-26:
I think we are mostly done :-) 
Fixed the FreePBX Section
@dandrzejewski is going to have another look, to see if anything is missing / requires updating etc... Thanks David.

dkgroot, 2016-04-26:
Done:
* Moved FreePBX related pages around and change the Menu order. 

Todo:
* Still need to merge the Setup FreePBX and the FreePBX page content into one (don't know which is better/newer).
* I think we need more screen captures (color: 7970 and 7960: if necessary)
* Setting up /etc/asterisk/logger.conf 
* More feature coverage

closed 2016-05-15 by dkgroot