chan-sccp/chan-sccp #293: Transfer leaves an open file descriptor after call completes

dkgroot, 2016-04-28:
Issues Reporter: Miguel
Date: 2016/04/28

**Reproduction Steps:**

ls -l /proc/`pidof asterisk`/fd/*;wc -l 
Make a simple transfer. 
ls -l /proc/`pidof asterisk`/fd/*;wc -l 
Number goes up

**Expected behavior:**
Number should go up by one during the transfer and go back down afterwards. We might have an issue with timer_fd (known from the hangup issue)

**Observed behavior:**

Number of open files goes up by one, during the transfer off a call. And does not drop back down after the transfer has completed.

**chan-sccp version:** V4.3.0_DEV

**asterisk version:** Asterisk-11.21.1
*

miguelsanzpardo, 2016-04-29:
The same occurs when you make a gpickup with directed_pickup_modeanswer = no

With directed_pickup_modeanswer = yes the problem doesnt appear.

dkgroot, 2016-04-29:
Good to know, thanks for the heads up. Have not had time to look at the issue yet.

dkgroot, 2016-04-29:
Asterisk compile flag: 'DEBUG_FD_LEAKS' will hopefully help while searching for the issue.

miguelsanzpardo, 2016-04-29:
I can reproduce the issue on my test machine, 

I have put maxfiles = 35 (very low) and then the same errors appears on /var/log/asterisk/full

I can compile again asterisk with Build Options:               
AST_DEVMODE, 
DONT_OPTIMIZE, 
DEBUG_THREADS, 
LOADABLE_MODULES, 
BETTER_BACKTRACES, 
MALLOC_DEBUG, 
DETECT_DEADLOCKS
DEBUG_FD_LEAKS

And send the full.log or whatever you think that can help us to solve these problems

miguelsanzpardo, 2016-04-29:
With DEBUG_FD_LEAKS activated,can we find what processes are increment the file descriptors?

dkgroot, 2016-04-29:
To be honest: I have not used that compilation flag before, so i am also checking that at the moment.

miguelsanzpardo, 2016-04-29:
Could you test on your system if the issue appears with transfer and gpickup with directed_pickup_modeanswer = yes?

dkgroot, 2016-04-29:
Using asterisk-13 i don't see the increase in fd's after a transfer. It returns to the base level.
Switching to asterisk-11 (recompiling with DEBUG_FD_LEAKS).

miguelsanzpardo, 2016-04-29:
I am recompiling Asterisk 11 with DEBUG_FD_LEAKS too

dkgroot, 2016-04-29:
Please switch to the gitter chat instead of doing this here !

dkgroot, 2016-04-29:
The issue is only detectable in asterisk-11
The issue is cause when calling 
ast_read is called without ast_waitfor() reporting that there is data to be read, see:

    main/channel.c: ast_read() : line 3925     (only active in AST_DEVMODE)

Will need to find what is causing the extra ast_read in our code. And why this only happens during a transfer / pickup. I guess the masquerading code is not stopping / started rtp flow during a masquerade.

I think we should re-introduce the ast_channel lock during read and write events. Will run a couple of tests to see if that does help fix the issue.

dkgroot, 2016-04-30:
The potential solution i had above, does not seem to help / work out. I still don't know why the timerfd in ast11 get's left open after hangup. It seems to have been a long standing problem in the past (ever since ast1.6) with function / applications and channel drivers.

miguelsanzpardo, 2016-04-30:
Ok, it doesnt sound so good, then If I have undestand in a good manner this problem, it looks like to appear only on 11 and maybe on 1.6 but not on 1.8 and 13.

By now, what do you recommend me to do, in addition to reboot the system every X days?

It is supposed that if I dont use debug the problem doesnt appear, true? I was using SCCP 4.3.0.6802 along 2 weeks without problems (and I was using Asterisk 11.21.2 with developer flags activated but with core verbose=0 and  core debug=0)
When I change to 4.3.0.6808 I put debug=1 and verbose=3 and then the problem appears.
Next monday I will test with debug=0, now I cant test anything at home

dkgroot, 2016-04-30:
I only said that i cannot reproduce it on 13. I can in 11, have not tested/checked 1.6 or 1.8.
The settings /  flags won't matter (as far as i can tell). 
Don't do anything for now. It will require some more work on my part. I need to find out how this timerrd is created and destroyed and how it is tight to channel hangup. It will require some tedious debugging on my part and will take time.

dkgroot, 2016-05-01:
@miguelsanzpardo 
This issue should be resolved in the 'develop' branch using revision 95d0648 (or higher).
You can easily monitor the open 'fds' which where causing the issue using:
watch -n3 "ls -l /proc/`pidof asterisk`/fd/* | grep timerfd" 
Without the patch, every transfer/ pickup / cfwdall would add 1 timerfd which would not be cleaned up. WItht the patch this is fixed.

commit 95d0648b (2016-05-01): Fix: File Descriptor Leak after channel masqueration (transfer / cfwd / pickup)

The TimerFD does not get moved/merged during fixup (masqueration).
I could not find the actual cause for this in the asterisk sources.
This patch does fix the issue if it does exist during channel cleanup though.

Fixes #293
closed 2016-05-01 by dkgroot