chan-sccp/chan-sccp #294: Blind transferring does not work while destination is still ringing

andybradfordcms, 2016-04-29:
Blind transfer seems to have stopped working in recent revisions beyond the stable release - the softkey fails to appear. 

**Reproduction Steps:**

1. Be connected to a party on the device
2. Invoke the transfer dialogue and dial the destination party number (internal or external) 
3. Note that the destination party rings, but will not show the transfer softkey until the destination party answers, thus "blind" transferring doesn't work. 

**Expected behavior:**

Expected behavior would be for the transfer softkey to show up during the destination party still being in the ringing state, pre-pickup. 

**Observed behavior:**

Softkey is not available when the steps are gone through. 

**Screenshots**

No screenshots available 

**chan-sccp version:** 7a9996b and be98253 

**asterisk version:** 11.16.0 

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes]
* Problem can be reliably reproduced, doesn't happen randomly: [Yes]
* Platform:
    * [CPU]
    * [32/64 Bit]
    * [Big/Small Endian]
    * [Operating System] CentOS 
    * [Operating System Version] release 6.7 (Final)  
* Direct Contact information: andy_bradford_cms@yahoo.com 
* Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]
* Comments: [Other Information]
*

dkgroot, 2016-05-01:
> Hi Diederik,
> Thanks for taking a look into this one so quickly. I went back to look at the closure details, but couldn't correlate this to any particular commit update. Was there a patch put in place for this, or anything identified?
> Thanks,
>       o [CPU]
>       o [32/64 Bit]
>       o [Big/Small Endian]
>       o [Operating System] CentOS
>       o [Operating System Version] release 6.7 (Final)
>   * Direct Contact information: andy_bradford_cms@yahoo.com <mailto:andy_bradford_cms@yahoo.com>

Hi Andy,

There was a commit (notified in the issue as well (just before the 'dkgroot closed this...'), namely revison: |ce188f1 <https://github.com/chan-sccp/chan-sccp/commit/ce188f1f100da630c89110c4fe1c08122426caaa>|
ie: https://github.com/chan-sccp/chan-sccp/commit/ce188f1f100da630c89110c4fe1c08122426caaa
in the 'develop' branch

git clone https://github.com/chan-sccp/chan-sccp chan-sccp_develop
cd  chan-sccp_develop
git checkout develop
./configure ....
make
make install
make reload

The develop branch is used for new commits (just like svn 'trunk' used to be). Once it has been tested and everything is still stable. The changes are moved to the master branch (takes 1-2 weeks).

Regards,

Diederik

On 04/29/2016 06:26 PM, andybradfordcms wrote:

dkgroot, 2016-05-02:
@andybradfordcms / @miguelsanzpardo

Can you guys test / check revision: ce188f1 or later please, to make sure that this change fixed the issue for these four cases:

                 | transfer_on_hangup = off | transfer_on_hangup = on
------------- | -------------------------------- | --------------------------------
blind          |       <OK / Not OK>           | <OK / Not OK>
attended   |       <OK / Not OK>            | <OK / Not OK>

If one of them is not working, please attach the output from
'sccp debug core, device, channel, pbx'

Thanks

miguelsanzpardo, 2016-05-03:
I have recompile the latest rev. bd49e2eM

By now I have made the test with: transfer_on_hangup = off, I never have used transfer_on_hangup = on, I need to figure out how it really works

;transfer_on_hangup = no / yes
Complete transfer on hangup, without pressing transfer a second time. Will complete transfer, when the transferer puts the receiver on hook, after the destination has been reached. To cancel the transfer, either press resume on the transfered channel, press the 'endcall' softkey, or have the receiving party hangup first

Attended transfer(Not OK):
1101 calls to 1105
1105 answer the call
1105 press TRANSFER button
1105 call 1104
1104 answer the call
1105 press TRANSFER button again and then the call is finished

Blind transfer(Not OK):
1101 calls to 1105
1105 answer the call
1105 press TRANSFER button
1105 call 1104
1105 press TRANSFER button before 1104 answer the call
1104 offhook the phone several times but it can answer the call unless press the line button 1104

[full-attended transfer.txt](https://github.com/chan-sccp/chan-sccp/files/246459/full-attended.transfer.txt)
[full-blind transfer.txt](https://github.com/chan-sccp/chan-sccp/files/246460/full-blind.transfer.txt)

miguelsanzpardo, 2016-05-03:
I have recompile the latest rev. e4528d7M

The test is the same but in this case I dont have any problems.

[full-attended transfer-OK.txt](https://github.com/chan-sccp/chan-sccp/files/246502/full-attended.transfer-OK.txt)
[full-blind transfer-OK.txt](https://github.com/chan-sccp/chan-sccp/files/246501/full-blind.transfer-OK.txt)

miguelsanzpardo, 2016-05-03:
[full-blind transfer-OK-after reboot phone.txt](https://github.com/chan-sccp/chan-sccp/files/246530/full-blind.transfer-OK-after.reboot.phone.txt)
[full-blind transfer-OK-after reboot.txt](https://github.com/chan-sccp/chan-sccp/files/246531/full-blind.transfer-OK-after.reboot.txt)

dkgroot, 2016-05-04:
Hi Miguel, the open rtp port issue should be fixed using the latest revision. Rolled back #297 and change indication handling. Would you mind rechecking that transfer works correctly again.

miguelsanzpardo, 2016-05-09:
Solved on commit f6edd

commit ce188f1f (2016-05-01): Fix: Blind transferring does not work while destination is still ringing

fixes #294
closed 2016-05-10 by dkgroot