chan-sccp/chan-sccp #295: Parked calls should flash amber

dandrzejewski, 2016-04-30:
When a call is placed in park and there's a hinted button on a phone for that parking slot, the button lights red and the screen indicates off-hook.  Instead, it should probably flash amber.

**Reproduction Steps:**

Put the following in a config:

button = speeddial,Park 71,71,71@park-hints
Park a call.

**Expected behavior:**

Button should flash amber.

**Observed behavior:**

Button shows solid red.

**Screenshots**

**chan-sccp version:** (include revision if relevant)

4.3, current trunk as of this evening.

**asterisk version:** (include revision if relevant)

13.7.2

**Addition / Optional Information:**

dkgroot, 2016-04-30:
@dandrzejewski The hinted speeddial does not have any knowledge about what it is observing, we only know the extension and the context, not what it is. So we can't react to it in a different way than any other hint. It would require parsing the asterisk dialplan to figure out what it might be and figuring out what the user might expect. We don't have the dialplan parsing capability, nor do we  have an AI module (yet) to predict the users wishes :-)

There is really not much we can do, sorry.

dandrzejewski, 2016-04-30:
Fair enough. I was thinking about something simpler, perhaps adding options to the button configuration (e.g. "this device state = this button state"

dkgroot, 2016-04-30:
More button config parameter. I think we have to many already.

closed 2016-05-01 by dkgroot