chan-sccp/chan-sccp #297: Current caller name on speed dial line doesn’t update after call transfer

drtmk, 2016-05-01:
Alice’s phone configuration:
button 1 – Alice’s line
button 2 – speeddial line Bob

Bob’s phone configuration:
button 1 – Bob’s line
button 2 – speeddial line Alice

Steps to Reproduce:
1. From Alice’s phone, call to Bob’s phone.
2. Pickup the call on Bob’s phone.
Now on Bob’s phone next to speeddial line, you see something like „Bob <=> Alice” – meaning that Alice and Bob are talking to each other.
3. From Bob’s phone, transfer this call to other phone – Eve’s.
Now Alice is talking with Eve, but the speeddial line on Bob’s phone that monitors Alice’s line, is still showing previous state: Bob <=> Alice which is incorrect. The label should be updated and display „Eve <=> Alice”.

Asterisk 13.8.2
chan_sccp 4.2.3
Two cisco 7942G phones

dkgroot, 2016-05-01:
Partial fix committed. The recipent part of the transfer now gets update. Still need to figure out how to force the hint update of the transferee (WIP).

dkgroot, 2016-05-04:
Change rolled back

dkgroot, 2016-05-26:
Sometimes the callinfo update does not always happen on attended trasnfer (3 out of 5) 
```
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_line.c: -- SCCP: Found line 1101
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: (getEmptyChannel) on line:1101, maybe_c:
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: (getEmptyChannel) getActiveChannel
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_device.c: -- SEP0026CBC03B37: Getting the active channel on device.
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: (getEmptyChannel) got channel already.
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_channel.c: -- SEP0026CBC03B37 can't put on hold an inactive channel SCCP/1101-00000007 with state DIALING (14)... cancelling hold action.
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_device.c: -- SEP0026CBC03B37: ( sccp_channel.c:1563:sccp_channel_hold ) sccp_dev_displayprompt '€-' for line 1 (5)
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_protocol.c: -- SEP0026CBC03B37: Display prompt on line 1, callid 7, timeout 5
[2016-05-26 16:15:15] ERROR[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: Putting Active Channel SCCP/1101-00000007 OnHold failed -> Cancelling new CaLL
[2016-05-26 16:15:15] ERROR[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: Can't allocate SCCP channel for line 1101
````
More debug requred, 
'sccp debug core, channel, pbx, callinfo`
To show when callinfo is updated and when it is not

miguelsanzpardo, 2017-03-29:
Using:
- The actual SCCP develop version: Release: 4.3.0 develop - 128c709M 
- Asterisk 11.21.2
- Phone Model: CISCO 7942-7962 with firmware 9.4.2
- Plain dialplan

Example:
1103-Ricard
1101-Oscar (Has Speedial buttons for: 1102-Miguel and 1103-Ricard)
1102-Miguel

* Blind Transfer:
Ricard call to Oscar and Oscar transfer to Miguel.
Once the transfer is done Oscar can see on his BLF Speedials something like that:
Ricard <=> 1102-Miguel (OK)
Miguel <=> 1103-Ricard (OK)
The info. showed on the left of the speedial is correct.

* Attended Transfer:
Ricard call to Oscar and Oscar transfer to Miguel.
Once the transfer is done Oscar can see on his BLF Speedials something like that:
Oscar <=> 1102-Miguel (Not OK)
Oscar <=> 1103-Ricard (Not OK)
The info. showed on the left of the speedial is not correct.