chan-sccp/chan-sccp #299: (Rejected) Transfer causing stuck/ghost channels

dkgroot, 2016-05-10:
Reported by: Valerio & Brian
Subject: 'sccp-chan-b goes in limbo after rejected transfer'
Subject: '4.3.0 built 6483 memory and channel leaking'

Issue: Something is causing a session disconnect and subsequent sccp_dev_cleanup in the middle of a transfer. The session is cancelled from the chan-sccp side, the device is not closing the socket.

SubIssue: sccp_dev_cleanup, should cleanup correctly even when a call is currently being transfered

Reference: e670e
Reference: 4efe58

Logging:
~~~~~
> <... misc dialplan messages about the incoming call ....>
> [2016-05-09 17:14:49] NOTICE[17467][C-000003c3] ast113.c: SCCP: (allocPBXChannel) Create New Channel with name: SCCP/100-00000EEE
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] app_stack.c: SCCP/100-00000EEE Internal Gosub(func-apply-sipheaders,s,1) start
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] pbx.c: Executing [s@func-apply-sipheaders:1] NoOp("SCCP/100-00000EEE", "Applying SIP Headers to channel") in new stack
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] pbx.c: Executing [s@func-apply-sipheaders:2] Set("SCCP/100-00000EEE", "SIPHEADERKEYS=") in new stack
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] pbx.c: Executing [s@func-apply-sipheaders:3] While("SCCP/100-00000EEE", "0") in new stack
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] app_while.c: Jumping to priority 7
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] pbx.c: Executing [s@func-apply-sipheaders:8] Return("SCCP/100-00000EEE", "") in new stack
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] app_stack.c: Spawn extension (from-internal-xfer, 100, 1) exited non-zero on 'SCCP/100-00000EEE'
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] app_stack.c: SCCP/100-00000EEE Internal Gosub(func-apply-sipheaders,s,1) complete GOSUB_RETVAL=
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] ast113.c:     -- SCCP: Asterisk request to call SCCP/100-00000EEE (dest:100, timeout: 0)
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] sccp_pbx.c:     -- 100: Asterisk request to call SCCP/100-00000EEE
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] app_dial.c: Called SCCP/100
> [2016-05-09 17:14:49] VERBOSE[17467][C-000003c3] app_dial.c: SCCP/100-00000EEE is ringing
> [2016-05-09 17:14:49] WARNING[17209] sccp_utils.c: SCCP: (backtrace)
> ================================================================================
> OPERATING SYSTEM: Linux, ARCHITECTURE: armv7l, KERNEL: 3.10.80-94
> ASTERISK: 13.8.2
> CHAN-SCCP-b: Skinny Client Control Protocol (SCCP). Release: 4.3.0 TRUNK - 6822M (built by 'root' on '2016-04-22 03:15:47 UTC')
>
> To get a better backtrace you would need to install libbfd (package binutils devel package)
> --------------------------------------------------------------------------(bt)--
> ================================================================================
>
> [2016-05-09 17:14:49] NOTICE[17209] sccp_device.c: SEPnnnnnnnnnn: (sccp_dev_clean) Called From: sccp_session.c:554:destroy_session
> [2016-05-09 17:14:49] VERBOSE[17209] sccp_device.c:  SCCP: Clean Device SEPnnnnnnnnnn
> [2016-05-09 17:14:49] VERBOSE[17209] sccp_device.c:   == SCCP: Remove Line 100 from device SEPnnnnnnnnnn
> [2016-05-09 17:14:49] VERBOSE[17209] sccp_device.c:   == SCCP: Unregister Device SEPnnnnnnnnnn
> [2016-05-09 17:14:49] VERBOSE[23133] sccp_devstate.c:     -- SEPnnnnnnnnnn: (sccp_devstate_deviceRegisterListener) device unregistered
> [2016-05-09 17:5:04] VERBOSE[17467][C-000003c3] app_dial.c: Nobody picked up in 15000 ms
> <.... there's a bunch of other dialplan-related messages related to "transferring" the call to the extension's voicemail. Then a few minutes later the phone reconnects ...>
> [2016-05-09 17:18:07] VERBOSE[23162] sccp_session.c:     -- SCCP: Accepted Client Connection from 10.x.x.y:53244
> [2016-05-09 17:18:07] VERBOSE[23162] sccp_session.c:     -- SCCP: Connected on server via 10.x.x.z:42337
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:   == Alarm Type: LastOutOfServiceInformation
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:   == Reason Enum: 14
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:   == Last Event Sent: 0:KeepAlive
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:   == Last Event Received: 330:CallInfoV2
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:     -- SEPnnnnnnnnnn: Device Not NATTED. Device IP '10.x.x.y' falls in localnet scope
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:     -- SEPnnnnnnnnnn: Joint protocol capability : 22
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:     -- SEPnnnnnnnnnn: Ask the phone to send keepalive message every 24 seconds
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:     -- SEPnnnnnnnnnn: using softkeyset: umc_softkeyset (0xbf7728)!
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:     -- SEPnnnnnnnnnn: Sending ConfigStatMessage, lines 1, speeddials 17
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:     -- SEPnnnnnnnnnn: Device has 8 Capabilities
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_actions.c:     -- SEPnnnnnnnnnn: Got Forward Status Request.  Line: 1
> [2016-05-09 17:18:08] VERBOSE[17551] sccp_device.c:     -- SEPnnnnnnnnnn: Device registered; performing post registration tasks...
> [2016-05-09 17:18:08] VERBOSE[23132] sccp_devstate.c:     -- SEPnnnnnnnnnn: (sccp_devstate_deviceRegisterListener) device registered
~~~~

mscdex, 2016-06-15:
I left a note on the mailing list, but I'll add the update here as well (which is better by the way?).

We are still experiencing this problem even with the develop branch from May 10. At this point we will need to restart Asterisk every night to ensure we don't continue to run into this issue. If there is something we can do to help debug this, let me know.

ValerioParisMitritsakis, 2016-07-04:
@mscdex so does restarting asterisk at night help? I tried unloading and loading the module at night but that did not help at all.

mscdex, 2016-07-04:
@ValerioParisMitritsakis Yes, I ended up setting up a daily cron job that restarts Asterisk at midnight. I think reloading just the sccp module will not work because I think it's an Asterisk (core) channel resource that's getting "stuck" although I am not an expert by any means when it comes to the Asterisk/chan-sccp-b internals.

jonasius, 2016-07-04:
Hi, I've the same issues. If I can help in any way please, let me now.

dkgroot, 2016-12-27:
@Jonasius / @ValerioParisMitritsakis  / @mscdex 
This could be related to Issue/358

Would you guys mind testing this with the issues/358 branch:
```
git fetch -a
git pull -a
git checkout issues/358
make -j2 && make install && make reload
```
I would really like to get rid of any `nightly restarts` if not absolutely required.

jonasius, 2017-01-26:
@dkgroot 
Hi Diederik, sadly I think I'm not able to test this before april.

dkgroot, 2017-01-27:
Hi Jonasius,

These transfer issues should have been resolved in the meantime (No further reports). There still is a issue with a blind transfer to a non existent extension that might cause issue. I am working on a solution for that issue.
If you can only test again in april, that new code will be committed by then :-)

dkgroot, 2017-02-25:
@ValerioParisMitritsakis / @mscdex / @jonasius 

I Started a little project, to cleanly rewrite the whole transfer process in a separate branch called feature/transfer. Would you mind helping out to test/check this (test environment only at this moment).
```
git checkout feature/transfer
./configure [... your configure flags...]
make -j2 && make reload
```
Mind you, i am still working on getting the callinfo correct, so please do not pay attention to that to much.

If you have a test environment available, and would like to help out, please do a couple of tests for me.

mscdex, 2017-04-05:
I thought I'd provide an update: we've had ce8bce8 (develop branch) running now since February 1st and have processed ~4600 calls and have had no stuck channels. Typically we would see stuck channels at around the little over 2 week and ~2200 calls mark (not sure which metric mattered).

We haven't tried the feature/transfer branch since the current build seems to be working well for now.

jonasius, 2017-04-14:
@dkgroot 
I'm able to do some testings for you, tell me what I should do.

Beside that I've issues transferring calls which I initiated over TAPI but for this I'll create a new issue.

dkgroot, 2017-09-14:
@jonasius,
Helping with testing 'feature/transfer' should only involve checking out the 'feature/transfer' branch and doing a couple of transfers.

Since yesterday, transfer on asterisk-11 should also include the correct callerid information. On asterisk-13 and above, a little more work is needed.

Also note, this branch contains a new method to do blind transfer. Press 'Transfer' Softkey and then Press the 'Transfer' softkey again, before entering the destination. Once the number has been entered/completed, the call will be removed from your device immediately and the destination starts ringing. This is using the standard asterisk method for blind transfer (compared to faking it with the old code).

closed 2019-12-17 by dkgroot