chan-sccp/chan-sccp #300: Visual parking lot menu comes back on screen after exiting and un-parking elsewhere

jamesmz, 2016-05-14:
Using latest commit in development branch and 7960 phones.

If two calls are parked using separate phones on the system, then on a third phone the visual parking lot menu is displayed and exited, when un-parking a call elsewhere the menu returns by itself on the third phone.

dkgroot, 2016-05-14:
Related: #139

dkgroot, 2016-05-14:
Note: Requires 'sccp debug core, pbx, channel, newcode, action' during testing, to show all the interaction, we are now handling the menu interaction via sccp_action.c: handle_device_to_user.

jamesmz, 2016-05-14:
Seems to be working correctly in the latest release, thank you.

dkgroot, 2016-05-14:
Ok, nice !
Closing this issue (You can re-open, report a new issue or add a comment on #139).

commit 09291470 (2016-05-14): Enh: Use IPPhoneMenu instead of IPPhoneDirectory for VisualParkingLot handling

Fix: Visual parking lot menu comes back on screen after exiting and un-parking elsewhere
fixes #300

Fix: '#' not dialled to instantly un-park when retrieving single parked call
New version does not depend on the device to do the dialing, we dial the parkinglot extension for the device instead.
fixes #301
closed 2016-05-14 by dkgroot