chan-sccp/chan-sccp #301: # not dialled to instantly un-park when retrieving single parked call

jamesmz, 2016-05-14:
Using latest commit in development branch and 7960 phones.

When using the feature button to instantly retrieve a single parked call without the visual menu, the extension number is dialled but no # is added afterwards. The dial button must also be pressed or wait for the digit timeout. When using the menu the # is always added as normally.

Only a problem with non-java phones I presume.

dkgroot, 2016-05-14:
When there is only a single parked call, we do not dial a number from the phone, we just create a new channel and send the number directly to the dialplan. The phone just get's the new channel presented to it. There should not be any need to start a digittimeout timer, nor include a '#' termination character. 
So it's a little curious that this does not work on your system (can you send me a copy of your sccp.conf) + debuglogging (sccp debug core, channel, pbx, newcode,) so i can see what's going on.

Besides: it's also curious why simulated enbloc dialing is not working in general on your system (even if we would be dialing digit for digit, which should not be the case in this instance). Do you have it disabled in sccp.conf ? If not, can you show me the debug output (sccp debug core, channel, action), when the phone is sending the digits. Maybe there is something off in the calculations.

jamesmz, 2016-05-14:
Ah I see. Sorry it was down to users pressing the button before taking the phone off hook or pressing loudspeaker. They were expecting the button to activate the loudspeaker, which is what our previous system did. So it isn't actually an issue, my bad.

On that subject though, is there any way to make the button activate the loud speaker when retrieving a single parked call, if the phone is not off hook? That would put an end to the confusion at least.

For the enbloc issue, my sccp.conf is as follows (I've removed all of the other phones but their settings are the same as the first):

;============================================================================$
;
; general definitions
;
;============================================================================$
[general]
servername = FreePbxSCCP
keepalive = 60
debug = core, newcode, filelinefunc
context = from-internal
dateformat = D.M.Y
bindaddr = 0.0.0.0
port = 2000
disallow=all
allow=alaw
allow=ulaw
allow=g729
firstdigittimeout = 16
digittimeout = 2
autoanswer_ring_time = 1
autoanswer_tone = 0x32
remotehangup_tone = 0x32
transfer_tone = 0
callwaiting_tone = 0x2d
musicclass=default
language=en
deny=0.0.0.0/0.0.0.0
permit=192.168.0.0/255.255.255.0
dndFeature = on
sccp_tos = 0x68
sccp_cos = 4
audio_tos = 0xB8
audio_cos = 6
video_tos = 0x88
video_cos = 5
echocancel = on
silencesuppression = off
private = on
callanswerorder=oldestfirst
directed_pickup_modeanswer = on
hotline_enabled=yes                                     ;can devices without $
hotline_context=default                                 ; context for hotline
hotline_extension=111                                   ; extension will be d$

;============================================================================$
;
; actual definitions
;
;============================================================================$
[SEP001647F247F5]
description = Middle Desk 301
devicetype = 7940
park = on
button=line,301
button = feature,Parked Calls,ParkingLot,default
cfwdall = off
type = device
keepalive = 60
;tzoffset = +2
transfer = on
park = on
cfwdall = off
cfwdbusy = off
cfwdnoanswer = off
directed_pickup = on
directed_pickup_context = from-internal
directed_pickup_modeanswer = on
deny=0.0.0.0/0.0.0.0
permit=192.168.0.0/255.255.255.0
dndFeature = on
nat=auto
directrtp=off
earlyrtp = progress
private = on
mwilamp = on
mwioncall = off
setvar=testvar=value
cfwdall = on

[301]
id=301
type=line
label=301
description=301
mailbox=510
vmnum=*97
cid_name=Middle Desk
cid_num=301
accountcode=501
callgroup=1
pickupgroup=1
context=from-internal
incominglimit=1
transfer=on
meetmenum=70
trnsfvm=510
musicclass=default
language=en
echocancel=on
regexten=510
silencesuppression=off

dkgroot, 2016-05-14:
> Ah I see. Sorry it was down to users pressing the button before taking the phone off hook or pressing loudspeaker. They were expecting the button to activate the loudspeaker, which is what our previous system did. So it isn't actually an issue, my bad.

The feature button should work either way. And the source should be fixed, if that is not the case. Will look into this.

dkgroot, 2016-05-14:
> For the enbloc issue, my sccp.conf is as follows (I've removed all of the other phones but their settings are the same as the first):

sccp.conf looks fine as far as i can see. Still waiting for the debug log information.

jamesmz, 2016-05-14:
I've just done a full power down and reboot (asterisk and all phones), and the speaker is now activating with the button as intended.

I'll let you know if it fails again.

jamesmz, 2016-05-14:
Retrieving a parked call by dialling the extension number manually and waiting for the digit timeout

[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_device.c:1859 sccp_device_getActiveChannel:     -- SEP001647BC225C: Getting the active channel on device.
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_device.c:1867 sccp_device_getActiveChannel:     -- SEP001647BC225C: No active channel on device.
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_actions.c:2374 handle_offhook:     -- SEP001647BC225C: Taken Offhook
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_channel.c:2783 sccp_channel_find_bystate_on_device:     -- SCCP: Looking for channel by state '21'
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_channel.c:2797 sccp_channel_find_bystate_on_device:     -- SEP001647BC225C: line: '302'
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_channel.c:2808 sccp_channel_find_bystate_on_device:     -- SEP001647BC225C: Could not find active channel with state RINGING(21) on device
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_actions.c:2409 handle_offhook:     -- SEP001647BC225C: Using line 302
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_channel.c:1210 sccp_channel_getEmptyChannel: SEP001647BC225C: (getEmptyChannel) on line:302, maybe_c:
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_channel.c:1215 sccp_channel_getEmptyChannel: SEP001647BC225C: (getEmptyChannel) getActiveChannel
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_device.c:1859 sccp_device_getActiveChannel:     -- SEP001647BC225C: Getting the active channel on device.
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_device.c:1867 sccp_device_getActiveChannel:     -- SEP001647BC225C: No active channel on device.
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_device.c:1883 sccp_device_setActiveChannel:     -- SEP001647BC225C: Set the active channel 7 on device
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_channel.c:195 sccp_channel_allocate:     -- 0003: New channel number: 7 on line 302
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:699 sccp_pbx_channel_allocate:     -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 302,
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:759 sccp_pbx_channel_allocate:     -- SCCP:              cid_num: "302"
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:760 sccp_pbx_channel_allocate:     -- SCCP:             cid_name: "MZ Rec"
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:761 sccp_pbx_channel_allocate:     -- SCCP:          accountcode: "501"
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:762 sccp_pbx_channel_allocate:     -- SCCP:                exten: ""
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:763 sccp_pbx_channel_allocate:     -- SCCP:              context: "from-internal"
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:764 sccp_pbx_channel_allocate:     -- SCCP:             amaflags: "0"
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:765 sccp_pbx_channel_allocate:     -- SCCP:            chan/call: "SCCP/302-00000007"
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:767 sccp_pbx_channel_allocate:     -- SCCP:combined capabilities: "[slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a, rfc2833]"
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:768 sccp_pbx_channel_allocate:     -- SCCP:  reduced preferences: "[alaw/64k, alaw/56k, ulaw/64k, ulaw/56k, g729, g729a, g729b, g729ab, g729/annex/b]"
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: ast113.c:1020 sccp_wrapper_asterisk113_allocPBXChannel: SCCP: (allocPBXChannel) Create New Channel with name: SCCP/302-00000007
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_channel.c:334 sccp_channel_recalculateReadformat:     -- SCCP/302-00000007: (recalculateReadformat)
        capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a, rfc2833]
        preferences: [alaw/64k, alaw/56k, ulaw/64k, ulaw/56k, g729, g729a, g729b, g729ab, g729/annex/b]
        remote caps: []
        READ:[alaw/64k]
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_channel.c:378 sccp_channel_recalculateWriteformat:     -- SCCP/302-00000007: (recalculateWriteformat)
        capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a, rfc2833]
        preferences: [alaw/64k, alaw/56k, ulaw/64k, ulaw/56k, g729, g729a, g729b, g729ab, g729/annex/b]
        remote caps: []
        Write:[alaw/64k]
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_pbx.c:861 sccp_pbx_channel_allocate:     -- 0003: Allocated asterisk channel SCCP/302-00000007
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_utils.c:179 sccp_pbx_setcallstate:     -- SEP001647BC225C: Set asterisk state OffHook (2) for call 7
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_indicate.c:441 __sccp_indicate:     -- SEP001647BC225C: Finish to indicate state SCCP (OFFHOOK) on call SCCP/302-00000007. New state on channel: OFFHOOK (10)
[2016-05-14 15:23:51] NOTICE[11472][C-00000001]: sccp_channel.c:1139 sccp_channel_schedule_digittimout:     -- SCCP/302-00000007: schedule digittimeout 16
[2016-05-14 15:23:52] NOTICE[11472][C-00000001]: sccp_actions.c:2911 handle_keypad_button:     -- SEP001647BC225C: SCCP (handle_keypad) digit:00000001, callid:7, buttonIndex:1 => lineInstance:1
[... cut ...]
[2016-05-14 15:28:35] NOTICE[12702][C-00000001]: sccp_channel.c:418 sccp_channel_send_callinfo:     -- SEP001647BC225C: send callInfo on SCCP/302-0000000B with lineInstance: 1
[2016-05-14 15:28:35] NOTICE[12702][C-00000001]: sccp_indicate.c:441 __sccp_indicate:     -- SEP001647BC225C: Finish to indicate state SCCP (CONNECTED) on call SCCP/302-0000000B. New state on channel: CONNECTED (30)
[2016-05-14 15:28:35] NOTICE[12702][C-00000001]: ast113.c:803 sccp_wrapper_asterisk113_indicate:   == SEP001647BC225C: (pbx_indicate) finish: send indication 'outofband signaling' (0)

dkgroot, 2016-05-14:
> Retrieving a parked call by dialling the extension number manually and waiting for the digit timeout

If you map parkinglot extensions to a single digit like '1' and there are other extensions in your dialplan which also start with '1', then the dialplan will always return 'match_more'. Because there are more/multiple matches there is no way for us to know that dialing is complete. So when an entered number to dial matches more, we have to wait for the full digit timeout period. There is no way around that. 

For instantationous dialing to succeed you need a finite dialplan, without overlap. That way once a number is complete, we get an exclusive match and know that dialling can comment. Setting up finite dialplan is possible/simple in countries which have a fixed max number length (for example 10 digits for all national extensions and 12 or more for international). And impossible in countries that do not (like germany for example).

Note: simulated enbloc dialing only kicks in after the first digit is entered, and needs a minimum of 3-4 digits to be able to calculated the inter digit interval time, if the interval time is similar for all digits we can assume that the digit's where dialed automatically by the device and we shorten the digittimeout. Normally when a person is dialing manually, they will not be able to enter all the digits with such a specific interdigit cadence and we can detect that, resulting in the normal timeout.

dkgroot, 2016-05-14:
I advise against using parkinglot numbers with 1 digit, that doesn't match a normal use case. Can you take the default asterisk example of 
700 park + 701-709 slots
And give this another try.
BTW: currently redesigning because of #300, so the whole digittimout issue should be fixed when that one is done.

jamesmz, 2016-05-14:
Ah, I see. Thinking back I believe I originally had 700 etc, but changed it to a single digit when testing something (so I didn't have to type the 3 digits over and over again), but never changed it back. I'll swap back to 700.

dkgroot, 2016-05-14:
Please retest the latest revision: 0929147

The new menu solution should fix both #300 and #301. chan-sccp now dials the parkingslot directly, instead of waiting for digits comming from the device. It also handles the 'Exit' event directly.

jamesmz, 2016-05-14:
Seems OK in this release, thank you. :+1: I'll report back if I come across anything.

dkgroot, 2016-05-14:
Ok, nice !
Closing this issue (You can re-open, report a new issue or add a comment on #139).

commit 09291470 (2016-05-14): Enh: Use IPPhoneMenu instead of IPPhoneDirectory for VisualParkingLot handling

Fix: Visual parking lot menu comes back on screen after exiting and un-parking elsewhere
fixes #300

Fix: '#' not dialled to instantly un-park when retrieving single parked call
New version does not depend on the device to do the dialing, we dial the parkinglot extension for the device instead.
fixes #301
closed 2016-05-14 by dkgroot