chan-sccp/chan-sccp #302: After parking the call, the display bar text always show extension 700

drtmk, 2016-05-15:
When I park a call, the notification message displayed on the phone always shows extension 700, although asterisk plays 701, 702 and so on, so the text on the phone is incorrect.

Calling from extension 103 (software SIP phone) to extension 102 (7942 firmware 9-3-1SR4-1S).
Debug: http://pastebin.com/uBb1tW8S

res_parking.conf:
```
[general]
[default]
parkext => 700                  ; What extension to dial to park.
parkpos => 701-720              ; What range of parking spaces to use - must be numeric
context => parkedcalls          ; Which context parked calls and the default park
```

Asterisk 13.8.2 built by root @ asterisk on a i686 running Linux
Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - ae4d27fM

dkgroot, 2016-05-17:
Hi Thomas,
I looked into this. I can confirm that we are displaying 700 instead of the slot where the call got parked. I checked the asterisk function used to park the call. And it is supposed to return the parkingslot or parking_space the call got parked at, according to the documentation. This is however not the case. I tired to solve the issue some other way, but access parking_space information is not available as far as i can see. 

I will have to open an issue in asterisk jira to get the issue resolved. We will need to wait for them to resolve this.

dkgroot, 2016-05-17:
Response from Digium: "The *only* caller of ast_parking_park_call() is chan_skinny.c.  Unfortunately in v12+, the parking space is no longer known when the function returns because other threads are now involved.  Off hand, I don't see a way to get that information synchronously."
I am not sure how to proceed, should we suppress the status bar output when running asterisk-13 ?

drtmk, 2016-05-17:
Can this information be acquired asynchronously and then presented on the status bar?

dkgroot, 2016-05-18:
Async is pretty tricky. we don't know how long it will take before the information is available (0-10-20-30-xx milli seconds). Nor do we know how long the information is going to be available. It could be unparked immediately. So if we don't wait long enough we miss the information, if we wait to long we miss the information. It should not be a guessing game. Both of these make it dangerous (segfault potential).

We could just monitor the AMI event stream and grab the information from there, that is possible (and we already do that for the visual parkinglot functionality). But it will change the current flow of events significantly (only for asterisk-13). I will have to think about this for a bit. 

I had hoped to persuade the digium guys to fix the issue. Notice: chan_skinny also has the same issue and the function implementation contradicts the documentation. I am afraid they are going to fix the documentation though.

Maybe our first step should just be to drop the statusbar update when using ast13. You can still hear the parking_space information.

dkgroot, 2016-05-20:
Received an update from Richard Mudgett via https://issues.asterisk.org/jira/browse/ASTERISK-26029

The workaround has several potential pitfalls and I will not implement this for this issue. I will however remove the statusbar message when running in > asterisk-12.

closed 2016-05-20 by dkgroot