chan-sccp/chan-sccp #304: Fax sending/receiving problem in Cisco  VG248

azaim, 2016-05-16:
Dear All,

I am having problem with sending/receiving fax from Cisco VG248 with chan_sccp module.
In my setup I have asterisk box and  call manager for testing purposes. They are connected each other as sip trunk.

In my test I  tried to send fax from asterisk (VG248 analog phone) line to the another analog line which exist in callmanager.

In my test, when I try to send fax, I am getting "[2016-05-16 09:35:25] WARNING[29170][C-00005c74]: res_rtp_asterisk.c:4152 ast_rtp_read: RTP Read too short" errors and its fails to send the fax.

and when I tried to receive  fax I am getting "NOTICE[29170][C-00005c74]: res_rtp_asterisk.c:4371 ast_rtp_read: Unknown RTP codec 96 received from '192.168.152.142:31708'" and NOTICE[29170][C-00005c74]: res_rtp_asterisk.c:4371 ast_rtp_read: Unknown RTP codec 96 received from '192.168.152.142:31708'" errors.

I am not sure %100 if its about chan_sccp problem.

You can find my full logs below.

**chan-sccp version:** 
Revision: 6633
Node Kind: directory
Schedule: normal
Last Changed Author: ddegroot
Last Changed Rev: 6630
Last Changed Date: 2016-02-16 01:29:14 +0200 (Sal, 16 Şub 2016)

**asterisk version:Asterisk 11.21.2

* Log Files (attach .txt / [pastbin](http://pastebin.com/iurZetH2) 

BR,

Ahmet

dkgroot, 2016-05-20:
Hi Ahmet,

At the moment we don't have any special provisions to support faxing in chan-sccp. We have global settings for jitterbuffer and echo cancellation, which would have to change (at least). We also don't do any fax detection and forcing codecs to a specific format (like ulaw).

At the moment there is not a lot we can / will do to make any changes to support faxing. because the use case is so very very small.

From your mails i notices that you are converting the incoming connectiong several times (ISDN->analog->SIP->SCCP->analog), the change of any fax/modem working after that are extremely small.

dkgroot, 2016-05-20:
If you want a stable / controlable solution:

* forget using the vg248 in sccp mode for faxing. It's never going to be a stable and reliable method (i think), to many conversions etc.
* Get 15 analog lines, and hook up the faxes directy to the outside. This is a proven method and it works (100%).
* Buy a couple of small SIP FXO converters and install them near each fax device. Only do this if this fax machine is crucial. This will work 97% of the time, but will require a little fiddling with timing and getting sip, codecs, jitter, echo cancellation and modem speed to work correctly.
* Get to the year 2000:
  * Convert incoming faxes on the asterisk machine and send a PDF to the recipient per mail (Your users will love you). (asterisk + iaxmodem + hylafax). This works 95% of the time and you can get it up to 99% with some fiddling.
  * Install a Software FaxPrinter driver on some of the users machines, who use faxing a lot. This faxprinter driver can send the PDF directly to the Asterisk machine for conversion (http://hylafax.sourceforge.net/howto/faxing.php#ss5.2). Advantages:
    * No more printing to paper ->  Save a couple of trees a year.
    * No more walking walking to the fax machine
    * No more manually entering the number
    * No more pushing the paper into it and it getting stuck...
    * No more unsent faxes nobody noticed
* Start using PGP/GPG 'Signed' e-mails instead of sending and receiving faxes, we are in the year 2016 already.

   Note: arguments like: 'but e-mail is not a legal document' are complete bullshit, because faxes are not an unforgable document either.

   Note: arguments like: 'but it requires a signature', that's why you are PGP/GPG signing the e-mail. It's traceble and unforgeable.

Sorry, but my attitude toward using faxes and the arguments people use to keep those ancient machine around are annoying the hell out of me.

azaim, 2016-05-20:
Thank you Diederik, I am so glad to hear your opinion to sending fax with sccp, which is really important for me.

closed 2016-05-23 by azaim