chan-sccp/chan-sccp #305: VG248 register problem - Session with this IP-address is already known

azaim, 2016-05-16:
Hello All,

Today I tried to update my chan-sccp rev 6633 to 6836.

I think in new version its limiting to an 1 ip per device.

In VG248 and VG224 it is not possible because there is 48 lines in one ip.
Example:
VGC13c496217a01
VGC13c496217a02
VGC13c496217a03
.
.
.
VGC13c496217a48

In normal situation all lines registered like below.

 -- SCCP: Connected on server via 192.168.12.135:45483
    -- VGC13c496217a01: nat=Off set manually in config file
       > VGC13c496217a01: Asked for our protocol capability (22).
       > VGC13c496217a01: Phone protocol capability : 0
       > VGC13c496217a01: Our protocol capability   : 22
    -- VGC13c496217a01: Joint protocol capability : 3
    -- VGC13c496217a01: Ask the phone to send keepalive message every 112 seconds
    -- Added line '5510' to Glob(lines)
    -- VGC13c496217a01: Device has 4 Capabilities
    -- VGC13c496217a01: using softkeyset: default (0x31b3550)!
    -- VGC13c496217a01: Device registered; performing post registration tasks...
    -- VGC13c496217a01: (sccp_devstate_deviceRegisterListener) device registered
    -- SCCP: Accepted Client Connection from 192.168.12.142:16387
    -- SCCP: Connected on server via 192.168.12.135:41698
    -- VGC13c496217a02: nat=Off set manually in config file
       > VGC13c496217a02: Asked for our protocol capability (22).
       > VGC13c496217a02: Phone protocol capability : 0
       > VGC13c496217a02: Our protocol capability   : 22
    -- VGC13c496217a02: Joint protocol capability : 3
    -- VGC13c496217a02: Ask the phone to send keepalive message every 119 seconds
    -- Added line '5001' to Glob(lines)
    -- VGC13c496217a02: Device has 4 Capabilities
    -- VGC13c496217a02: using softkeyset: default (0x31b3550)!
    -- VGC13c496217a02: Device registered; performing post registration tasks...
    -- VGC13c496217a02: (sccp_devstate_deviceRegisterListener) device registered
    -- SCCP: Accepted Client Connection from 192.168.12.142:16388

But in new revision chan-sccp not allowing other lines register and giving an error below.

   -- SCCP: Session with this IP-address is already known 192.168.12.142:16626. wait !
    -- VGC13c496217a47: nat=Off set manually in config file
       > VGC13c496217a47: Asked for our protocol capability (22).
       > VGC13c496217a47: Phone protocol capability : 10
       > VGC13c496217a47: Our protocol capability   : 22
    -- VGC13c496217a47: Joint protocol capability : 10
    -- VGC13c496217a47: Ask the phone to send keepalive message every 105 seconds
    -- Added line '5046' to Glob(lines)
    -- VGC13c496217a47: Device has 4 Capabilities
    -- VGC13c496217a47: using softkeyset: default (0x35f8250)!
    -- VGC13c496217a47: Device registered; performing post registration tasks...
    -- VGC13c496217a47: (sccp_devstate_deviceRegisterListener) device registered
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16626. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16626. wait !
-- SCCP: Session with this IP-address is already known 192.168.12.142:16626. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16626. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16626. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16626. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16626. exceeding wait time, denied connection !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16627. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16627. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16627. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16627. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16627. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16627. exceeding wait time, denied connection !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16628. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16628. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16628. wait !
    -- SCCP: Session with this IP-address is already known 192.168.12.142:16628. wait !

BR,

Ahmet

dkgroot, 2016-05-17:
> Today I tried to update my chan-sccp rev 6633 to 6836.

Where are you getting this 6836 revision from ? This looks like a svn revision, but github revisions don't look like that.

To update you need to clone the github repository: 
git clone https://github.com/chan-sccp/chan-sccp chan-sccp_develop
cd chan-sccp_develop
git checkout develop
./configure [....]
....

The issue which prevents multiple devices using a single ip-address was a regression and is already fixed.

azaim, 2016-05-17:
Thank you diederik.
I was updated via svn.

dkgroot, 2016-05-17:
Ok :-)

closed 2016-05-17 by dkgroot