chan-sccp/chan-sccp #306: [Regression] Missing AMI event DeviceStateChange on RINGING

pnlarsson, 2016-05-17:
This event is in 4.2 but not in master och develop branch:

Event: DeviceStateChange
Privilege: call,all
SystemName: s50
Device: SCCP/519
State: RINGING

**asterisk version:** 13.9

./configure --enable-debug --disable-park --disable-dirtrfr --disable-pickup --with-astmoddir=/usr/local/ngcore/service/pbx/usr/lib/asterisk/modules --enable-experiment-mode --no-create --no-recursion

No change if configured with or without --enable-experiment-mode

dkgroot, 2016-05-19:
Is this directly related to issue #307 ?

pnlarsson, 2016-05-20:
Kind of, we used the DeviceStateChange as a band aid fix to get the line state ringing. So if #307 works this is not needed (for us).

dkgroot, 2016-05-20:
Even better ! Thanks for the update ! Closing this issue as well !

closed 2016-05-20 by dkgroot