chan-sccp/chan-sccp #307: PBX state RINGING is missing

pnlarsson, 2016-05-17:
The PBX does not know that the sccp device/line is ringing:
```
   -- Called SCCP/519
    -- SCCP/519-00000001 is ringing
 uc50*CLI> sccp show channels 

 +--- Channels ----------------------------------------------------------------------------------------------------------------------------------------------+
 | ID    Name                      LineName   DeviceName       NumCalled  PBX State  SCCP State ReadCodec  WriteCodec RTPPeer                Direct DTMFmode |
 + ===== ========================= ========== ================ ========== ========== ========== ========== ========== ====================== ====== ======== +
 | 1     SCCP/519-00000001         519        SCCP                        Down       RINGING    alaw/64k   alaw/64k                          no     RFC2833  | 
 +-----------------------------------------------------------------------------------------------------------------------------------------------------------+
```

And it wont send the ExtensionStatus Rining AMI event

This has been the same for as long as I used chan_sccp.

A fix (not recomended) is to uncomment the 
```
//                      iPbx.set_callstate(c, AST_STATE_RINGING);                                               /*!\todo thats not the right place to update pbx state */
```

line in src/sccp_indicate.c

**asterisk version:** 13.9

dkgroot, 2016-05-19:
Can you check again using this revision, to see if it fixes the issue on your system as well ?

pnlarsson, 2016-05-20:
Tested latest develop bransh and it's now working as it should. Thanks

dkgroot, 2016-05-20:
Nice, We finally fixed this one, thanks for the update. 

Closing this issue !

dkgroot, 2016-05-26:
Reverting related changes back to af666b3 for now. Reopening issue

marcelloceschia, 2016-08-26:
I've check this @ develop branch and it looks like this issue is fixed.
@pnlarsson can you confirm this?

pnlarsson, 2016-08-29:
Yepp, using revert_session_handler I get:

Event: DeviceStateChange
Privilege: call,all
SystemName: s1
Device: SCCP/519
State: RINGING

So I will close this issue.

closed 2016-08-29 by pnlarsson

marcelloceschia, 2016-08-29:
Is it possible to check this for develop branch too?
Otherwise I have to merge this changes back to develop branch.

pnlarsson, 2016-08-29:
With develop branch: 

Event: ExtensionStatus
Privilege: call,all
SystemName: s1
Exten: 519
Context: company_4_hints
Hint: SCCP/519
Status: 8
StatusText: Ringing

Event: DeviceStateChange
Privilege: call,all
SystemName: s1
Device: SCCP/519
State: RINGING

So I'm happy :-)

marcelloceschia, 2016-08-29:
Thanks you for this positive feedback :)