chan-sccp/chan-sccp #308: Message "Unsupported payload type received" appears continuously while the call is established

miguelsanzpardo, 2016-05-19:
[If you have debug level activated, once a sccp phone has established a call with another sccp phone (or with a sip phone) this line appears continously until the call is finished:
DEBUG[26115][C-00000001] res_rtp_asterisk.c: Unsupported payload type received ]

**Expected behavior:**
[It shouldnt write this line continously during the call]

**Observed behavior:**
[It writtes continously this line during a call. It is possible that chan_sccp is using incorrect payload with respect to asterisk payload]

**chan-sccp version:** 4.3.0.b768c55

**asterisk version:** 11.21.2 & 13.8.2

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: [No, on 4.2.0-6168 it appears too, at least on my case]
* Problem can be reliably reproduced, doesn't happen randomly: [Yes]
* Platform:
    * [VMWARE:  AMD Opteron(TM) Processor 6238 - 2600 Mhz]
    * [64 Bit]
    * [Little Endian]
    * [Debian 8.2]

dkgroot, 2016-05-27:
Note: When using a dialplan like:
```
exten => 666,1,NoOp(TEST)
same => n,Progress();
same => n,Wait(1);
same => n,Ringing();
same => n,Wait(1);
same => n,Progress();
same => n,Wait(1);
same => n,Answer();
same => n,Wait(1);
same => n,MusicOnHold(default);
same => n,Hangup();
```
It shows this in the debug log switching from ALAW to SLIN and back to ALAW again. 
```
[2016-05-27 09:53:07] VERBOSE[18374][C-00000004] sccp_indicate.c:     -- SEP0026CBC03B37: Finish to indicate channel state:RINGOUT on call:SCCP/1101-00000001, lineInstance:1. New channel state:RINGOUT
[2016-05-27 09:53:07] VERBOSE[18374][C-00000004] sccp_utils.c:     -- SEP0026CBC03B37: Set asterisk state Ring (4) for call 1
[2016-05-27 09:53:07] VERBOSE[18374][C-00000004] ast111.c:   == SEP0026CBC03B37: (pbx_indicate) finish: send indication 'inband signaling' (-1)
[2016-05-27 09:53:07] DEBUG[18374][C-00000004] channel.c: Driver for channel 'SCCP/1101-00000001' does not support indication 3, emulating it
[2016-05-27 09:53:07] DEBUG[18374][C-00000004] channel.c: Set channel SCCP/1101-00000001 to write format slin
[2016-05-27 09:53:07] DEBUG[18374][C-00000004] channel.c: Scheduling timer at (50 requested / 50 actual) timer ticks per second
[2016-05-27 09:53:07] DEBUG[18374][C-00000004] channel.c: Prodding channel 'SCCP/1101-00000001'
[2016-05-27 09:53:07] VERBOSE[18374][C-00000004] ast111.c:     -- SEP0026CBC03B37: Asterisk prodded channel SCCP/1101-00000001.
[2016-05-27 09:53:07] DEBUG[18374][C-00000004] res_rtp_asterisk.c: Received frame with no data for RTP instance '0x7fd5b001f1c8' so dropping frame
[2016-05-27 09:53:07] DEBUG[18374][C-00000004] pbx.c: Launching 'Wait'
[2016-05-27 09:53:07] VERBOSE[18374][C-00000004] pbx.c:     -- Executing [666@from-internal-moviles:5] Wait("SCCP/1101-00000001", "1") in new stack
[2016-05-27 09:53:07] DEBUG[18374][C-00000004] res_rtp_asterisk.c: Ooh, format changed from unknown to alaw
[2016-05-27 09:53:07] DEBUG[18374][C-00000004] res_rtp_asterisk.c: Created smoother: format: alaw ms: 20 len: 160
[2016-05-27 09:53:08] DEBUG[18374][C-00000004] pbx.c: Launching 'Progress'
[2016-05-27 09:53:08] VERBOSE[18374][C-00000004] pbx.c:     -- Executing [666@from-internal-moviles:6] Progress("SCCP/1101-00000001", "") in new stack
[2016-05-27 09:53:08] VERBOSE[18374][C-00000004] ast111.c:     -- SEP0026CBC03B37: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS' (14) condition on channel SCCP/1101-00000001 (readStat:2, receiveChannelState:2, rtp:yes, default res:inband signaling (-1))
[2016-05-27 09:53:08] VERBOSE[18374][C-00000004] sccp_indicate.c:     -- SEP0026CBC03B37: Indicate SCCP new state:PROGRESS, current channel state:RINGOUT on call:SCCP/1101-00000001, lineInstance:1 (previous channelstate:PROGRESS)
[2016-05-27 09:53:08] VERBOSE[18374][C-00000004] sccp_indicate.c:   == SEP0026CBC03B37: SCCP_CHANNELSTATE_PROGRESS
[2016-05-27 09:53:08] VERBOSE[18374][C-00000004] sccp_indicate.c:     -- SCCP: Asterisk requests to change state to (Progress) from (RINGOUT)
```

miguelsanzpardo, 2016-11-22:
Could be that SCCP doesnt support RTCP packets?

I share with you a full log related with a call realised from one SCCP phone to a SIP phone with:
- core verbose level : 5
- core debug: 5
- sip debug on
- sccp debug all

[full.txt](https://github.com/chan-sccp/chan-sccp/files/605925/full.txt)

dkgroot, 2016-11-22:
We do support RTCP. You can see the RTCP stream being setup in your full log, just before the payload issue starts. I can remember trying to find the cause for this issue back in may, i don't remember the details though. Could the revert_branch have something to do with trying to find this issue, back then ?

miguelsanzpardo, 2016-11-22:
I tried with:
- feature/revert_session_handler
- feature/revert_to_af666b3b
but the problem persist.

You need to have debug activated to see this problem. I remember that I shared with you an ISO in which the problem can be reproduced (in theory)

The full.txt that I have sent has been obtaining from an Asterisk Vanilla (the same Asterisk Vanilla that I have used with the SIP NOTIFY testing)

miguelsanzpardo, 2016-11-22:
I share with you 2 .txt related with:
- 1 Call from SIP to SIP  (1004 to 1001)
- 1 Call from SCCP to SIP (1104 to 1004)

On both cases I have executed:
- "core show channel XXX" on both legs
Before and after the call has been established

[SCCP_to_SIP.txt](https://github.com/chan-sccp/chan-sccp/files/606247/SCCP_to_SIP.txt)
[SIP_to_SIP.txt](https://github.com/chan-sccp/chan-sccp/files/606246/SIP_to_SIP.txt)

dkgroot, 2016-11-22:
FYI: Full sccp debug level is not required in this instance... It generates way to much information. sccp debug core, channel, rtp should be enough. It might be helpfull to include a little bit of 'rtp debug on' so we can verify which payload is faulty in this case.

I don't know where to start with this issue. I remember looking at the issue for an extended period of time back then and could not reproduce the issue on my system. 

@marcelloceschia Do you see anything in the debug logging, that gives you an indication of where the problem might be coming from ?

miguelsanzpardo, 2016-11-22:
I share with you a debug of a call from 1104(SCCP) to 1001(SIP):

- core verbose level : 5
- core debug: 5
- rtp debug on
- sccp debug core, channel, rtp

[full_RTP.txt](https://github.com/chan-sccp/chan-sccp/files/606349/full_RTP.txt)

marcelloceschia, 2016-11-22:
@miguelsanzpardo interessting log, but it does not include the payload type that causes this log entry. maybe we have to add some logs to asterisk or need a tcpdump

dkgroot, 2016-11-22:
@marcelloceschia We would have to add some logging to res_rtp_asterisk.c: bridge_p2p_rtp_write, to find out why the bridge instance payload codecs are not matching.

marcelloceschia, 2016-11-22:
@dkgroot i know ;)

marcelloceschia, 2016-11-22:
Untested, change:
```
if (ast_rtp_codecs_find_payload_code(ast_rtp_instance_get_codecs(instance1), bridged_payload) == -1) {
		ast_debug(1, "Unsupported payload type received \n");
		return -1;
	}
```
to:
```
if (ast_rtp_codecs_find_payload_code(ast_rtp_instance_get_codecs(instance1), bridged_payload) == -1) {
	ast_rtp_instance_get_remote_address(instance1, &remote_address);
	res = rtp_sendto(instance1, (void *)rtpheader, len, 0, &remote_address, &ice);
	ast_debug(1, "Unsupported payload type received. Packet to %s%s (type %-2.2d, len %-6.6d)\n",
			    ast_sockaddr_stringify(&remote_address),
			    ice ? " (via ICE)" : "",
			    bridged_payload, len - hdrlen);
	return -1;
}
```

dkgroot, 2016-11-22:
Found the cause. We are missing the initial setup of the rtp instance. 

It got zeroed out while trying to copy payloads into the rtp instance using `ast_rtp_codecs_payloads_set_rtpmap_type_rate()` ->  Remarked out this copy function.

Adding: `ast_rtp_codecs_payloads_default(ast_rtp_instance_get_codecs(instance), instance);`
To make sure the basic set of asterisk codec rtp payload are available.

Reinstated the previous method of adding payload mappings for cisco specific types like DTMF.

Guess: This change  is related to the switch to dynamic payloads in asterisk-10.
Note: `ast_rtp_codecs_payloads_default(ast_rtp_instance_get_codecs(instance), instance)` does not exist for asterisk-13.
Note: update required in #342 when pull request is merged

dkgroot, 2016-11-24:
Issue is still present in asterisk-13 and up. Either we are adding an unknown codec to the payloads or are not declaring it correctly. Need to add some debug output to asterisk-13 to figure out which one of the two it is. First need to sort out how to emulate this issue reliable on my local test system.

It might turn out that my fix for asterisk-11 works but is/was incorrect, we will see.

miguelsanzpardo, 2016-12-12:
Hi Diederik, how about the issue, did you can to analyse the problem with asterisk 13? By the way, with actual develop version the problem doesnt appears on asterisk 11 but you are not sure if the patch is 100% correct, true?

dkgroot, 2016-12-12:
I can't find a way to do the same thing in asterisk 13 and up. Which leads me to think that the solution in asterisk-12  and below is not 100% correct either. Its working for you in asterisk-11 so that's ok. No body else seems to notice the issue at all. For now i am not to concerned about it, but the issue is still open. and i am thinking about it.

dkgroot, 2017-03-13:
@miguelsanzpardo 
Created a temporary branch called 'issues/308' for you to run a couple of tests (Asterisk-13 only).

dkgroot, 2017-03-14:
Update the issues/308 branch. Trying to be a little more carefull when setting up RTP and reading/writing to the RTP channel. If you allready started testing this branch, then please update

dkgroot, 2018-12-23:
This issue seems to also have been resolved by merging #420

commit 838dfecc (2016-11-22): Fix: 'Unsupported payload type received' continuesly

It got zeroed out while trying to copy payloads into the rtp instance using ast_rtp_codecs_payloads_set_rtpmap_type_rate() -> Remarked out this copy function.

Adding: ast_rtp_codecs_payloads_default(ast_rtp_instance_get_codecs(instance), instance);
To make sure the basic set of asterisk codec rtp payload are available.

Reinstated the previous method of adding payload mappings for cisco specific types like DTMF.

Guess: This change is related to the switch to dynamic payloads in asterisk-10.
Note: ast_rtp_codecs_payloads_default(ast_rtp_in [... cut]
closed 2018-12-23 by dkgroot