chan-sccp/chan-sccp #309: VG 248 some lines not registering

azaim, 2016-05-23:
I started to test latest github_develop version of chan-sccp on VG248.
I am writing down the problem I am facing when registering devices.

I realized that some lines of the VG248 not registering and its stucking on RegState = Progress.
Sometimes 2, sometimes 3 lines are stuck in progress state.
you can see the:

sccp show devices: http://pastebin.com/uECbN9h6
full debug log of unregisted lines: http://pastebin.com/7y8gu5Ze

Same problem also happens when I change line of the device .  It keeps also unregistered and RegState = Progress.
full debug log of unregisted changed line: 
http://pastebin.com/AZdbLBx5

**chan-sccp version:** Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - 8c38d34M (built by 'root' on '2016-05-23 13:08:25 UTC')

**asterisk version:** Asterisk 11.21.2 built by root @ jenkins-builder1.schmoozecom.net on a x86_64 running Linux on 2016-02-11 22:14:18 UTC

**Addition / Optional Information:** There was no problem with rev:6630
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes]
* Problem can be reliably reproduced, doesn't happen randomly: [No]
* Direct Contact information: [You can contact me here]

dkgroot, 2016-05-26:
Researching solution using the 'feature/revert_session_handler' branch

dkgroot, 2016-12-14:
@azaim 
The pastebin records are not available any more, so cannot look into them.

There have been a couple of commits to sccp_session.c which might have fixed or influenced this behaviour: 9e00eb , 3aabf4a, 46decc, 11eb22

Could you re-test if the behaviour still exists with the current develop branch.

Note: There is still an open issue with devices occasionally getting into a loop between Registration State: None and Progress over and over again, but that is handled in a separate issue. That does not seem to be directly related to this report, where many devices seem to get into such a state.

If the issue does still occur, please attach the log files directly to the issue (which is allowed now). And do set 'sccp debug core, action, device, socket' while testing.

dkgroot, 2016-12-20:
While working on issue #358 i found the cause for this issue as well. The solution has already been comitted in revision ebe7837. Will need a little bit of testing, before it can be moved to mainline.

dkgroot, 2016-12-28:
@azaim 
Hi Ahmet,
I just merged fixes regarding #358 back into the 'develop' branch. Can you check if this also fixes the issues regarding VG248/V224 not being able to register a device/line

dkgroot, 2017-02-25:
Closing issue report, because of no further response

closed 2017-02-25 by dkgroot