chan-sccp/chan-sccp #310: Phones are restarted when you add one extension and execute sccp reload

miguelsanzpardo, 2016-05-25:
When you add an extension and execute CLI>sccp reload all the sccp phones which arent on a conversation are restarted. 
The phones which were on a conversation are restarted when the conversation has finished and have passed 15-30 seconds.

**Reproduction Steps:**

1. [Edit sccp_conf (or sccp_hardware.conf + sccp_extensions.conf if you are using FreePBX) and add a new extension]
2. [Execute CLI > sccp reload]

**Expected behavior:**

[A new extension has been added]

**Observed behavior:**

[A new extension has been added, but the rest of the sccp extensions have been reloaded]

**Screenshots**

**chan-sccp version:** from 4.2.0 4.2.0 - 6168M to 4.3.0-fba2d

**asterisk version:** 11.21.2

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: [No]
* Problem can be reliably reproduced, doesn't happen randomly: [Yes]
* Platform:
    * [64 Bit]
    * [Little Endian]
    * [Debian 8.2]

dkgroot, 2016-05-25:
>       o [64 Bit]
>       o [Little Endian]
>       o [Debian 8.2]

Hi Miguel,

Check the debug logs to see why sccp_config comes to the conclusion that devices need to be reset. Those lines look like:
[2016-05-25 10:30:45] NOTICE[11505] sccp_config.c:   == SCCP: config parameter mailbox='' in line 810 changed. (causes device reset)

Do you actually have a line
mailbox = ''
In your sccp.conf file. If you do not want to define / set something, you should not provide and empty string, you should leave the entry out completely.

Diederik

miguelsanzpardo, 2016-05-25:
Umh, in theory I dont have any mailbox line on my sccp.conf file.

dkgroot, 2016-05-25:
Should be Fixed now

closed 2016-05-25 by dkgroot

miguelsanzpardo, 2016-05-26:
I have made several test and it looks like to be fine now, thanks Diederik :)

dkgroot, 2016-05-26:
Thanks for the update !

> NOTICE[9208] sccp_config.c: -- SCCP: parsing general parameter: privacy = 'yes' (line: 55)
WARNING[9208] sccp_config.c: SCCP: Unknown param at general:55:privacy='yes'

'privacy' was renamed to 'private' some time ago. 'privacy' has been deprecated for more than a year and was removed more than 3-4 months ago.

Conference related parameters like: conf_allow, conf_play_general_announce, conf_play_part_announce, :conf_mute_on_entry, conf_music_on_hold_class and conf_show_conflist require that you ./configure --enable-conference, for them to be enabled and available. Without enabling conference these parameter are not known to the parser and will result in an error message.

I am not too worried about this netsock error message at the end, when a device is being restarted. It is somewhat expected. Maybe it should not be declared an ERROR but more a WARNING / NOTICE in this case.

miguelsanzpardo, 2016-05-26:
Nice to know that all of that has sense, perfect :+1:

commit 3cc74f5d (2016-05-25): Fix: All phones are restarted when you add a new extension to sccp.conf and execute sccp reload

Issue reported by Miguel

Fixes #310
commit 37b66390 (2016-05-25): Fix: All phones are restarted when you add a new extension to sccp.conf and execute sccp reload

Issue reported by Miguel

Fixes #310
commit 60e576b7 (2016-05-25): Fix: All phones are restarted when you add a new extension to sccp.conf and execute sccp reload

Issue reported by Miguel

Fixes #310
commit dbe1f15e (2016-05-26): Fix: All phones are restarted when you add a new extension to sccp.conf and execute sccp reload

Issue reported by Miguel

Fixes #310