chan-sccp/chan-sccp #311: Outgoing calls show "To Unknown Number"

dandrzejewski, 2016-05-25:
When I dial a number, the phone shows "To Unknown Number" on the screen.

**Reproduction Steps:**

# Dial a number

**Expected behavior:**

The phone should show either a name (if possible) or the phone number that was dialed.

**Observed behavior:**

Phone shows "To Unknown Number"

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)
[url]

**chan-sccp version:** (include revision if relevant)

Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - b725c2cM (built by 'root' on '2016-05-19 22:11:43 UTC')

**asterisk version:** (include revision if relevant)

13.7.4

miguelsanzpardo, 2016-05-26:
With the last revision (4.3.0 develop - 3cc74f5M ) this problem doesnt appear, maybe from this rev. to the last rev. there are several changes that are affecting to this behaviour.

dkgroot, 2016-05-26:
In an attempt to fix the 9.4 firmware / Placed calls issue reported by drtmk, i mayh have broken more than i should have. I am gonna move the 'develop' branch back to the point where i started on that issue. And slowly move forward from there. Instead of trying the multitude of new issues reported since then.

miguelsanzpardo, 2016-05-26:
Using af666b3:

- Dialed numbers is working fine with 7942/7962 9.4.2 firm.
- Placed calls doesnt show name info, but the rest of the directory phonebook looks like to be fine.
- Busy indications between phones which are registered on the same IP-PBX are ok
- We have another problems like: fba2d24db and 3cc74f5d

BTW, with this rev. when I have made a capture (on 2 times with the phone offhook) I have seen that the phone which was called and was captured by other was restarted. Im taking the log.

dkgroot, 2016-05-26:
Hi miguel,
So everything is back to the status for two weeks. I will use this revision as a starting point. I will recommit fba2d24 and 3cc74f5 on top of that, don't worry. And the session_handler issue is currently handled in a different branch as well. 
I just want to get back to a 'last known good situation' for the indication issues.

miguelsanzpardo, 2016-05-26:
I know Diederik, but If I can help you showing more information I will share with you ;)

dkgroot, 2016-05-26:
Thanks.. Working on it

dkgroot, 2016-05-26:
Hi Miguel / David:
Can you checkout 'feature/revert_to_af666b3b' and check if everything is still working as expected. 
I started over at af666b3b and committed all the changes up to 8c38d34, which did not change callerid status and ringing indication.
I am looking to commit a0814c6, but i am not sure if i should. So some testing before doing that one might be in order.

miguelsanzpardo, 2016-05-26:
All looks like to be fine, but when I made a capture on 2 times (with the phone offhook) I have seen that the phone which was called and was captured by other was restarted (I know that is related with session_handler issue but I would want to report you again that making the same thing that before the phone has been restarted).

dkgroot, 2016-05-26:
Work Log
```
commit 3cc74f5d3c2d03dd60d24181d43bc13e0bfebdc2
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Wed May 25 20:35:46 2016 +0200

    Fix: All phones are restarted when you add a new extension to sccp.conf and execute sccp reload
    
    Issue reported by Miguel
    
    Fixes #310

APPLIED 

commit 8464a3394df99d54691035fd380c5020de5009f1
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Wed May 25 20:34:28 2016 +0200

    Fix: some sccp_log(...) line ends missing ';'

APPLIED 
 
commit fba2d24db38366d412e9dcb57eb844d3af1925dc
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Mon May 23 15:23:05 2016 +0200

    Enh: Renamed .writeState to .receiveChannelState
    Enh: Renamed .readState to .mediaTransmissionState
    
    To Prevent an issue like a0814c691 in the future.

APPLIED

commit ef7f9957ed22e01f165f64915be4f58133ff7e81
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Mon May 23 15:11:52 2016 +0200

    Fix Session Timeout issue
    - Increase Keepalive additional timeout
    - Added Double Keepalive alive time when on a call
    - Reset keepalive before and after processing a message

SKIPPED 

commit dd6cd7bd99ef46d23b77058a77f98e75a06f218f
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Mon May 23 15:11:26 2016 +0200

    Enh: Make sure all RTP ports get/are closed during channel destruction and transfer_cancel

SKIPPED

commit a0814c691bf48ce74474e99734c962394a5de7b2
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Mon May 23 15:04:55 2016 +0200

    Fix: reversal of RTP readState and writeState
    
    readState is related to the MediaTransmission Status
    writeStae is related to the OpenRecieveChannel Status
    
    Changeing the names, might make this a little less confusing.

APPLIED

commit 8c38d344504a1811785df2a331031651f8a919eb
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Mon May 23 12:10:52 2016 +0200

    Fix: prevent segfault in sccp_handle_time_date_req when we have lost the session and we are still processing a hangup event. Reported by Niklas.
[... 227 lines cut ...]
     - We should send the protocol level we are actually using, instead of forcing protocolversion=17. Otherwise it causes issues in wireshark with alignment.
     - Reorder normal_dialing to follow CM order.
    Fix: Make sure we only start mediatransmission once. CM updates callinfo after starting media transmission.
    Enh: cleanup sccp_protocol.c
    Enh: sccp_indicate: Simplify getting innstance. Renamed instance to lineInstance.

SKIPPED 

commit e8a40a0e6d850e8bd46ecec2c72e17cb2d020dcf
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Wed May 18 15:40:51 2016 +0200

    Fix: 7936 not able to do inband DTMF. Issue reported by Mark Cianfaglione.

APPLIED

commit af666b3b13886b4665623d59720b974ea7114d40
Author: Diederik de Groot <dkgroot@talon.nl>
Date:   Wed May 18 11:36:58 2016 +0200

    Fix: clang compile warning

STARTING POINT

```

dkgroot, 2016-05-26:
Everything uptill the head has been commit except for:

- ef7f9957
- dd6cd7b
- fe4a0fcd
- da3bc71
- e9b5856
- e50e28e
- 18d0515
- ac76eb9
- 315d54f21
- e50e28e3f
- 3fb6042
- 01dbea1
- 1dfc06e9
- e78251a
- 6ce88d2

Which were either skipped or partially applied

dkgroot, 2016-05-26:
Issue related to Unknown Number above is fixed using feature/revert_to_af666b3b

closed 2016-05-26 by dkgroot