chan-sccp/chan-sccp #312: Placed calls phonebook directory shows incorrect information

miguelsanzpardo, 2016-05-26:
[It is showed the number correctly but the name incorrectly. It is showed the name of the calling party instead of the name of the called party]

**Reproduction Steps:**
Example: Direct call
When a user sends a call it appears incorrect info on "placed calls" phone directory
1101-Edi calls to 1105-Oscar
1101 go to Placed Calls directory and see "To Edi 1105" instead of "To Oscar 1101".
It doesnt use the correct cid_name.
Even without using any flag on Dial app (like "I") the name that is showed is incorrect.

Example: Transfer call
When you do a transfer you have the same problem showed on "placed calls" phone directory

Example: Capture call
When you do a capture on two times (indirect capture) with the phone onhook there is no problem. 
If you do the capture with the phone offhook the "answer calls" phone directory info is incorrect.
1101-Edi calls to 1104-Albert
1105-Oscar offhook the phone and press 2 times GPickup
When the call is finished if you go to "Answer calls" phone directory you see that:
From Edi (1105) To Albert
(Details)
From Edi-1101
Alternative 1105
A 1105-Oscar
To Albert
Instead of see that:
From Edi (1101) To Albert
(Details)
From Edi-1101
A 1105-Oscar
To Albert

**chan-sccp version:** (4.3.0-3cc74f5d3 develop)

**asterisk version:** (11.21.2)

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes]
* Problem can be reliably reproduced, doesn't happen randomly: [Yes]
* Platform:
    * [CPU]
    * [64 Bit]
    * [Small Endian]
    * [Debian 8.2]

dkgroot, 2016-05-26:
Ok this is pretty easy to fix:
9.4 firmware requires more information before it will display the right information
CallingParty
OrigCallingParty
CalledParty
OrigCalledParty
LastRedirectingParty

See revision: e50e28e
This version had side effect in other situations, because it was setting the PartyNames, when they where not available, yet. This revision needs to be redone. Correctly. Including carefull testing of :
Incoming Call
Outgoing Call
Incoming SIP Call
Outgoing SIP Call
Call Transfer
Call Pickup

marcelloceschia, 2016-10-10:
@miguelsanzpardo ist this related to #326

dkgroot, 2016-11-18:
please also reference e50e28e

dkgroot, 2016-12-14:
Re-tested this issue with current 'develop' branch and does not seem to occur any more, after some of the patches to sendCallInfoV16 and sendCallInfoV20 in the past weeks.
Closing issue report for now.

closed 2016-12-14 by dkgroot