chan-sccp/chan-sccp #314: Callinfo update on attended transfer does ot always contain the correct parties

dkgroot, 2016-05-26:
Sometimes the callinfo update does not always happen on attended trasnfer (3 out of 5) 
```
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_line.c: -- SCCP: Found line 1101
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: (getEmptyChannel) on line:1101, maybe_c:
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: (getEmptyChannel) getActiveChannel
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_device.c: -- SEP0026CBC03B37: Getting the active channel on device.
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: (getEmptyChannel) got channel already.
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_channel.c: -- SEP0026CBC03B37 can't put on hold an inactive channel SCCP/1101-00000007 with state DIALING (14)... cancelling hold action.
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_device.c: -- SEP0026CBC03B37: ( sccp_channel.c:1563:sccp_channel_hold ) sccp_dev_displayprompt '€-' for line 1 (5)
[2016-05-26 16:15:15] NOTICE[24138][C-00000013] sccp_protocol.c: -- SEP0026CBC03B37: Display prompt on line 1, callid 7, timeout 5
[2016-05-26 16:15:15] ERROR[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: Putting Active Channel SCCP/1101-00000007 OnHold failed -> Cancelling new CaLL
[2016-05-26 16:15:15] ERROR[24138][C-00000013] sccp_channel.c: SEP0026CBC03B37: Can't allocate SCCP channel for line 1101
````
More debug requred, 
'sccp debug core, channel, pbx, callinfo`
To show when callinfo is updated and when it is not

miguelsanzpardo, 2016-05-26:
[full-transfer-attended-notOK.txt](https://github.com/chan-sccp/chan-sccp/files/284766/full-transfer-attended-notOK.txt)
[full-transfer-attended-OK.txt](https://github.com/chan-sccp/chan-sccp/files/284767/full-transfer-attended-OK.txt)

marcelloceschia, 2016-10-10:
@miguelsanzpardo can you give me a hint what line is wrong

miguelsanzpardo, 2017-01-09:
With plain dialplan:

**- BLIND TRANSFER:**
Callplane Updating
1104 calls to 1102
1102 takes the call
1102 press transfer button and press 1103
1102 press again transfer button
1103 see on the Screen "From Albert To Albert" : Bad behaviour
1103 takes the call
1104 see on the screen To Ricard (1103) : Good behaviour
1103 see on the screen To Albert (1104) : Good behaviour 

Phonebook Directory
On the phonebook directory of 1104 on Placed Calls you see: "To Miguel 1102" : Good behaviour
On the phonebook directory of 1102 on Received Calls you see: "From Albert 1104" : Good behaviour
On the phonebook directory of 1102 on Placed Calls you see: "To 1103" instead of "To Ricard 1103": Bad behaviour
On the phonebook directory of 1103 on Received Calls you see: "From Albert (1104) To Albert" : Bad behaviour

BLF updating
1103 has mapped 1104-Albert on BLF. When 1102 press the second time the transfer button the BLF updating is still incorrect. It appears Miguel <- 1104-Albert. Once 1103 takes the call the BLF updating is fine: Ricard <=> 1104-Albert

**- ATTENDED TRANSFER:**
Callplane Updating
1104 calls to 1102
1102 takes the call
1102 press transfer button and pres 1103
1104 takes the call
1102 press again transfer button
1104 see on the screen To Miguel (1102) instead of seeing To Ricard (1103) : Bad behaviour

Phonebook Directory
On the phonebook directory of 1104 on Placed Calls you see: "To 1102" instead of see "To Miguel 1102" : Bad behaviour
On the phonebook directory of 1102 on Received Calls you see: "From Albert 1104" : Good behaviour
On the phonebook directory of 1102 on Placed Calls you see: "To 1103" instead of "To Ricard 1103": Bad behaviour
On the phonebook directory of 1103 on Received Calls you see: "From Albert (1104) To Albert" : Bad behaviour

BLF updating
1102 and 1103 have mapped 1104-Albert on a BLF. After the transfer you can see:
Miguel < = > 1104-Albert instead of seeing Ricard < = > 1104-Albert

dkgroot, 2017-01-10:
@marcelloceschia : the 9.3 / 9.4 firmware issue might be causing some issues with this issue as well (have not had a chance to look into it). (possibly related: #366). Maybe we should fix #366 first and then see if #314 still exists.

Note: When i added 'Hunt' CID information to the gpickup/pickup code, the callinfo worked for both firmware version.

miguelsanzpardo, 2017-03-22:
It looks like to be working fine now with the 9.4 firmware ;)

dkgroot, 2017-03-24:
@miguelsanzpardo 
Thanks for reporting back. Closing this issue