chan-sccp/chan-sccp #315: Generate Busy Tone when receiving Hangup Cause= 'Busy' from remote system (Via SIP)

dkgroot, 2016-05-26:
Generate Busy Tone when receiving Hangup Cause= 'Busy' from remote system (Via SIP)

Busy indication from remote system (for example when call limit reached). 
We put the device onhook before generating BUSY Sound

````
WorkAround:
Dial(xxxx)
if (DialStatus == Busy) {
  Answer()
  Busy()
}
```

Reported by @Valerio

dkgroot, 2016-12-10:
Please verify if ed676e6 fixes this issue correctly

dkgroot, 2016-12-14:
@valerioparis_twitter : Can you verify that this issue does not occur anymore using the latest 'develop' branch revision ?
Issue will be closed for now, please reopen if necessary

ValerioParisMitritsakis, 2016-12-19:
With this dial plan:
> exten => 100,1,Dial(SCCP/08100,60)
> exten => 100,n,GotoIf($["${DIALSTATUS}" = "BUSY"]?${EXTEN}-BUSY:)
> exten => 100,n,Playback(vm-nobodyavail)
> exten => 100,n,Hangup()
> exten => 100-BUSY,n,Busy

I am getting the vm-nobodyavail played back while the extension is busy.

dkgroot, 2016-12-19:
I am not that good with extension.conf (AEL user, sorry). 

Not sure if the GotoIf is correct this way, should that not use "==" instead ? Just to make sure, it might it's  a good idea to add 'Noop(DialStatus: ${DialStatus});' in front. 

I have
```
macro handle_dialstatus(ext) {
        Noop(dialstatus: ${DIALSTATUS});
        switch(${DIALSTATUS}) {
                case BUSY:
                        Voicemail(${ext},b);
                        if (${VMSTATUS}=="FAILED") {
                                PlayTones(Busy);
                                Busy();
                                Wait(10);
                        }
                        break;
                default:
                        Voicemail(${ext},u);
                        if (${VMSTATUS}=="FAILED") {
                                PlayTones(Congestion);
                                Congestion();
                                Wait(10);
                        }
        };
        return;
};
```
And
```
100 => {
        Dial(SCCP/${EXTEN}, 60);
        &handle_dialstatus(${EXTEN});
        Hangup();
}
```
Which does seem to work.

dkgroot, 2016-12-20:
@ValerioParisMitritsakis 
Found the cause for the issue. revision f256055 should fix the incoming_calllimit -> returning UNAVAILABLE instead of BUSY.
Please recheck and close the issue, if fixed

dkgroot, 2016-12-22:
Can you try this patch on current 'develop' branch:
```
diff --git a/src/sccp_pbx.c b/src/sccp_pbx.c
index 2819ef6..6086754 100644
--- a/src/sccp_pbx.c
+++ b/src/sccp_pbx.c
@@ -221,8 +221,8 @@ int sccp_pbx_call(sccp_channel_t * c, char *dest, int timeout)
        if ((l->incominglimit && SCCP_LIST_GETSIZE(&l->channels) > l->incominglimit)) {
                sccp_log((DEBUGCAT_CORE)) (VERBOSE_PREFIX_3 "Incoming calls limit (%d) reached on SCCP/%s... sending busy\n", l->incominglimit, l->name);
                iPbx.queue_control(c->owner, AST_CONTROL_BUSY);
-               iPbx.set_callstate(c, AST_STATE_BUSY);
-               return 0;
+               //iPbx.set_callstate(c, AST_STATE_BUSY);
+               return -1;
        }
 
        /* Reinstated this call instead of the following lines */
@@ -391,7 +391,8 @@ int sccp_pbx_call(sccp_channel_t * c, char *dest, int timeout)
                sccp_channel_send_callinfo(ForwardingLineDevice->device, c);
        } else if (hasDNDParticipant) {
                iPbx.queue_control(c->owner, AST_CONTROL_BUSY);
-               iPbx.set_callstate(c, AST_STATE_BUSY);
+               //iPbx.set_callstate(c, AST_STATE_BUSY);
+               res = -1;
        } else {
                iPbx.queue_control(c->owner, AST_CONTROL_CONGESTION);
                res = -1;
```
I am not sure if returning '-1' makes the differnce or is not setting the callstate to BUSY.

dkgroot, 2016-12-22:
@ValerioParisMitritsakis 
Ran some more tests and fixed the issue as far as i can tell
Committed in branch 'issues/315'

Please review/check/test using
```
git checkout issues/315
make -j2 && make install && make reload
```
Needs to be available on both PBX's has a fix for the calling and the called side.

commit f256055b (2016-12-20): Fix: incoming_call_limit reached should return BUSY state to dialplan

Fixes #315
commit ddeecc3f (2016-12-27): Fix: When busy then queue_control=BUSY, hangupcause=AST_CAUSE_USER_BUSY and return '0' as exit status to DIAL command (#363)

Fix: When we receive a channel PROD indication, do not start audio flow when there is a hangupcause
     otherwise we would not generate busy tones when we receive AST_CONTROL_BUSY

Fixes #315
commit 6fe7eb21 (2016-12-27): Fix: When busy then queue_control=BUSY, hangupcause=AST_CAUSE_USER_BUSY and return '0' as exit status to DIAL command (#363)

Fix: When we receive a channel PROD indication, do not start audio flow when there is a hangupcause
     otherwise we would not generate busy tones when we receive AST_CONTROL_BUSY

Fixes #315
closed 2016-12-27 by dkgroot