chan-sccp/chan-sccp #316: Progress Before Ringout is causing 'Ask the device to open a RTP port' being called Twice (Emergency / Mediat Error)

dkgroot, 2016-05-26:
SIP issuesing Progress indications can cause 
Call to 'Ask the device to open a RTP port' being done 2 times
Which results in Openreceive MediatError

````
FreePBXPri*CLI> core set verbose 3
Console verbose was OFF and is now 3.

 -- SEPACA016FC076B: Looking for line with instance 1.
    -- SCCP: Found line 10502
    -- SEPACA016FC076B: Getting the active channel on device.
    -- SEPACA016FC076B: No active channel on device.
SEPACA016FC076B: (getEmptyChannel) on line:10502, maybe_c:
SEPACA016FC076B: (getEmptyChannel) getActiveChannel
    -- SEPACA016FC076B: Getting the active channel on device.
    -- SEPACA016FC076B: No active channel on device.
 SCCP: Adding channel 7735 to line 10502
    -- SEPACA016FC076B: Set the active channel 7735 on device
    -- SEPACA016FC076B: Set the active line 10502
    -- 0007: New channel number: 7735 on line 10502
    -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 10502,
    -- SCCP:              cid_num: "10502"
    -- SCCP:             cid_name: "Tecnic Sistemes"
    -- SCCP:          accountcode: "10502"
    -- SCCP:                exten: ""
    -- SCCP:              context: "from-internal-internacionales"
    -- SCCP:             amaflags: "0"
    -- SCCP:            chan/call: "SCCP/10502-00001E37"
    -- SCCP:combined capabilities: "[slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a, rfc2833]"
    -- SCCP:  reduced preferences: "[alaw/64k, alaw/56k]"
    -- SCCP/10502-00001E37: (recalculateReadformat)
        capabilities: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a, rfc2833]
        preferences: [alaw/64k, alaw/56k]
        remote caps: []
        READ:[alaw/64k]
    -- SCCP/10502-00001E37: (recalculateWriteformat)
        capabilities: [slin16, ulaw/64k, alaw/64k, ilbc, g729b, g729ab, g729, g729a, rfc2833]
        preferences: [alaw/64k, alaw/56k]
        remote caps: []
        Write:[alaw/64k]
    -- 0007: Allocated asterisk channel SCCP/10502-00001E37
    -- SEPACA016FC076B: Set asterisk state OffHook (2) for call 7735
    -- SEPACA016FC076B: Indicate SCCP new state 10 (OFFHOOK), current channel state 0 (DOWN) on call SCCP/10502-00001E37 (previous channelstate 0 (DOWN))
    -- SEPACA016FC076B: Send speaker mode 'on'
    -- SEPACA016FC076B: Send and Set the call state offhook(1) on call 7735 (visibility:default)
    -- SEPACA016FC076B: Send activate call plane on line 1
    -- SEPACA016FC076B: ( sccp_indicate.c:91:__sccp_indicate ) sccp_dev_displayprompt '▒ ' for line 1 (3)
    -- SEPACA016FC076B: Display prompt on line 1, callid 7735, timeout 3
    -- SEPACA016FC076B: Sending tone Inside Dial Tone (33) on line 1 with callid 7735 (direction: User)
    -- SEPACA016FC076B: Set softkeyset to OFFHOOK(4) on line 1  and call 7735
    -- SEPACA016FC076B: validKeyMask 4294967295
    -- SEPACA016FC076B: Finish to indicate state SCCP (OFFHOOK) on call SCCP/10502-00001E37. New state on channel: OFFHOOK (10)
    -- SCCP/10502-00001E37: schedule digittimeout 16
    -- SEPACA016FC076B: Accessory 'Speaker' is 'Off Hook'
    -- SEPACA016FC076B: EnblocCall, party: 0061, lineInstance 1
    -- SEPACA016FC076B: Getting the active channel on device.
    -- SCCP/10502-00001E37: stop schedule digittimeout 21
    -- SEPACA016FC076B: (sccp_pbx_softswitch) New call on line 10502
    -- SEPACA016FC076B: (sccp_pbx_softswitch) Dial Extension 0061
    -- SEPACA016FC076B: Indicate SCCP new state 14 (DIALING), current channel state 10 (OFFHOOK) on call SCCP/10502-00001E37 (previous channelstate 0 (DOWN))
    -- SEPACA016FC076B: Stop tone on line 1 with callid 7735
    -- SEPACA016FC076B: Send the dialed number:0061 channel 7735
    -- SEPACA016FC076B: Set softkeyset to RINGOUT(8) on line 1  and call 7735
    -- SEPACA016FC076B: validKeyMask 4294967293
    -- SEPACA016FC076B: Send and Set the call state proceed(12) on call 7735 (visibility:default)
    -- SEPACA016FC076B: Finish to indicate state SCCP (DIALING) on call SCCP/10502-00001E37. New state on channel: DIALING (14)
    -- SCCP: (sccp_pbx_softswitch) set variable SKINNY_PRIVATE to: 0
    -- SCCP: (sccp_pbx_softswitch) set variable SKINNY_PRIVATE to: 0
    -- +- pbx extension matcher (0061           ): ---+
    -- |ignore     |exists     |can match  |match more|
[... 195 lines cut ...]
    -- SCCP: destroying PBX rtp server on channel SCCP/10502-00001E37
    -- SEPACA016FC076B: Got OpenChannel ACK.  Status: 'Media Error: Unknown' (1), Remote RTP/UDP '172.17.32.29:0', Type: Indirect RTP, PassThruPartyId: 4294959560, CallID: 7735
    -- SEPACA016FC076B: Getting the active channel on device.
    -- SEPACA016FC076B: No active channel on device.
    -- SCCP: Looking for channel by PassThruId 4294959560 on device SEPACA016FC076B
    -- SEPACA016FC076B: Found line: '10502'
    -- SEPACA016FC076B: Could not find active channel with Passthrupartyid 4294959560 on device
[2016-05-23 13:52:35] ERROR[18226][C-00000672]: sccp_actions.c:3242 handle_open_receive_channel_ack: SEPACA016FC076B: (OpenReceiveChannelAck) Device returned: 'Media Error: Unknown' (1) !. Giving up.
    -- SEPACA016FC076B: Getting the active channel on device.
    -- SEPACA016FC076B: No active channel on device.
    -- SCCP: Looking for channel by PassThruId 4294959560 on device SEPACA016FC076B
    -- SEPACA016FC076B: Found line: '10502'
    -- SEPACA016FC076B: Could not find active channel with Passthrupartyid 4294959560 on device
[2016-05-23 13:52:35] WARNING[18226][C-00000672]: sccp_actions.c:3419 handle_startmediatransmission_ack: SEPACA016FC076B: Channel with passthrupartyid 4294959560 / callid 7735 / callid1 7735 not found, please report this to developer
    -- SEPACA016FC076B: Clear the display notify message
    -- SEPACA016FC076B: Clear the status prompt on line 0 and callid 0
    -- SEPACA016FC076B: ( sccp_device.c:1937:sccp_dev_check_displayprompt ) sccp_dev_displayprompt '▒' for line 0 (0)
    -- SEPACA016FC076B: Display prompt on line 0, callid 0, timeout 0
    -- SEPACA016FC076B: Set softkeyset to ONHOOK(0) on line 0  and call 0
    -- SEPACA016FC076B: validKeyMask 4294967295
    -- SEPACA016FC076B: Accessory 'Speaker' is 'On Hook'
````

Which prevents Miguel from dialing Emergency number for example.
Reported by @miguelsanzpardo

dkgroot, 2016-05-26:
Related 9f53c83

dkgroot, 2016-05-26:
Reproducting the events using dialplan on remote server containing:
```
Progress();
Wait(1);
Ringing();
Wait(1);
Answer();
Wait(1);
MusicOnHold(default);
Hangup();
```
Generated the following debug on the calling side when using the new revision: 9f53c83
```
  SCCP: (sccp_pbx_softswitch) exit
    -- Executing [10005@internal:1] NoOp("SCCP/98041-00000004", "Test guest busy") in new stack
    -- Executing [10005@internal:2] Dial("SCCP/98041-00000004", "IAX2/xxx:[xxxxt]@xxx.xxxx.xxx/908, 40") in new stack
    -- Called IAX2/dkgroot:[dkgroot]@asterisk.chan-sccp.net/908
    -- Call accepted by xxx.xxx.xxx.xxx:4569 (format g722)
    -- Format for call is (g722)
    -- IAX2/asttest-4814 is making progress passing it to SCCP/98041-00000004
    --     -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS'
    --     -- SEP0023043403F9: (pbx_indicate) start indicate 'AST_CONTROL_PROGRESS: Indicate PROGRESS' (14) condition on channel SCCP/98041-00000004 (readStat:0, receiveChannelState:0, rtp:no, default res:inband signaling (-1))
    --     -- SEP0023043403F9: Indicate SCCP new state:PROGRESS, current channel state:DIALING on call:SCCP/98041-00000004, lineInstance:1 (previous channelstate:OFFHOOK)
  ==   == SEP0023043403F9: SCCP_CHANNELSTATE_PROGRESS
    --     -- SCCP: Asterisk requests to change state to (Progress) from (DIALING)
  == Using SCCP RTP TOS bits 184
  == Using SCCP RTP TOS bits 184 in TCLASS field.
  == Using SCCP RTP CoS mark 6
    --     -- SEP0023043403F9: Open receive channel with format G.711 A-law 64k[2], payload 2, echocancel: YES, passthrupartyid: 4294967291, callid: 4
    --     -- SEP0023043403F9: Finish to indicate channel state:PROGRESS on call:SCCP/98041-00000004, lineInstance:1. New channel state:PROGRESS
  ==   == SEP0023043403F9: (pbx_indicate) finish: send indication 'inband signaling' (-1)
    --     -- SEP0023043403F9: Getting the active channel on device.
    --     -- SEP0023043403F9: Set the active channel 4 on device
    --     -- SEP0023043403F9: send callInfo on SCCP/98041-00000004 with lineInstance: 1
    --     -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    --     -- SEP0023043403F9: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/98041-00000004 (readStat:2, receiveChannelState:2, rtp:yes, default res:inband signaling (-1))
  ==   == SEP0023043403F9: (pbx_indicate) finish: send indication 'inband signaling' (-1)
    --     -- SEP0023043403F9: Getting the active channel on device.
       > 0x2e61240 -- Probation passed - setting RTP source address to [xxxx:xxxx:xxxx:xxxx:xxxx]:22812
    -- IAX2/asttest-4814 is ringing
    --     -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_RINGING: Remote end is ringing'
    --     -- SEP0023043403F9: (pbx_indicate) start indicate 'AST_CONTROL_RINGING: Remote end is ringing' (3) condition on channel SCCP/98041-00000004 (readStat:2, receiveChannelState:2, rtp:yes, default res:inband signaling (-1))
    --     -- SEP0023043403F9: Indicate SCCP new state:RINGOUT, current channel state:PROGRESS on call:SCCP/98041-00000004, lineInstance:1 (previous channelstate:DIALING)
    --     -- SEP0023043403F9: Finish to indicate channel state:RINGOUT on call:SCCP/98041-00000004, lineInstance:1. New channel state:RINGOUT
    --     -- SEP0023043403F9: Set asterisk state Ring (4) for call 4
  ==   == SEP0023043403F9: (pbx_indicate) finish: send indication 'inband signaling' (-1)
    --     -- SEP0023043403F9: Asterisk prodded channel SCCP/98041-00000004.
    -- IAX2/asttest-4814 answered SCCP/98041-00000004
    --     -- SCCP: sccp_pbx_answer
    --     -- SEP0023043403F9: sccp_pbx_answer checking parent channel
    --     -- SEP0023043403F9: (sccp_pbx_answer) Outgoing call SCCP/98041-00000004 has been answered by remote party
    --     -- SEP0023043403F9: Indicate SCCP new state:PROCEED, current channel state:RINGOUT on call:SCCP/98041-00000004, lineInstance:1 (previous channelstate:PROGRESS)
    --     -- SEP0023043403F9: Finish to indicate channel state:PROCEED on call:SCCP/98041-00000004, lineInstance:1. New channel state:PROCEED
    --     -- SEP0023043403F9: Indicate SCCP new state:CONNECTED, current channel state:PROCEED on call:SCCP/98041-00000004, lineInstance:1 (previous channelstate:RINGOUT)
    --     -- SEP0023043403F9: Finish to indicate channel state:CONNECTED on call:SCCP/98041-00000004, lineInstance:1. New channel state:CONNECTED
    --     -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel'
    --     -- SEP0023043403F9: (pbx_indicate) start indicate 'AST_CONTROL_PROD: Kick remote channel' (-1) condition on channel SCCP/98041-00000004 (readStat:2, receiveChannelState:2, rtp:yes, default res:inband signaling (-1))
  ==   == SEP0023043403F9: (pbx_indicate) finish: send indication 'inband signaling' (-1)
    -- Channel IAX2/asttest-4814 joined 'simple_bridge' basic-bridge <97ccf7bd-2e24-4273-8906-b477fa31a1cb>
    -- Channel SCCP/98041-00000004 joined 'simple_bridge' basic-bridge <97ccf7bd-2e24-4273-8906-b477fa31a1cb>
    --     -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_SRCCHANGE: Media source has changed and requires a new RTP SSRC'
    --     -- SEP0023043403F9: (pbx_indicate) start indicate 'AST_CONTROL_SRCCHANGE: Media source has changed and requires a new RTP SSRC' (26) condition on channel SCCP/98041-00000004 (readStat:2, receiveChannelState:2, rtp:yes, default res:inband  [... cut ...]
    --          Last Call        : CallID: 4 Packets sent: 180 rcvd: 31 lost: 0 jitter: 1 latency: 0
    --          Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.5123;CCR=0.5123;ICRmx=0.5123|CS=2;SCS=2
    --          Mean Statistics  : #Calls: 4 Packets sent: 291 rcvd: 175 lost: 0 jitter: 2 latency: 0
    --          Mean Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.2242;CCR=0.2250;ICRmx=0.5123|CS=2;SCS=1

```
Which seems to prove that the change is helping

dandrzejewski, 2016-05-26:
Verified as fixed in 4.3.0 develop - b725c2cM.

dkgroot, 2016-05-27:
@dandrzejewski 
Bute this is a problem in the current 'develop' branch. It only got fixed in the feature/revert_to_af666b3b branch. So I don't understand you comment really, can you elaborate ?

dkgroot, 2016-12-14:
Retested with current develop branch, and does not occur any more. No further reports. Closing issue

closed 2016-12-14 by dkgroot