chan-sccp/chan-sccp #318: Conferencing does not work correctly in develop branch

instipod, 2016-06-06:
When creating a conference, the leader (device that created the conference) is unable to hear or send audio to the conference.  This occurs when using the develop branch.  If the stable release 4.2.3 version is installed, conferencing works as expected.

**Reproduction Steps:**

1. Install develop branch chan-sccp
2. Create a conference with a phone (conference should have at least 3 parties total)

**Expected behavior:**

The leader phone is able to communicate with the conference.

**Observed behavior:**

The leader is unable to hear or speak to members of the conference.  Other members of the conference can correctly hear and speak to each other.  

**chan-sccp version:** develop branch commit 3cc74f5d3c2d03dd60d24181d43bc13e0bfebdc2 (latest)

**asterisk version:**  13.9.1 (from source)

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes, works as expected in STABLE release 4.2.3
* Problem can be reliably reproduced, doesn't happen randomly: Yes
* Platform:
    * Intel Xeon 
    * 64 Bit
    * Ubuntu
    * 14.04.4 LTS
* Direct Contact information: instipod@instipod.com

instipod, 2016-06-07:
Commit f6eddf1a0796717b39d59aeb88dada500857f382 (May 4, 2016; Fix: setting CHANNELSTATE_CONNECTED should be reentrant) appears to be the breaking source, running it or later commits causes conferencing to be broken.  

Running commit 656e2fba5731ea1a437c6bfda42a96d58ae8cf60 (May 3, 2016; Revert bd49e2e: Current caller name on speed dial line doesn’t update) causes conferencing to work as expected.

dandrzejewski, 2016-06-15:
Set this to Priority 2. Conferencing is a pretty integral feature.

dkgroot, 2016-06-18:
Thanks for reporting this. Because of some major back problems i currently don't have the time to investigate the cause. Nothing has changed in the last weeks/months regarding conference. So must be caused by changes in indications i guess

sdstorts, 2016-10-12:
I am seeing this same issue on our systems.  I did find that you can add another line to the bridge and the leader can hear again and everyone else can hear the leader.  I am on Release: 4.3.0 develop - ae9ae62M (built by 'root' on '2016-10-05 11:10:02 UTC'.

marcelloceschia, 2016-10-12:
@sdstorts thank you for your feedback, I will check this issue at the end of this week

sdstorts, 2016-10-24:
Have you gotten a chance to take a look at this issue?  I have some users wanting an update on getting this fixed.  
Thanks for all you do!

sdstorts, 2016-11-09:
Any update on this?

dkgroot, 2016-11-09:
Hi @sdstorts ,
I issues the original conference module in the past, and have not been able to work on the chan-sccp project for a while.
I will see what I can do in the coming days.
@marcelloceschia might be easier for me to look into this, if you are ok with that.

sdstorts, 2016-11-09:
That would be awesome!  Thanks!

dkgroot, 2016-11-18:
Hi instipod
I just noticed you tried merging the revert_branch and the current master branch in your own repository, and ran into some difficulties. I had the same issue back then, trying to bisect what changed and what was causing the issues. I never made any further progress because of illnes and the revert branch is now sort of left behind.
I don't want to discourage you in any way, and if you manage to get it into working order we would certainly take that into account.
We will try to make careful steps forward to get the current develop branch back into running order. Merging the revert_branch will not be an option as far as i can see.

dkgroot, 2016-11-19:
@instipod, @dandrzejewski  and @sdstorts 

Just commited a fix for conference to re-enable audio on the moderator channel. Can you guys check the latest development branch to see if the fixes the issue for you as well ?

dkgroot, 2016-11-19:
Also links to #316

instipod, 2016-11-20:
@dkgroot 

I pulled the changes from the development branch as of an hour ago and the moderator audio is working correctly!  Thanks so much.

dkgroot, 2016-11-20:
Hi Michael,
Thanks for checking/testing and reporting back. Closing this issue.

commit 942fcec3 (2016-11-19): Fix conference moderator.

During CONNECTEDCONFERENCE indication the previousChannelState check was missing a cause,
causing the openReceiveChannel not to be (re)opened. Audio for the moderator was missing
from the conference.

closes #318
closed 2016-11-20 by dkgroot