chan-sccp/chan-sccp #323: Multiple calls stop ringing after first call is answered.

stuartykins, 2016-08-02:
I seem to have an issue when I have more than one call ringing at the same time, when the first call is answered, the ringing stops on the other phone. I have searched for similar support requests but was unable to find anything.
We have two phones in our reception area that ring on incoming calls (the two SCCP extensions are in a ring group) and today we had two calls coming in at the same time, so both the phones were ringing and one of the phones was answered leaving the other call "ringing" on the system - but the other phone stopped physically ringing so the person at the other desk didn't realise there was another incoming call.
We later recreated the scenario by placing two calls to our incoming number, and the two SCCP extensions started ringing, but as soon as one of the calls was answered, the other one stopped ringing - but the handset LED was still flashing on the second phone, and the softkeys had Answer/End Call as the second call hadn't actually been answered.
This also happens on a single phone when there is call waiting activated, when the first call is ended, the second call is still presented as a "Call Waiting" on the screen, but the phone does not actually ring.

Thanks in advance

chan-sccp version: 4.2.3 STABLE 6625M
asterisk version: 11.22.0

marcelloceschia, 2016-08-02:
@stuartykins What phone type is the receptions phone and what protocol version / firmware version?

stuartykins, 2016-08-02:
@marcelloceschia Thanks for the quick response.

One phone is a Cisco 7970 running on image version SCCP70.9-4-2-1S using protocol version 22.
The second is a Cisco 7961 running on image version SCCP41.9-4-2-1S using protocol version 22.

marcelloceschia, 2016-08-02:
@stuartykins thank you. I will need some time to find an appropriate solution (counting incoming calls and the calculated ringtone)

jamesmz, 2016-11-24:
I'm having a similar issue. Person A calls us, then Person B calls a few seconds later. All phones ring saying Person A is calling. That call is answered. Person B is still calling but none of the chan sccp phones are ringing so we can't answer the call on them. We're aware that they are still calling as they also go through to a cordless SIP phone which we are able to answer if it is not already in use. We're using Cisco 7940Gs and 7960Gs. Unlike stuartykins however, we do not see the LED flashing or the answer softkey. The phones just remain idle as if nobody is actually calling.

dkgroot, 2016-12-14:
@stuartykins / @jamesmz 
I tried reproducing the issue (using 'develop' branch), but did not find any abnormalities. 
Further tests performed:
- Even set incoming_limit=1 on one of the devices and that was obeyed correctly as well. 
- If both calls are answered, then no 'missed calls' entry was produced.

Can you guys verify if this issue (still) exists with current 'develop' branch ? 
```
git fetch
git checkout develop
make -j2 && make install && make reload
```

jamesmz, 2016-12-17:
I can confirm that the issue no longer occurs after updating to the latest version.

dkgroot, 2016-12-17:
@jamesmz Thanks for reporting back
Closing this issue

stuartykins, 2016-12-18:
Thanks for the fix, will try it once I'm back to work

jamesmz, 2017-01-19:
Hi. This issue seems to have returned after updating to the latest version I'm afraid.

dkgroot, 2017-01-19:
Hi Jamesmz, what was your provious revision (makes it a little easier to got through he changelog for me). You can look this up using git reflog, if necessary.

jamesmz, 2017-01-20:
@dkgroot I can't be 100% certain, as I deleted the folder (thought I had to do this each time before git cloning a new version, sorry). But looking at the date I'd say it was either 1114bea or ef9be53.

dkgroot, 2017-01-20:
@jamesmz Ok thanks, i will see what i can find.

dkgroot, 2017-01-20:
@jamesmz 
Would you be able to do a 'git bisect' to help find the revision that introduced the bug.

`git bisect start --term-good=1114bea --term-bad=610c086`
`make && make install && make reload`
do your test and decide if good or bad
`git bisect [bad|good]`

Bisect will make the decision which intermediate versions you should check/test to get to the revision. 
which first introduced the bug.

stuartykins, 2017-01-26:
I know there's been a lot of testing going on while I've not been at work, so thank you guys for your help so far, and thanks to jamesmz for testing :)

jamesmz, 2017-01-26:
Hi. Just to let you know I've not forgotten about this. I've just been struggling to find a time when I can do the git bisect testing when the system is not in use. I'll report back as soon as I can.

jamesmz, 2017-02-01:
OK, now I'm really confused, as I can't get it to go back to working correctly with any commit, even with the one it worked with previously.

jamesmz, 2017-02-01:
I forgot to mention that the second part of the problem involves hanging up one call whilst other phones are ringing, but not having the incoming call present on the phone you just put down. Could this be linked or should I start a new issue thread?

Let me know if I can help in any way by posting logs etc.

stuartykins, 2017-02-01:
I've not had that issue on the answering phone, when the call is ended, the second call is still displayed on the screen and the red handset light flashes along with the line key LED.
We have a mix of different phones, 7940, 7960, 7961 and 7970 and the calls appear on all of them when they answer the first call. I added them all to the incoming ring group to check this after I reported the initial bug. Maybe you have a different firmware with the phone, I'm not sure :(

jamesmz, 2017-02-01:
Under firmware versions I'm seeing "App Load ID P0030801SR02" "Boot Load ID PC0303010200" "Version 8.1(SR.2)"

jamesmz, 2017-02-12:
I've sorted this now. It was my fault as I somehow managed to set incominglimit to 1 for all phones in sccp.conf. Sorry for the confusion.

dkgroot, 2017-02-15:
Thanks for reporting back. Closing this issue now 😊

closed 2017-02-15 by dkgroot