chan-sccp/chan-sccp #331: GPickup Enabled When Not Listed in Softkey Config

dandrzejewski, 2016-08-28:
I have set up a softkey configuration that does not have gpickup, yet the gpickup button still appears.  This is on a 7970G.

**Reproduction Steps:**

# Configure a softkeyset without gpickup
# Note that gpickup still appears

**Expected behavior:**

gpickup should not appear.

**Observed behavior:**

gpickup appears.

**chan-sccp version:** (include revision if relevant)

Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - 575aa22M (built by 'root' on '2016-08-14 22:36:39 UTC')

**asterisk version:** (include revision if relevant)

13.7.2

**Addition / Optional Information:**

    softkeyset=softkeys

    [softkeys]
    onhook=redial,newcall
    connected = hold,endcall,park,select
    onhold = resume,newcall,endcall,transfer,confrn,select,dirtrfr
    ringin = answer,endcall
    offhook = redial,endcall
    conntrans = hold,endcall,transfer,confrn,park,select
    digitsfoll = back,endcall
    connconf = hold,endcall,join
    ringout = endcall,transfer,cfwdall,idivert
    offhookfeat = redial,endcall
    onhint = pickup,barge

marcelloceschia, 2016-08-28:
@dandrzejewski the context [softkeys] is missing the config type type=softkeyset
Please check if you set type=softkeyset in your configuration and check the parsing result via asterisk cli: `sccp show softkeyssets`

marcelloceschia, 2016-09-03:
seems to work

closed 2016-09-03 by marcelloceschia