chan-sccp/chan-sccp #333: 7925/7921 phones suddenly are unregistered and cant be registered again until asterisk service is rebooted

miguelsanzpardo, 2016-09-21:
The problem is that suddenly the phone is unregistered and if we power off and then power on the phone, then the phone is constantly changing between these RegStates:
None --> Progress --> None --> Progress
Instead of having OK state

# asterisk -rx "sccp show devices" | grep 11111
| Ext 11111                                                           -- SEP7CAD7442568A  None       No To  Wed Sep 21 13:18:02 2016 Yes 1     (Auto)Off |

# asterisk -rx "sccp show devices" | grep 11111
| Ext 11111                                                           -- SEP7CAD7442568A  Progress       No To  Wed Sep 21 13:19:02 2016 Yes 1     (Auto)Off |

**Reproduction Steps:**
The problem appears suddenly on randomly manner (Or at least I think so)

**Expected behavior:**
RegState : OK

**Observed behavior:**
RegState : None --> Progress -->  None --> Progress 

**chan-sccp version:** 4.2.0.6168

**asterisk version:** 11.21.2

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: 4.2.0.6168
* Problem can be reliably reproduced, doesn't happen randomly: No
* Platform:
    * 64 Bit
[Info_7921-7925-Progress.txt](https://github.com/chan-sccp/chan-sccp/files/484891/Info_7921-7925-Progress.txt)
[Info_7921-7925-Progress.txt](https://github.com/chan-sccp/chan-sccp/files/484892/Info_7921-7925-Progress.txt)

    * Operating System: Debian 8.3

miguelsanzpardo, 2016-11-28:
What debug flags should activate when this problem appears?

miguelsanzpardo, 2016-12-12:
Diederik, Marcello, what debug flags should activate to analyze this problem when it appears again?

dkgroot, 2016-12-12:
Hi Miguel. The issue has existed for some time now, but we still don't know what causes it. We have not found a debug level that gives any conclusive evidence to what happened exactly. At least not when this debug level is added after the looping of the devices has started. There might be evidence from before that time, but we have not seen it so far.

It's especially hard to debug because the issue is hard to reproduce. It only seems to occur with any regularity on bigger systems (many phones) and those systems can't wait for a long time, before they need to restart to continue work.

I did make a small change to sccp_session.c last week, that might be use full with this issue (not sure). If you are running the latest develop revision you might be able to check if the issue still occurs with it. I might take another couple of weeks to find out if it does help.

miguelsanzpardo, 2016-12-13:
Umh, I am using chan_sccp 4.2.0.6168 right now, do you think that the session issue which appeared a months ago comes from previous chan sccp versions (like 4.2.0) or is related to new releases (like 4.2.3-4.3.0)? Maybe is not exactly the same issue? What do you think?

dkgroot, 2016-12-13:
Marcello fixed the session issue a couple of weeks ago.
I changed a setting to make the devices use a different port for each connection, which caused another issue with devices looping. 

The issue with the looping between None and Progress, most likely comes from a refcount error. Finding that refcount error on a large server with lots of phones is quite a difficult and daunting task.

miguelsanzpardo, 2016-12-13:
Ok, nice to know all of this info.

You said that: "We have not found a debug level that gives any conclusive evidence to what happened exactly.At least not when this debug level is added after the looping of the devices has started"

It would be interesting to know what debug flags could help us to detect this refcount issue.
Do you intuit what flags could help us to detect this problem being activated before the problem appears? Maybe I could activate any flags and if the segfault appears we can catch it.

dkgroot, 2016-12-13:
If the looping issue is indeed created by a refcount problem, then 
```
./configure --enable-refcount-debug
and/or
sccp debug core, refcount
```
Would potentially help. But with a large amount of phones the amount of debug information would be impossible to check/read/investigate. 

You would have to have a large amount of free diskspace and/or a correct log rotation script in place. Note: a single call can produce 40-60 lines of refcount debug output.

If the issues takes multiple days or weeks before it happens / is detected, you would have to gather all that data for all that amount of time (from starting the server until the issue arises). With some intelligent scripting and comparing differences you might be able to reduce the data set quite a bit, but i am not sure you are up to the task. You would have to understand refcounting and how we use it, to make intelligent decisions about what data can be scrapped and what needs to be kept.

And then we would need a large amount of time to go through all that logging to maybe find a potential cause (not sure if you find it, because you don't know what exactly you are looking for). For which we would then have to write a possible patch. And then start all over again, to check if it still occurs.

The amount of data that needs to be gathered plus the time needed to investigate that data is huge. And we don't even know if the issue is caused by refcount (just a guess). It's what i would call an insurmountable task.

miguelsanzpardo, 2016-12-13:
Ufff, it sounds so complicated ^_^", maybe with the last develop branch the problem doesnt appear, when develop branch is more debugged I will install on production, by now I am going to wait

dkgroot, 2016-12-20:
Closing this issue report until further notice. Testing of the solution will commence in #358. 
Revision ebe7837 fixes this issue as well.

miguelsanzpardo, 2017-02-13:
With the Release: 4.3.0 issues/370 - f19b107M I have seen again this issue.

I was using sccp debug with these flags: core,action,channel,pbx,message because i was analyzing another different issues.

[Re-register.zip](https://github.com/chan-sccp/chan-sccp/files/771602/Re-register.zip)

dkgroot, 2017-02-14:
Found the cause and submitted a patch
Thanks for reporting the issue.

miguelsanzpardo, 2017-03-13:
I share with you the info related with the last re-registering issue that has appeared on a 7921 phone with MAC ADDRESS:  SEP001DA2319FB5

I have activated:
sccp debug core, refcount, device, socket, action, line , channel, filelinefunc , filelinefunc
[full.txt](https://github.com/chan-sccp/chan-sccp/files/838866/full.txt)

I have executed several times the command "sccp show refcount" meanwhile the device is registering-unregistering:
[refcount_01.txt](https://github.com/chan-sccp/chan-sccp/files/838856/refcount_01.txt)
[refcount_02.txt](https://github.com/chan-sccp/chan-sccp/files/838855/refcount_02.txt)
[refcount_03.txt](https://github.com/chan-sccp/chan-sccp/files/838858/refcount_03.txt)
[refcount_04.txt](https://github.com/chan-sccp/chan-sccp/files/838857/refcount_04.txt)
[refcount_05.txt](https://github.com/chan-sccp/chan-sccp/files/838859/refcount_05.txt)
[refcount_06.txt](https://github.com/chan-sccp/chan-sccp/files/838860/refcount_06.txt)
[refcount_07.txt](https://github.com/chan-sccp/chan-sccp/files/838862/refcount_07.txt)
[refcount_08.txt](https://github.com/chan-sccp/chan-sccp/files/838863/refcount_08.txt)
[refcount_09.txt](https://github.com/chan-sccp/chan-sccp/files/838861/refcount_09.txt)
[refcount_10.txt](https://github.com/chan-sccp/chan-sccp/files/838864/refcount_10.txt)

dkgroot, 2017-03-13:
Obviously there is a refcount problem in this case. The device refcount has gone up to 24 when the device register, which is way to much. A more realistic value would be between 3 or 4 like the other devices. We now need to figure out why / where this refcount has gone wrong. Because the device is using wireless, there is a chance that this might be part of the cause.

What was the uptime of the chan-sccp module before the issue showed itself. If not to long, then increasing the debug level might help to figure out when/where the issue is.

dkgroot, 2017-03-13:
One other note: looking at the refcount fill factor, you need to increase the hash size when running configure. You should use a value which is a prime, and make sure the fill factor goes below 1.00. I would start with --with-hash-size=5003 and recompile.

dkgroot, 2017-08-31:
References #410

commit ea797e96 (2017-02-14): Fix: Device Cleaning causing Registration issues

Note: This does not fix the cause for the re-registration. I have not found out why devices sometimes:
- loose there connection
- do not respond until there session times out
- go into a device restart

Fixes #378
Fixes #333
closed 2019-12-17 by dkgroot