chan-sccp/chan-sccp #334: Wrong presence state after transfer

epasqualotto, 2016-09-21:
Dear all, I'm testing in my lab the chan_sccp v. 4.3.0  with some 7945 and 7965 phones and I found a strange issue.
When a phone transfer a call to another exten it remain with the presence state InUse and can't receive calls.
If I put the handset of that phone up/down the presence state is cleaned back to Idle and can receive call, seems that is only an issue with the presence state becase the sccp line is Idle.

PBX*CLI> core show hint 6100
                   6100@ext-local           : SCCP/6100,CustomPres  State:InUse           Watchers  0
1 hint matching extension 6100
    -- SEP00235EB66620: Taken Offhook
    -- SEP00235EB66620: Using line 6100
SEP00235EB66620: (getEmptyChannel) on line:6100, maybe_c:
SEP00235EB66620: (getEmptyChannel) getActiveChannel
    -- SCCP/6100-00000122: schedule digittimeout 16
    -- SEP00235EB66620: is Onhook (buttonIndex: 0, callid: 0)
  == SEP00235EB66620: Ending call SCCP/6100-00000122 (state:OFFHOOK)
    -- SEP00235EB66620: Sending hangupRequest to Call SCCP/6100-00000122 (state: OFFHOOK)

PBX*CLI> core show hint 6100
                   6100@ext-local           : SCCP/6100,CustomPres  State:Idle            Watchers  0
1 hint matching extension 6100
PBX*CLI>

Trying to troubleshoot that issue I found also that I can't change the state with "devstate change" Asterisk command.

**Reproduction Steps:**

1. PhoneA call PhoneB
2. PhoneB transfer call to PhoneC (with consult)
3. PhoneA talk with PhoneC
4. PhoneB is marked "InUse" on the Asterisk presence state

**Expected behavior:**

PhoneB has state Idle.

**Observed behavior:**

PhoneB has state InUse, until the handset is put up/down

**chan-sccp version:** 4.3.0

**asterisk version:** 11.x
[debug_transfer_clean.txt](https://github.com/chan-sccp/chan-sccp/files/485095/debug_transfer_clean.txt)

* Comments: [Other Information]

I can't found also a way to change the device_state of the PhoneB by using "devstate change" command.

epasqualotto, 2016-09-21:
as a wordaround is there a way to change the presence_state with a AGI script,shell cmd or other?

marcelloceschia, 2016-09-21:
@epasqualotto no, there is no workaround.

epasqualotto, 2016-09-22:
Other info about the issue:

======= OK ==============
    -- SCCP: we got a fixup request for SCCP/6100-00000034<ZOMBIE> and SCCP/6101-00000032
    -- SCCP/6101-00000032: set c->hangupRequest = requestQueueHangup
    -- SCCP: we got a fixup request for SCCP/6101-00000032 and SCCP/6100-00000034<ZOMBIE>
    -- SCCP/6100-00000034: set c->hangupRequest = requestHangup
    -- Stopped music on hold on SCCP/6100-00000034<ZOMBIE>
    
======= END OK ======================

======= FAIL ==========================   
    -- SCCP: we got a fixup request for SCCP/6100-0000001E<ZOMBIE> and SCCP/6101-0000001C
    -- SCCP/6101-0000001C: set c->hangupRequest = requestQueueHangup
    -- SCCP: we got a fixup request for SCCP/6101-0000001C and SCCP/6100-0000001E<ZOMBIE>
    -- SCCP/6100-0000001E: set c->hangupRequest = requestHangup
[2016-09-22 14:08:48] WARNING[17175][C-00000002]: rtp_engine.c:1496 ast_rtp_instance_bridge: Got hangup while attempting to bridge 'SCCP/6100-0000001E<ZOMBIE>' and 'SCCP/6100-0000001D'
    -- Native bridging SCCP/6100-0000001E<ZOMBIE> and SCCP/6100-0000001D ended
    -- Stopped music on hold on SCCP/6100-0000001E<ZOMBIE>    
    
======== END FAIL ============================   
Looking into the attached SCCP DEBUG seems that channel try to notify Asterisk about the new state but I don't know what's happen after.
During my test seems that issue appears about 1 time on 5, not always

[sccp_debug.txt](https://github.com/chan-sccp/chan-sccp/files/487652/sccp_debug.txt)

dkgroot, 2016-12-11:
@epasqualotto 

I have created a patch for another issue: 324, which also involves hints and presence state, can you check if this also fixes your issue ?
```
git checkout issues/324
make && make install && make reload
```
If it does we can move ahead and merge this into the develop branch.

dkgroot, 2016-12-27:
@epasqualotto No further reply from you
The issues/324 branch has been merged into develop. It should also take care of the issue you reported.

PR #362 (2016-12-27): Issues/324
Fixes #324
Fixes #334
closed 2016-12-27 by dkgroot