chan-sccp/chan-sccp #335: Can't do transfer on phone 6911 with

voipAsteriskGuy, 2016-09-28:
When i receive a call to a phone model 6911 i cant transfer the call

Reproduction Steps:
1. Call from extension A to extension B (B must be a 6911 phone)
2. On extension B press the transfer button after this you will ear the signal tone
3. if you start dialing an extension you will continue listening to the signal tone

Expected behavior:
After you press the transfer button and start dialing an extension the signal tone must stop and after finished dialing the extension the call must be transfered

chan-sccp version:
Release: 4.3.0 TRUNK
Release: 4.3.0 develop
Release: 4.2.3 STABLE
asterisk version:11.17.1
Addition / Optional Information:
Problem started happening recently, didn't happen in an older version of Chan-SCCP: [No] it append on versions 4.3.0 TRUNK, 4.3.0 develop and 4.2.3 STABLE 
Problem can be reliably reproduced, doesn't happen randomly: [Yes]
Platform:
     [CPU] Intel(R) Xeon(R) CPU E5-2630 v4 @ 2.20GHz
     [32/64 Bit] 64 Bit
     [Big/Small Endian]
     [Operating System] Centos
     [Operating System Version] 5.9 and 7.0 append in both also tried on ubuntu 14.04
Phone:
     Firmware used: SCCP6911.9-3-1-SR1-3 protocol version 20
                              SCCP6911.9-0-2-0 or    protocol version 15
                              SCCP6911.9-3-1-SR2-3 protocol version 15

Comments: 
I have a pair of 6911 telephone if you need me to get some packet captures for the feature just let me know.

marcelloceschia, 2016-09-28:
Hello @voipAsteriskGuy 

I do not have a 6911 to reproduce this issue locally. Can you send me some cli logs from asterisk (with `sccp debug channel device indicate message`) at the stage where you have answered the call.

voipAsteriskGuy, 2016-09-28:
Hi @marcelloceschia 

I uploaded the file bug_6911.txt to the logs i send to you are using the  debug option with
 `sccp debug channel device indicate message`
and asterisk verbose level 3 
i get the first log searching for the status of CONNECTED for the extension 57114 wich is the phone 6911

let my know if you need more information.

[bug_6911.txt](https://github.com/chan-sccp/chan-sccp/files/499344/bug_6911.txt)

marcelloceschia, 2016-09-29:
Hi @voipAsteriskGuy 

i have check the debug log. Is it right, that the phone hangs up immediately after pressing transfer?

```
[2016-09-28 16:45:09] VERBOSE[23918] sccp_actions.c:     -- SEP0008308B0D5B: Got Softkey: Transfer (4) line=1 callid=3
[2016-09-28 16:45:12] VERBOSE[23918] sccp_actions.c:     -- SEP0008308B0D5B: >> Got message On-Hook Message (0x7)
[2016-09-28 16:45:12] NOTICE[23918] sccp_actions.c: SEP0008308B0D5B: SCCP Handle Message: On-Hook Message(0x0007) 20 bytes length
[2016-09-28 16:45:12] VERBOSE[23918] sccp_channel.c:   == SEP0008308B0D5B: Ending call SCCP/57114-00000004 (state:OFFHOOK)
```

voipAsteriskGuy, 2016-09-29:
hi @marcelloceschia the phone does not hangs i pressed the transfer key and ear the tone, after that i pressed some buttons to dial a extension and still earing the tone, so i hanged the call.

voipAsteriskGuy, 2016-09-30:
Hi @marcelloceschia I was doing more test with this phone, and think the problem could be something related to RTP, when I make a call from a 6911 phone to another phone I have a one way audio issue, I have captured some pcap files involving the 6911 phone, one is from a Phone calling to the 6911 and trying to do a transfer and the other one is a call between a phone and the 6911.

I analyzed them using wireshark and get a Malformed Packet: SKINNY in both cases
on the pcap file with the name 6911_call you can check it at sequence number 20 
on the pcap file with the name 6911_transfer you can check it at sequence number 20 

6911_call.zip](https://github.com/chan-sccp/chan-sccp/files/503622/6911_call.zip)

dkgroot, 2016-11-20:
Hi @voipAsteriskGuy 

Looking at the tcpdump / wireshark dump shows that we are never receiving any dialed digits after transfer has been initiated. I suspect that the phone has entered an invalid channelstate/keymode preventing it to send any digits (because it is not expecting that it should, in the state that its in). The 69xx series behaves quite a bit different than all other sccp phones to do, in this respect.

We would have to do some annoying remote debugging or require local access to one of these phones (meaning buy one), to be able to figure out what's going on. The 6911 not having a display does not help either in this respect because you cannot tell us the current state either.

dkgroot, 2017-02-25:
Related Revision develop: e6fd157
Please pull the latest changes, an recheck

We are sending SelectSoftKeys to the transferee call, after we just started a new outbound (transferer) call, this might be the cause of the Focus to switch back to the first call, which bad side effects.

I found the location in source that caused this, and fixed it in revision: 'e6fd157'
I also added some code to make the line lamp blink, while you are in a transfer, so that you can see that you are doing something specific.

Note: Do make sure you have (in you SEP...cnf.xml):
```xml
<vendorConfig>...<autoSelectLineEnable>0</autoSelectLineEnable><autoCallSelect>0</autoCallSelect></vendorConfig>
```

BTW:
I Started a little project, to cleanly rewrite the whole transfer process in a separate branch called feature/transfer. Would you mind helping out to test/check this (test environment only at this moment).
```
git checkout feature/transfer
./configure [... your configure flags...]
make -j2 && make reload
```
Mind you, i am still working on getting the callinfo correct, so please do not pay attention to that to much.

This new version, will also get rid of the above requirement for autoSelectLineEnable + autoCallSelect :-)

dkgroot, 2017-02-28:
@voipAsteriskGuy : Issues regarding 6901 and the audio fixes related to that issue, should also take care of this issue. The issues/381 branch was merged back into the develop branch. So you can just perform an `git pull` of develop and recheck.
Closing this issue (on this assumption), please re-open if there still is an issue with the 6911.

commit d106aeb0 (2017-02-28): Enh: Remove commented out lines (regarding 6901)

Fixes #381
Fixes #335
closed 2017-02-28 by dkgroot