chan-sccp/chan-sccp #339: Ringin or Ringout not showing in CTI with TAPI driver

technorocka, 2016-10-18:
Hi,

there is a Problem with the SCCP version 4.3.0 and asterisk 13 (FreePBX Distro maybe others, too?).

The Tapi driver (Activa TSP) isn't getting the ringin or ringout State. 
The same happened at FOP2 (Flash operator panel). 
So you can't get missed calls or see who is calling you on the PC CTI.

The hint status in CLI is working perfectly. 
Idle -> Ringing -> InUse

I don't know, if this is the same problem ( https://sourceforge.net/p/chan-sccp-b/mailman/message/33283935/ )?

**Reproduction Steps:**
1. Phone A calls SCCP Phone B
2. SCCP Phone B calls Phone A

**Expected behavior:**
Show the number of the caller or the dialed number in CTI

**Observed behavior:**
Only after picking up the phone, the CTI is showing the number and the line as busy.
SIP Phones works without any problems.

dkgroot, 2016-11-18:
This might be a hint / devicestate / channelstate issue. Marcello and I both have a little difficulty reproducting TAPI related issues, because neither of us has a windows installation available.

dkgroot, 2016-12-27:
@technorocka 
Some questions
- Is this issue still present ?
- Does Activa TSP actually support asterisk-13 ? From their website it looks like its only supported up to asterisk-1.8
- Does the Activa TSP debug log give any extra information ?
- I guess they connect using an AMI/manager connection, that means they are not looking at the hint information directly, but at manager events. We send some extra CID information to these events, that they might not be able to parse. 

An example of SCCP/98031 calling SCCP/98011 generates these kinds of events
```
Event: DeviceStateChange
Privilege: call,all
SequenceNumber: 1401
File: manager.c
Line: 1817
Func: manager_default_msg_cb
Device: SCCP/98011
State: RINGING

Event: ExtensionStatus
Privilege: call,all
SequenceNumber: 1402
File: manager.c
Line: 6957
Func: manager_state_cb
Exten: 98011
Context: hints
Hint: SCCP/98011
Status: 8
StatusText: Ringing

Event: Newstate
Privilege: call,all
SequenceNumber: 1404
File: manager_channels.c
Line: 734
Func: channel_snapshot_update
Channel: SCCP/98011-00000006
ChannelState: 5
ChannelStateDesc: Ringing
CallerIDNum: 98011
CallerIDName: Diederik-Phone1
ConnectedLineNum: 98031
ConnectedLineName: Diederik-Phone3
Language: en
AccountCode: 98031
Context: internal
Exten: 98011
Priority: 1
Uniqueid: 1482831964.19
Linkedid: 1482831964.18

Event: DeviceStateChange
Privilege: call,all
SequenceNumber: 1406
File: manager.c
Line: 1817
Func: manager_default_msg_cb
Device: SCCP/98031
State: RINGINUSE

Event: ExtensionStatus
Privilege: call,all
SequenceNumber: 1409
File: manager.c
Line: 6957
Func: manager_state_cb
Exten: 98031
Context: hints
Hint: SCCP/98031
Status: 9
StatusText: InUse&Ringing
```
You would need to look at the Activa TSP source code to see which of these events they have problems with. You should see these events in their debug.log as well (i hope).

- Could it be that CTI cannot deal with combined states like "InUse&Ringing" (We set the hint to this state when we dial out and the other side is ringing)

dkgroot, 2017-01-13:
Closing issue report, no further reply from issuer.

closed 2017-01-13 by dkgroot