chan-sccp/chan-sccp #340: Sometimes can't dial a number from Cisco 7912

jozzus, 2016-10-19:
Sometimes can't dial a number from Cisco 7912

**Reproduction Steps:**

1. Get Offhook
2. try to dial any number

**Expected behavior:**

It must be dialed.

**Observed behavior:**

The number is not dialed.
In Asterisk CLI:

    --     -- SEP001B5434F07C: Taken Offhook
    --     -- SEP001B5434F07C: Using line 77212
SEP001B5434F07C: (getEmptyChannel) on line:77212, maybe_c:
SEP001B5434F07C: (getEmptyChannel) getActiveChannel
[2016-10-19 17:09:56] NOTICE[28110]: ast113.c:1004 sccp_wrapper_asterisk113_allocPBXChannel: SCCP: (allocPBXChannel) Create New Channel with name: SCCP/77212-00000071
    --     -- SCCP/77212-00000071: schedule digittimeout 45
    --     -- SEP001B5434F07C: SCCP (handle_keypad) digit:00000007, callid:60, buttonIndex:1 => lineInstance:1
[2016-10-19 17:09:58] NOTICE[28110]: sccp_actions.c:2912 handle_keypad_button: SEP001B5434F07C: Device sent a Keypress, but there is no (active) channel! Exiting
    --     -- SEP001B5434F07C: SCCP (handle_keypad) digit:00000000, callid:60, buttonIndex:1 => lineInstance:1
[2016-10-19 17:09:58] NOTICE[28110]: sccp_actions.c:2912 handle_keypad_button: SEP001B5434F07C: Device sent a Keypress, but there is no (active) channel! Exiting
    --     -- SEP001B5434F07C: SCCP (handle_keypad) digit:00000007, callid:60, buttonIndex:1 => lineInstance:1
[2016-10-19 17:09:59] NOTICE[28110]: sccp_actions.c:2912 handle_keypad_button: SEP001B5434F07C: Device sent a Keypress, but there is no (active) channel! Exiting
    --     -- SEP001B5434F07C: SCCP (handle_keypad) digit:00000009, callid:60, buttonIndex:1 => lineInstance:1
[2016-10-19 17:09:59] NOTICE[28110]: sccp_actions.c:2912 handle_keypad_button: SEP001B5434F07C: Device sent a Keypress, but there is no (active) channel! Exiting
    --     -- SEP001B5434F07C: SCCP (handle_keypad) digit:00000000, callid:60, buttonIndex:1 => lineInstance:1
[2016-10-19 17:10:00] NOTICE[28110]: sccp_actions.c:2912 handle_keypad_button: SEP001B5434F07C: Device sent a Keypress, but there is no (active) channel! Exiting

**chan-sccp version:** Release: 4.3.0 TRUNK - 7a9996bM

**asterisk version:** Asterisk 13.11.2

**Addition / Optional Information:**
 It happenned randomly

* Platform:
    * [CPU]
    * [64 Bit]
    *Centos 7
* Comments: [Other Information]

marcelloceschia, 2016-11-18:
@jozzus can you send us a full debug in this case. 
```
sccp debug core acction channel device line
```

dkgroot, 2016-11-19:
(0e64a1e Tested with 7936 on my local system)

@marcelloceschia 
Can you have a look at the  0e64a1e 
- To check if you like this better than the previous if/then/else tree
- If this works correctly on your 7912

@jozzus 
Can you verify if the issue stil occurs with this branch : 0e64a1e as well

jozzus, 2016-11-23:
It doesn't happens anymore. Thank you!

dkgroot, 2016-11-23:
Thanks for reporting and updating the issue. Closing it now.

closed 2016-11-23 by dkgroot