chan-sccp/chan-sccp #344: Problem with SIP NOTIFY in SIP phones when chan_sccp is activated

miguelsanzpardo, 2016-11-15:
When chan_sccp module is activated the SIP NOTIFY sent by Asterisk between SIP phones are incorrect.
When chan_sccp module is deactivated the SIP NOTIFY sent by Asterisk between SIP phones are correct.

**Reproduction Steps:**
1004 calls to 1001
1005 has BLF which monitorised both: 1004 and 1001
NOTE: In my test I am using Grandstream phones (2130,2160 and 2170). I am going to install several GS phones on the same place where I have about 400 CISCO phones with chan_sccp.

**Expected behavior:**
1005 should show on the BLF: From 1004 To 1001

NOTIFY sip:1005@192.168.1.169:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.1.41:5060;branch=z9hG4bK2d1bcc18;rport
Max-Forwards: 70
From: <sip:1001@192.168.1.40>;tag=as392d3a2d
To: <sip:1005@192.168.1.40>;tag=60618332
Contact: <sip:1001@192.168.1.41:5060>
Call-ID: 801912290-5060-4@BJC.BGI.B.BGJ
CSeq: 130 NOTIFY
User-Agent: FPBX-12.0.76.4(11.21.2)
Subscription-State: active
Event: dialog
Content-Type: application/dialog-info+xml
Content-Length: 544

<?xml version="1.0"?>
<dialog-info xmlns="urn:ietf:params:xml:ns:dialog-info" version="28" state="full" entity="sip:1001@192.168.1.40">
<dialog id="1001" call-id="pickup-801912290-5060-4@BJC.BGI.B.BGJ" local-tag="60618332" remote-tag="as392d3a2d" direction="recipient">
<remote>
<identity display="1001">sip:1001@192.168.1.40</identity>
<target uri="sip:1001@192.168.1.40"/>
</remote>
<local>
<identity display="1004">sip:1004@192.168.1.40</identity>
<target uri="sip:1004@192.168.1.40"/>
</local>
<state>early</state>
</dialog>
</dialog-info>

**Observed behavior:**
1005 show on the BLF: From 1001 To 1001

NOTIFY sip:1005@192.168.1.169:5060 SIP/2.0
Via: SIP/2.0/UDP 192.168.1.41:5060;branch=z9hG4bK2d1bcc18;rport
Max-Forwards: 70
From: <sip:1001@192.168.1.40>;tag=as392d3a2d
To: <sip:1005@192.168.1.40>;tag=60618332
Contact: <sip:1001@192.168.1.41:5060>
Call-ID: 801912290-5060-4@BJC.BGI.B.BGJ
CSeq: 130 NOTIFY
User-Agent: FPBX-12.0.76.4(11.21.2)
Subscription-State: active
Event: dialog
Content-Type: application/dialog-info+xml
Content-Length: 544

<?xml version="1.0"?>
<dialog-info xmlns="urn:ietf:params:xml:ns:dialog-info" version="28" state="full" entity="sip:1001@192.168.1.40">
<dialog id="1001" call-id="pickup-801912290-5060-4@BJC.BGI.B.BGJ" local-tag="60618332" remote-tag="as392d3a2d" direction="recipient">
<remote>
<identity display="1001">sip:1001@192.168.1.40</identity>
<target uri="sip:1001@192.168.1.40"/>
</remote>
<local>
<identity display="1001">sip:1001@192.168.1.40</identity>
<target uri="sip:1001@192.168.1.40"/>
</local>
<state>early</state>
</dialog>
</dialog-info>

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)

**chan-sccp version:** I have test this with the last chan_sccp develop version and with chan_sccp 4.2.0.6168 and the problem appears with both

**asterisk version:** I have test it with several Asterisk versions: 11.17.0, 11.21.2, 11.24.1

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: No, the problem came from 4.2.0.6168 or even before
* Problem can be reliably reproduced, doesn't happen randomly: Yes, I have doing the test with two FreePBX distros and with one Asterisk Vanilla (Asterisk without web interface) and the problem appears in all the cases if chan_sccp is activated.

* Platform:
Debian 8.3, 64 bits.

NOTE:
It sounds quite strange that chan_sccp is affecting the SIP NOTIFY send by Asterisk but it is happening, I think that you can reproduce the issue without problems.

marcelloceschia, 2016-11-20:
@miguelsanzpardo is 1004 a sip or an sccp extension?

miguelsanzpardo, 2016-11-20:
All the extensions of the example are sip extensions. If chan sccp is activated (having or not sccp extensions registered on asterisk)  the problem exist, if chan sccp is deactivated the problem disappear.

dkgroot, 2016-11-20:
@miguelsanzpardo : That is very very strange behaviour, and does not sound logical to me. That would mean we are changing channelstates / devicestates for devices that are not linked/monitored to/by chan-sccp at all. I don't even know where to start investigating that behaviour.
Can you check that with an empty sccp.conf file, to make sure we are not monitoring any of the sip phones.
@marcelloceschia : do you have any idea how that might be possible ?

miguelsanzpardo, 2016-11-21:
Hi again,

I have tested with an empty sccp.conf file and the problem doesnt dissappear.

sccp.conf:
[general]
servername = Asterisk

CLI> sccp show globals

--- SCCP channel driver global settings ----------------------------------------------------------------------------------
Config File                                    : sccp.conf
Platform byte order                            : LITTLE ENDIAN
Server Name                                    : Asterisk
Bind Address                                   : 0.0.0.0:2000
Extern IP                                      : Not Set -> falling back to Incoming Interface IP-addres (expect issue if running natted !).
Localnet                                       : permit:127.0.0.0/255.0.0.0,permit:10.0.0.0/255.0.0.0,permit:172.0.0.0/255.224.0.0,permit:192.168.0.0/255.255.0.0,
Deny/Permit                                    : deny:0.0.0.0/0.0.0.0,permit:127.0.0.0/255.0.0.0,permit:10.0.0.0/255.0.0.0,permit:172.0.0.0/255.224.0.0,permit:192.168.0.0/255.255.0.0,
Direct RTP                                     : off
Nat                                            : Auto
Keepalive                                      : 60
Debug                                          : (1) core
Date format                                    : M/D/Y
First digit timeout                            : 16
Digit timeout                                  : 8
Digit timeout char                             : #
SCCP tos (signaling)                           : 104
SCCP cos (signaling)                           : 4
AUDIO tos (rtp)                                : 184
AUDIO cos (rtp)                                : 6
VIDEO tos (vrtp)                               : 136
VIDEO cos (vrtp)                               : 5
Context                                        : default (exists)
Language                                       : en
Accountcode                                    : skinny
Musicclass                                     : default
AMA flags                                      : 0 (Unknown)
Callgroup                                      :
Pickupgroup                                    :
Pickup Mode Answer                             : on
Codecs preference                              : [ulaw/64k, ulaw/56k, alaw/64k, alaw/56k]
CFWDALL                                        : on
CFWBUSY                                        : on
CFWNOANSWER                                    : on
Call Events                                    : on
DND Feature enabled                            : on
Park                                           : off
Private softkey                                : on
Echo cancel                                    : on
Silence suppression                            : off
Trust phone ip (deprecated)                    : off
Early RTP                                      : Progress
AutoAnswer ringtime                            : 1
AutoAnswer tone                                : 50
RemoteHangup tone                              : 50
Transfer tone                                  : 0
Transfer on hangup                             : off
Callwaiting tone                               : 45
Callwaiting interval                           : 0
Registration Context                           : Unset
Jitterbuffer enabled                           : off
Jitterbuffer forced                            : off
Jitterbuffer max size                          : -1
Jitterbuffer resync                            : -1
Jitterbuffer impl                              :
Jitterbuffer log                               : off
Token FallBack                                 : no
Token Backoff-Time                             : 60
Hotline_Enabled                                : on
Hotline_Context                                : sccp
Hotline_Exten                                  :
Threadpool Size                                : 0/2

sccp show lines:
CLI> sccp show lines

+--- Lines ------------------------------------------------------------------------------------------------------------------------------------------------------+
| Ext           Suffix    Label                          Description      Device           MWI  Chs  Active Channel                                              |
+ ============= ========= ============================== ================ ================ ==== ==== =========================================================== +
| Hotline                 (null)                         (null)           --               OFF  0    --         --                                               |
+----------------------------------------------------------------------------------------------------------------------------------------------------------------+

sccp show devices:
CLI> sccp show devices

+--- Devices --------------------------------------------------------------------------------------------------------------------------------------------+
| Descr                     Address                                      Mac              RegState   Token RegTime                   Act Lines Nat       |
+ ========================= ============================================ ================ ========== ===== ========================= === ===== ========= +
+--------------------------------------------------------------------------------------------------------------------------------------------------------+

I have made the test with 3 distributions and have the same problem with all when module chan_sccp.so is loaded. 
- Centos 6.7 + Asterisk 11.17 (without FreePBX or another web interface). I have recompiled to 11.17.0, 11.21.2 and 11.24.1 and the problem doesnt disappear.
- FreePBX Distro (with Asterisk 11.21.2)
- Debian 8.3 + Asterisk 11.21.2 + FreePBX module

Do you have 3 SIP phones(Grandstream, Yealink, Astra,etc) to simulate the problem? It is very strange but it is happening and can affect to SIP phones which are on the same place as SCCP phones.

marcelloceschia, 2016-11-21:
@miguelsanzpardo and is the hint for 1001 and 1004 defined? `core show hints`, `core show hint 1001`, `core show hint 1004`

marcelloceschia, 2016-11-21:
@miguelsanzpardo Do you use the same line name for sccp and sip?

miguelsanzpardo, 2016-11-21:
```
CLI> core show hints

    -= Registered Asterisk Dial Plan Hints =-
                   1004@from-internal-gs    : SIP/1004              State:Idle            Watchers  2
                   1005@from-internal-gs    : SIP/1005              State:Idle            Watchers  2
                   1001@from-internal-gs    : SIP/1001              State:Idle            Watchers  3
                   1008@from-internal-gs    : SIP/1008              State:Idle            Watchers  2
----------------

CLI> core show hint 1001
                   1001@from-internal-gs    : SIP/1001              State:Idle            Watchers  3
1 hint matching extension 1001

CLI> core show hint 1004
                   1004@from-internal-gs    : SIP/1004              State:Idle            Watchers  2
1 hint matching extension 1004

CLI> sip show settings
Global Settings:
----------------
  UDP Bindaddress:        192.168.1.40:5060
  TCP SIP Bindaddress:    Disabled
  TLS SIP Bindaddress:    Disabled
  Videosupport:           No
  Textsupport:            No
  Ignore SDP sess. ver.:  No
  AutoCreate Peer:        Off
  Match Auth Username:    No
  Allow unknown access:   Yes
  Allow subscriptions:    Yes
  Allow overlap dialing:  No
  Allow promisc. redir:   No
  Enable call counters:   Yes
  SIP domain support:     No
  Realm. auth:            No
  Our auth realm          asterisk
  Use domains as realms:  No
  Call to non-local dom.: Yes
  URI user is phone no:   No
  Always auth rejects:    Yes
  Direct RTP setup:       No
  User Agent:             Asterisk PBX 11.21.2
  SDP Session Name:       Asterisk PBX 11.21.2
  SDP Owner Name:         root
  Reg. context:           (not set)
  Regexten on Qualify:    No
  Trust RPID:             Yes
  Send RPID:              Yes
  Legacy userfield parse: No
  Send Diversion:         Yes
  Caller ID:              asterisk
  From: Domain:
  Record SIP history:     Off
  Call Events:            Off
  Auth. Failure Events:   Off
  T.38 support:           No
  T.38 EC mode:           Unknown
  T.38 MaxDtgrm:          4294967295
  SIP realtime:           Disabled
  Qualify Freq :          60000 ms
  Q.850 Reason header:    No
  Store SIP_CAUSE:        No

Network QoS Settings:
---------------------------
  IP ToS SIP:             CS0
  IP ToS RTP audio:       CS0
  IP ToS RTP video:       CS0
  IP ToS RTP text:        CS0
[... 96 lines cut ...]
  Generic PLC:                 Enabled
  Min DTMF duration::          80

* Subsystems
  -------------
  Manager (AMI):               Disabled
  Web Manager (AMI/HTTP):      Disabled
  Call data records:           Enabled
  Realtime Architecture (ARA): Disabled

* Directories
  -------------
  Configuration file:
  Configuration directory:     /etc/asterisk
  Module directory:            /usr/lib64/asterisk/modules
  Spool directory:             /var/spool/asterisk
  Log directory:               /var/log/asterisk
  Run/Sockets directory:       /var/run/asterisk
  PID file:                    /var/run/asterisk/asterisk.pid
  VarLib directory:            /var/lib/asterisk
  Data directory:              /var/lib/asterisk
  ASTDB:                       /var/lib/asterisk/astdb
  IAX2 Keys directory:         /var/lib/asterisk/keys
  AGI Scripts directory:       /var/lib/asterisk/agi-bin
```

marcelloceschia, 2016-11-21:
@miguelsanzpardo can you try somthin on the problematic asterisk box?

please change on https://github.com/chan-sccp/chan-sccp/blob/develop/src/sccp_hint.c#L1327
```
#if ASTERISK_VERSION_GROUP >= 108
```
into
```
#if 0
```
recompile chan_sccp and load the module again. Does this change has any effect?
Thank you

miguelsanzpardo, 2016-11-21:
On what code line?
163? 1327? both?

By the way, what sccp debug options should activate to help us to find the problem?

marcelloceschia, 2016-11-21:
1327 see the link above

miguelsanzpardo, 2016-11-21:
I have test it and the problem still appears, what sccp debug flags should activate to analyse this problem?

miguelsanzpardo, 2016-11-21:
I have test another thing more, if I register one sccp extensions (for example 1104 extension) and call to one sip extension (for example 1001) --> I can see on BLF (1001 is being monitorised on 1004 and 1005) that 1001 is calling to 1001 instead of seeing that 1104 is calling to 1001.

I share with you a share of sccp debug all (138 KB):
[full.txt](https://github.com/chan-sccp/chan-sccp/files/603900/full.txt)

...
extensions.conf:
[from-internal-gs]
exten => _10XX,1,NoOp()
same  => n,Dial(SIP/${EXTEN},60,)
same  => n,Hangup()

exten => _11XX,1,NoOp()
same  => n,Dial(SCCP/${EXTEN},60,)
same  => n,Hangup()

exten  => 1001,hint,SIP/1001
exten  => 1004,hint,SIP/1004
exten  => 1005,hint,SIP/1005
exten  => 1008,hint,SIP/1008
exten  => 1104,hint,SCCP/1104

; Capturas de llamada:
; Mediante ** + nยบ extension
exten => _**1XXX,1,Pickup(${EXTEN:2}@from-internal-gs)
same  => n,Hangup()
...

marcelloceschia, 2016-11-21:
@miguelsanzpardo thank you for the detailed log, just a question. Is this a general issue or does this only occur if SCCP starts the call? Does this also occur if 1001 calls 1104 ? (just searching for a starting point ;) )

miguelsanzpardo, 2016-11-21:
With chan_sccp.so activated:

If 1005 (SIP) is monitoring:
- 1001 (SIP)
- 1004 (SIP)
- 1104 (SCCP)

Then:
- If 1004(SIP) calls to 1001 (SIP) the problem appear --> 1005 is watching:  1001 is calling to 1001
- If 1104(SCCP) calls to 1001 (SIP) the problem appears --> 1005 is watching:  1001 is calling to 1001
- If 1004(SIP) calls to 1104 (SCCP) the problem doesnt appear --> 1005 is watching: 1004 is calling to 1104

marcelloceschia, 2016-11-21:
@dkgroot should we uncomment the whole hint module to see if this issue is within hint or somewhere else?
One way is to comment everything in sccp_hint_module_start() function

miguelsanzpardo, 2016-11-21:
Should I comment:

void sccp_hint_module_start(void)
{
	sccp_log((DEBUGCAT_CORE)) (VERBOSE_PREFIX_2 "SCCP: Starting hint system\n");
	SCCP_LIST_HEAD_INIT(&lineStates);
	SCCP_LIST_HEAD_INIT(&sccp_hint_subscriptions);
	sccp_event_subscribe(SCCP_EVENT_DEVICE_REGISTERED | SCCP_EVENT_DEVICE_UNREGISTERED | SCCP_EVENT_DEVICE_DETACHED | SCCP_EVENT_DEVICE_ATTACHED | SCCP_EVENT_LINESTATUS_CHANGED, sccp_hint_eventListener, TRUE);
	sccp_event_subscribe(SCCP_EVENT_FEATURE_CHANGED, sccp_hint_handleFeatureChangeEvent, TRUE);
#ifdef CS_USE_ASTERISK_DISTRIBUTED_DEVSTATE
	ast_eid_to_str(default_eid_str, sizeof(default_eid_str), &ast_eid_default);
#endif
}

and then compile and try again?

marcelloceschia, 2016-11-21:
do not comment the signature of the function
```
void sccp_hint_module_start(void)
{
	//sccp_log((DEBUGCAT_CORE)) (VERBOSE_PREFIX_2 "SCCP: Starting hint system\n");
	//SCCP_LIST_HEAD_INIT(&lineStates);
	//SCCP_LIST_HEAD_INIT(&sccp_hint_subscriptions);
	//sccp_event_subscribe(SCCP_EVENT_DEVICE_REGISTERED | SCCP_EVENT_DEVICE_UNREGISTERED | SCCP_EVENT_DEVICE_DETACHED | SCCP_EVENT_DEVICE_ATTACHED | SCCP_EVENT_LINESTATUS_CHANGED, sccp_hint_eventListener, TRUE);
	//sccp_event_subscribe(SCCP_EVENT_FEATURE_CHANGED, sccp_hint_handleFeatureChangeEvent, TRUE);
#ifdef CS_USE_ASTERISK_DISTRIBUTED_DEVSTATE
	//ast_eid_to_str(default_eid_str, sizeof(default_eid_str), &ast_eid_default);
#endif
}
```

miguelsanzpardo, 2016-11-21:
I have comment:
void sccp_hint_module_start(void)
{
/* 	sccp_log((DEBUGCAT_CORE)) (VERBOSE_PREFIX_2 "SCCP: Starting hint system\n");
	SCCP_LIST_HEAD_INIT(&lineStates);
	SCCP_LIST_HEAD_INIT(&sccp_hint_subscriptions);
	sccp_event_subscribe(SCCP_EVENT_DEVICE_REGISTERED | SCCP_EVENT_DEVICE_UNREGISTERED | SCCP_EVENT_DEVICE_DETACHED | SCCP_EVENT_DEVICE_ATTACHED | SCCP_EVENT_LINESTATUS_CHANGED, sccp_hint_eventListener, TRUE);
	sccp_event_subscribe(SCCP_EVENT_FEATURE_CHANGED, sccp_hint_handleFeatureChangeEvent, TRUE); */
#ifdef CS_USE_ASTERISK_DISTRIBUTED_DEVSTATE
	/* ast_eid_to_str(default_eid_str, sizeof(default_eid_str), &ast_eid_default); */
#endif
}

and compile again, but the problem still appears.

dkgroot, 2016-11-21:
@marcelloceschia 
Could also be related to devstate (active in changing state even without hint being monitored) and/or registration context (small chance / not sure how)

marcelloceschia, 2016-11-21:
@miguelsanzpardo @dkgroot I think we should start commenting all "modules" in https://github.com/chan-sccp/chan-sccp/blob/develop/src/chan_sccp.c#L184 

```
// sccp_devstate_module_start();
// sccp_mwi_module_start();
// sccp_hint_module_start();
// sccp_manager_module_start();
// sccp_conference_module_start();
```

I hope one of them solves this issue and we can start searching for the solution :(

miguelsanzpardo, 2016-11-21:
Commented and compiled, the problem still appears :(

dkgroot, 2016-11-21:
As far as i know you also have a vanilla asterisk machine (without freepbx). Would be a lot easier to reason about, without freepbx changing states as well. Can you rebuild the same situation there ?

miguelsanzpardo, 2016-11-21:
I am doing the test with an Asterisk Vanilla version since this morning Diederik.

Although at the beggining of the issue you can see on the sip NOTIFY: User-Agent: FPBX-12.0.76.4(11.21.2) I have made all the test of today from an Asterisk Vanilla version:
- Centos 6.7 + Asterisk 11.21.2

dkgroot, 2016-11-21:
Ooh ok, good to know. Thanks for the update.
It's a difficult issue, which is hard to wrap your head around. We have tried most options so far.

@marcelloceschia : commenting out the same entries in my local system prevent all hint/devstate interaction on my system. So they should not be able to cause any of this issue.

miguelsanzpardo, 2016-11-21:
Diederik, do you have sip phones to simulate the same scenario?

dkgroot, 2016-11-21:
Hi Miguel,
I only have one SNOM which uses this type of BLF (chan_sip supports multiple sip devices, which use different encodings for blf messages). Currently this device is not connected. So having a hard time simulating this behaviour. Marcello has several phones which use this encoding, but will not be able to do much testing until at least next weekend.

miguelsanzpardo, 2016-11-22:
Umh, can we do anything more to try to solve this problem meanwhile or should wait until Marcello can simulate the same behaviour on his system? By the way, did you take a look of the full.txt that I shared?

marcelloceschia, 2016-11-22:
@miguelsanzpardo we checked the full.txt but chan_sccp has no debugging for this issue, because everything for notifycid for sip is done in chan_sip or asterisk and it uses channels to get this data

miguelsanzpardo, 2016-11-22:
This sounds like is going to be very hard to analyse and solve, true? :)

marcelloceschia, 2016-11-22:
@miguelsanzpardo I will check this on sunday, I need to prepare the SIP devices

miguelsanzpardo, 2016-11-29:
I Marcello, could you do a test?

miguelsanzpardo, 2016-12-12:
Hi again, any news about this issue? Marcelo, did you can test It? On my installation i have sccp and sip phones and this problem is a little frustrating. If i can help you with the test in any way say me please.

dkgroot, 2016-12-27:
Note chan_sip.c source states:

> It may seem odd to base the remote_target on the To header here, but testing by reporters on issue ASTERISK-16735 found that basing on the From header would cause ringing state hints to not work properly on certain SNOM devices. If you are using notifycid properly (i.e. in the same extension and context as the dialed call) then this should not be an issue since the data will be overwritten shortly with channel caller ID

Sip.conf has this entry:
```
notifycid = yes                ; Control whether caller ID information is sent along with
                                ; dialog-info+xml notifications (supported by snom phones).
                                ; Note that this feature will only work properly when the
                                ; incoming call is using the same extension and context that
                                ; is being used as the hint for the called extension.  This means
                                ; that it won't work when using subscribecontext for your sip
                                ; user or peer (if subscribecontext is different than context).
                                ; This is also limited to a single caller, meaning that if an
                                ; extension is ringing because multiple calls are incoming,
                                ; only one will be used as the source of caller ID.  Specify
                                ; 'ignore-context' to ignore the called context when looking
                                ; for the caller's channel.  The default value is 'no.' Setting
                                ; notifycid to 'ignore-context' also causes call-pickups attempted
                                ; via SNOM's NOTIFY mechanism to set the context for the call pickup
                                ; to PICKUPMARK.
```
This could mean that overlapping sip and sccp extensions in a single context would cause these issues. The method they used to match extension+context to the hint is faulty at best. They should at least make this change: after finding a potential candidate (using the to match) the should lookup the hint and track back if it belongs to the 'from'.

closed 2019-12-17 by dkgroot