chan-sccp/chan-sccp #346: Making a "sccp reload" unregister the phones

miguelsanzpardo, 2016-11-22:
If I do changes on sccp.conf and execute "sccp reload" the SCCP phones are unregistered and register again.

**Reproduction Steps:**
Modify sccp.conf:

Incially:
[general]
servername = Asterisk
; keepalive = 60
; debug = 1
; context = default
; dateformat = D.M.Y
; bindaddr = 0.0.0.0
; port = 2000
; disallow=all
; allow=alaw
...

Finally:
[general]
servername = Asterisk
; keepalive = 60
; debug = 1
; context = default
; dateformat = D.M.Y
; bindaddr = 0.0.0.0
; port = 2000
disallow=all
allow=alaw

**Expected behavior:**
The changes take effect on sccp configuration with no problems.

**Observed behavior:**
It appears this error and suddenly all the phones deregister and register again:

ERROR[26066]: sccp_session.c:206 socket_get_error: SCCP: (sccp_session.c:803:sccp_netsock_device_thread) Socket returned error: 'No such file or directory (2)')

**chan-sccp version:** Skinny Client Control Protocol (SCCP). Release: 4.3.0 develop - c7dae08M 
**asterisk version:** 11.21.2

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: Not sure
* Problem can be reliably reproduced, doesn't happen randomly: Yes, it can be reproduced
* Platform:
    bedian 8.3

**Logs**
On this example I have donde 2 sccp reload:
1º) I have commented disallow=all and allow=alaw and then I have executed "sccp reload"
The changes has taken effect without ptoblems.
2º) I have uncommented disallow=all and allow=alaw and then I have executed "sccp reload"
The changes has taken effect but the phones have been unregistered and registered again
[full-unregister_on reload.txt](https://github.com/chan-sccp/chan-sccp/files/606146/full-unregister_on.reload.txt)

[sccp.conf.txt](https://github.com/chan-sccp/chan-sccp/files/606155/sccp.conf.txt)

dkgroot, 2016-11-22:
Changing the allowed codecs should not require a restart (agreed). This particular codec change should not happen very regulary, so it's a minor issue.

dkgroot, 2016-11-22:
Running the config change with 'debug = core, config, high', shows the reasoning behind the reset
```
  == Parsing '/usr/local/asterisk-13-branch/etc/asterisk/sccp.allow.conf': Found
[Nov 22 13:09:41] NOTICE[2184]: sccp_config.c:2703 sccp_config_file_status_t sccp_config_getConfig(boolean_t): Config file '/usr/local/asterisk-13-branch/etc/asterisk/sccp.allow.conf' loaded.
    --     -- SCCP: parsing general parameter: servername = 'Asterisk'  (line: 2)
    --     -- SCCP: (sccp_config_object_setValue) Set Entry[0] = TRUE for servername
    --     -- SCCP: parsing general parameter: debug = 'core, config, high'  (line: 3)
    --     -- SCCP: (sccp_config_object_setValue) Set Entry[2] = TRUE for debug
    --     -- SCCP: parsing general parameter: disallow = 'all'  (line: 10)
sccp_config_parse_codec preference: name: disallow, value:all
sccp_config_parse_codec preference: name: allow, value:alaw
  ==   == SCCP: config parameter disallow='all' in line 10 changed. (causes device reset)
    --     -- SCCP: (sccp_config_object_setValue) Set Entry[7] = TRUE for disallow|allow
    --     -- SCCP: parsing general parameter: allow = 'alaw'  (line: 11)
    --     -- SCCP: (sccp_config_object_setValue) Set Entry[7] = TRUE for MultiEntry disallow|allow -> SKIPPING
  ==   == Update Needed (1)
```
The Error Message
...
ERROR[26066]: sccp_session.c:206 socket_get_error: SCCP: (sccp_session.c:803:sccp_netsock_device_thread) Socket returned error: 'No such file or directory (2)')
...
Is not really an ERROR, but was added in the time when we were investigating session issues. It signals that the phone has closed the socket from it's side and we need to take action to tear down the connection on our side. We can lower the error status later on, once the open issues regarding cross device connection and loss of connectivity have been investigated.

Not taking action to change config parsing at this moment. Codec changes in the [general] section do require a device restart to make sure the new codec get's parsed into the device sections, if they don't have one set. So the action to reset all devices in this occasion makes sense.

closed 2016-11-22 by dkgroot

miguelsanzpardo, 2016-11-22:
It occurs the same If I add:

deny=0.0.0.0/0.0.0.0
permit=192.168.1.0/255.255.255.0

and execute a sccp reload.

Is this behaviour correct?