chan-sccp/chan-sccp #347: Missing wiki page on BLF?

jrodder, 2016-11-22:
Not an issue with the module itself, but really trying to work on some BLF blinking when in a queue and another extension is ringing. (To allow a ringing plan of one phone at a time, but being able to see a call in queue and doing a pickup)

However, it seems that the page in the wiki dealing with BLF doesn't exist? Did it ever? I tried permutations of the link using some dashes and % for spaces, but can't resolve it.

dkgroot, 2016-11-22:
Which wiki page are you looking at, which contains the link to the  missing page ?

jrodder, 2016-11-23:
The link I was trying to view was under: _https://github.com/chan-sccp/chan-sccp/wiki/Features
BLF	Busy Lamp Field support advanced presence information via Hints_
Maybe that page just simply never existed?

As to what I was actually trying to accomplish, it's pretty much as you had stated, other than needing to actually pick up the waiting call by using the same BLF button. Currently my users are used to using the Cisco 7975's tied into a UC520, where when a call comes into the queue even though the phones ring one at a time, every user in that group can see the handset blink and a message 'X number of people in queue' or something to that effect. I believe I can press the softkey now gpickup to grab a call in the queue as is, but no one gets any notification that there's even a call in queue. 

Of course if this isn't possible or way more trouble than it's worth, I can always just ring all the phones at once. I am just trying to keep the transition for the end users as transparent as possible. People hate change.

dkgroot, 2016-11-23:
Hi J,

>
> The link I was trying to view was under: /https://github.com/chan-sccp/chan-sccp/wiki/Features
> BLF Busy Lamp Field support advanced presence information via Hints/
> Maybe that page just simply never existed?
The page exists (it has a title :-)), its just doesn't contain much information. We still need some help filling some of wiki pages. So if you have some information that should go on this page, please contribute.
> As to what I was actually trying to accomplish, it's pretty much as you had stated, other than needing to actually pick up the waiting call by using the same BLF button. Currently my users are used [...]
If you say that pickup works even when the call is being handled by the Queue, then we should be able to get something going.
_/
queues.conf/_

... [it_queue] musicclass = default announce = queue-markqstrategy = rrmemory announce-frequency = 90 min-announce-frequency = 15 timeout = 5 retry = 3 timeoutpriority = app|conf periodic-announce = queue-periodic-announce,your-call-is-important,please-wait member => SCCP/111 member => SCCP/112 member => SCCP/113

_/extensions.conf/_

[trusted]
exten => 8501,hint,Queue:it_queue        ;Provide a hint for the queue
exten => 8501,1,Goto(it-queue,it,1)      ;Ring the queue

exten => _**8501,1,Pickup(itqueue@PICKUPMARK)

[it-queue]
exten => it,1,Set(_PICKUPMARK=itqueue)
exten => it,n,Queue(it_queue,crhH,,,127)

_/sccp.conf/_

... [SEPXXXXX] button = speeddial, ITQueue, **8501, 8501@trusted

This seems to do the trick.

Calls come in through trusted context, hint is set for the queue (apparently that was added in asterisk-1.8). Call is routed to the it-queue context using goto. For the queue is entered we set a pickupmark, to make it easier to the call up. A speeddial is added to the phone, which monitors the hint for the queue. When pressing the speeddial it calls **8501, which forces the pickup using the pickupmark set before.
>
> Of course if this isn't possible or way more trouble than it's worth, I can always just ring all the phones at once. I am just trying to keep the transition for the end users as transparent as possible. People hate change.
Hope this solves the thing you were trying to do. Creative dialplan logic is your friend (ok not always, but in this case).

Please help us out writing some content for the wiki page. I already added this solution to it as an example
https://github.com/chan-sccp/chan-sccp/wiki/Busy-Lamp-Field

If you have more, please do supply them. If you want to edit some pages directly, let me know.

Regards,

Diederik

jrodder, 2016-11-23:
Wow that was a lot more help than I ever expected! I should  have mentioned at the outset, great work on this driver. It really saved the day when I was researching how to make best use of our current hardware and transitioning off an EOL Cisco product. (I hated that thing anyway)

As far as the wiki page goes, it turns out that when I logged in to github, I could see a page existed but was simply not populated. If you're not logged in, it redirects back to the main page. This is probably totally normal expected behavior I just didn't realize it as such.

I am still learning pretty much everything at this point, and was scouring the internet and these documentation pages to try and figure out how all this works. I didn't even really understand the hint system up until yesterday. If and when the day comes that I feel like I know enough to contribute to your documentation, I most certainly would like to do so. I really appreciate the help, busy with holidays coming up here but afterwards I will dig back in and try to implement your solution. 

-Thanks!

Jared

closed 2016-11-23 by jrodder

dkgroot, 2016-11-23:
Hi Jared,

>
> Wow that was a lot more help than I ever expected! I should have mentioned at the outset, great work on this driver. It really saved the day when I was researching how to make best use of our current hardware and transitioning off an EOL Cisco product. (I hated that thing anyway)
I even went overboard a little further with: https://github.com/chan-sccp/chan-sccp/wiki/Sign-In-Out-of-Queue-using-BLF.-Monitor-Queue-using-BLF%2C-allow-pickup

Which combines 'your' Queue Monitor + Pickup  solution, with Queue Sign-in/Sign-out. Most of the time users will want to opt-in and out of queues.

If you are just starting out with asterisk, it might be easier to use extensions.ael instead of extensions.conf (i find AEL easier to comprehend), but this one is in extensions.conf format (converting it to ael is simple though). I normally use AEL throughout.

> As far as the wiki page goes, it turns out that when I logged in to github, I could see a page existed but was simply not populated. If you're not logged in, it redirects back to the main page. This is probably totally normal expected behavior I just didn't realize it as such.
:-) Our wiki needs a little more work. So if you want to lent a hand, let me know.
> I am still learning pretty much everything at this point, and was scouring the internet and these documentation pages to try and figure out how all this works. I didn't even really understand the [...]
Don't wait until you understand all of it, because you will never remember to write any wiki pages. Just write something up, even if it is partial or false. We/I will check / proofread it before it gets submitted anyway. Problem with us developers/maintainers is we don't know what's simple and clear from the get go and what isn't. So if setting up a devstate feature button is not completely clear, just write up what you think it does, even add a question in there, one of us will pick it up, extend it etc. Like we did your original question. If you wait, it won't happen :-)

Regards,

Diederik

P.S. Donations are always welcome as well :-)