chan-sccp/chan-sccp #349: Answered Calls Show as Missed Calls on Other Extensions

jamesmz, 2016-11-24:
Hi guys

I've updated to the latest version of chan sccp today, which has given me a new issue. When a call comes in to a ring group where several extensions ring together, and the call is answered on one phone, all of the other phones in that ring group display it as a missed call. This wasn't the case before I updated.

I'm using Asterisk 13.7.2. I see this in the CLI when the call is answered:

[2016-11-24 14:37:35] NOTICE[11050][C-000000cf]: ast113.c:1252 sccp_wrapper_asterisk113_hangup:     -- SCCP: This call was answered elsewhere
[2016-11-24 14:37:35] NOTICE[11050][C-000000cf]: ast113.c:1252 sccp_wrapper_asterisk113_hangup:     -- SCCP: This call was answered elsewhere
[2016-11-24 14:37:35] NOTICE[11050][C-000000cf]: ast113.c:1252 sccp_wrapper_asterisk113_hangup:     -- SCCP: This call was answered elsewhere

Thank you.

Glasswire, 2016-12-06:
I'm having the same problem.

dkgroot, 2016-12-11:
Calls which are answered elsewhere should be suppressed from entering the phonebook. It would be helpfull to know from which revision you came, so we can use git bisect to figure out where the regression occured.

jamesmz, 2016-12-17:
I'd be struggling to say which version I came from unfortunately, as I didn't do an update for a few weeks. Unless there's any way of showing a history of versions that have been installed?

dkgroot, 2016-12-17:
@jamesmz I don't know of a way to figure that out, sorry.

jamesmz, 2016-12-17:
I've re-tested that branch but the issue is still there I'm afraid.

dkgroot, 2016-12-17:
Ok. thanks for checking. Just needed to make sure. Dealing with multiple issues at the same time at the moment :-)

dkgroot, 2016-12-18:
@jamesmz 
Revision 65cd538 should take care of this. I removed the use of CallHistoryDisposition which only seems to be supported by a limited set of devices and firmware revisions, we are back to the old method (which should work). Please confirm.

jamesmz, 2016-12-18:
@dkgroot 
Yes it's working normally again now, thank you :)

closed 2016-12-18 by jamesmz

dkgroot, 2016-12-18:
@jamesmz Thanks for helping out on this one and reporting back ! I though i had reverted that patch a long time ago :-)

PR #359 (2016-12-27): Issues/354
Fixes Issue: #354

functionality: Including Intercept softkey in your onhookstealable softkeyset, will make the remotely connected call visable in the callplane. If the intercept softkey is not available, then the call is collapsed.

Note: bit of cleanup of stale unused/testing code in sccp_indicate.c also happened.

Related issues: #356,  #350, #349

Completely fixed by this Fix: #349, #350 
Partially fixed by this Fix: #356