chan-sccp/chan-sccp #350: Hold/Pickup error up Shared Line

Glasswire, 2016-12-02:
When I have a call come in on a shared line (ie. 5214), I put it on hold on one phone and try to pick it up on another. Normally when you put a shared line on hold, the call box pops up on all handsets that have that line and gives you your resume option. But sometimes (not always) it just show as if the line key as if it was a speeddial key where to left it shows  <##########>  --------------------------------------  John Smith. The box that shows up when the call is on hold doesn't pop up. With that being said, you can't pick up the call because there is no resume option. You can press the BLF, and you will pickup the call, but then the options displayed are your idle ones (Redial, New Call, CFwdALL, etc)

**Reproduction Steps:**

1. Answer the shared line
2. Put the call on hold 
3. Look at another phone with the shared line & Resume it
4. Keep putting the call on hold & switching between phones until you no longer see the "call on hold box"

**Expected behavior:**

You should always have the option to Resume a shared line on hold

**Observed behavior:**

Sometimes the Resume feature along with hold time box is not shown on other phones.

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)
When the initial call is answered
http://assets.glasswiresolutions.com/initial.png

When the call was placed on hold and was attempted to be picked up on another extention
http://assets.glasswiresolutions.com/hold_error.png

**chan-sccp version:** 4.3.0 12f96a9M

**asterisk version:** 13.12.2

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: This also happened in the current stable release
* Problem can be reliably reproduced, doesn't happen randomly: Yes
* Platform:
    * [CPU]
    * [32/64 Bit] 64 Bit
    * [Big/Small Endian]
    * [Operating System] CentOS
    * [Operating System Version]

Glasswire, 2016-12-02:
Here is a video

https://youtu.be/YFGRGkeeWN0

marcelloceschia, 2016-12-02:
@Glasswire I will have a look for this issue on sunday or monday

dkgroot, 2016-12-11:
possibly related to issue 354 

sounds like a race condition, which are notoriously difficult to reproduce and debug. If you can reproduce it with some level of certainty, then you might want to supply some debugging with:
```
logger rotate
sccp debug core, indicate, device
```

dkgroot, 2016-12-11:
Note: Using branch 'issues/354' apparently fixes the racecondition (according to bug report in issue 354)

dkgroot, 2016-12-27:
The issues/354 branch has now been merged. Can you still reproduce the issue. If not, please close this issue. If you can, we will need to dig a little deeper.

Glasswire, 2016-12-27:
@dkgroot I will be home Thursday night. I will check it and let you know :)

closed 2017-08-21 by Glasswire

dkgroot, 2017-08-21:
@Glasswire thanks for checking !