chan-sccp/chan-sccp #351: Session Numbers increase when on hold caller disconnects

Glasswire, 2016-12-02:
In the new developer version, the session number has been fixed (for the most part). I have noticed if you put someone on hold & the call disconnects, the next time you receive/make a call, the session # increases by 1. The only way to fix it is 'core restart now'.

**Reproduction Steps:**

1. Call phone (ie. 211)  and put call on hold
2. Hangup phone you made the initial call on (ie. 210)
3. Call the phone you answered on the first time (211)
4. Notice the increased session number

**Expected behavior:**

If caller disconnects while on hold, channel should hangup & session number will return to 1

**Observed behavior:**

When on-hold caller disconnects, the session number increases by 1.

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)

https://youtu.be/8oqd-13RXwY

**chan-sccp version:** Release: 4.3.0 develop - 12f96a9M 

**asterisk version:** 13
**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No] Not Sure
* Problem can be reliably reproduced, doesn't happen randomly: [Yes/No] Yes
* Platform:
    * [CPU]
    * [32/64 Bit]
    * [Big/Small Endian]
    * [Operating System]
    * [Operating System Version]
* Direct Contact information: [You can contact me here]
* Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]
* Comments: [Other Information]
*

marcelloceschia, 2016-12-02:
related to issue #313 and f53c1ad

Glasswire, 2016-12-02:
I do want to mention normal calls are keeping the correct session numbers.

PR #352 (2016-12-02): fix for issue #351
Closes #351.

Changes proposed by this Pull Request:
- decrement the numberOfHeldChannels on hangup (moved from channel to pbx)

Inform: @dkgroot 

@Glasswire can you chechout branch issue/351 and check if this fixes your issue? Thank you
commit 3bbfd20f (2016-12-02): Merge pull request #352 from chan-sccp/issue/351

fixes #351
closed 2016-12-02 by marcelloceschia