chan-sccp/chan-sccp #354: Call answered on Shared Line

Glasswire, 2016-12-10:
[Short description of problem here]

**Reproduction Steps:**

1. 1st call received on shared extension 5210
2. While on the phone, 2nd call comes in on 5210 & is answered on another phone
3. Incoming call 'bubble" is still displayed as if no one has answered yet along with the incoming to/from caller info at the bottom of the screen (in this case 5210: From: anonymous)

**Expected behavior:**

Call bubble should disappear and the screen should look like it did before the second call comes in.

**Observed behavior:**

Incoming call 'bubble" is still displayed as if no one has answered yet along with the incoming to/from caller info at the bottom of the screen (in this case 5210: From: anonymous)

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)

https://youtu.be/2i9bsXMiojo

**chan-sccp version:** (include revision if relevant)

Skinny Client Control Protocol (SCCP). Release: 4.3.0 origin/issue/351 - 3bf9a27M

**asterisk version:** 11

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No] Not sure
* Problem can be reliably reproduced, doesn't happen randomly: [Yes/No] Yes
* Platform:
    * [CPU]
    * [32/64 Bit]
    * [Big/Small Endian]
    * [Operating System]
    * [Operating System Version]
* Direct Contact information: [You can contact me here]
* Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]
* Comments: [Other Information]
*

dkgroot, 2016-12-11:
@Glasswire : I ran a couple of tests using develop and it looks ok to me. On a shared line you can see the other calls going on on that same line. If anything should change, then the second call between two remote phone might be display as a collapsed callplane instead of a full size one. 

Just to make sure i went back to revisions from the beginning of this year to check if there was some kind of regression in this piece of code, but could not verify this.

@marcelloceschia : Can you give a definitive answer if and how this second active call between two remote parties on a shared should have been displayed ?

dkgroot, 2016-12-11:
Provided a fix in case this is/was a regression. It is available in 'issues/354' branch.
```
git checkout issues/354
make && make install && make reload
```
@Glasswire : please verify if this does what you expected to happen
@marcelloceschia : can you confirm/deny if this is what it was intended to do

Glasswire, 2016-12-11:
I'm getting error "issues/354" did not match any files known to git.

dkgroot, 2016-12-11:
@Glasswire 
The branch https://github.com/chan-sccp/chan-sccp/tree/issues/354 does exist.
Can you do
```
git fetch
git pull
git checkout issues/354
make && make install && make reload
```
That should fix it.

Glasswire, 2016-12-11:
This did fix issue 354 & 350. Now the only problem is that you can't see who the other calls on that line, preventing you from using features like intercept. The calls do show up if you put them on hold though.

dkgroot, 2016-12-11:
@Glasswire : (I forgot about intercept when creating the patch). For intercept to work we need an active callplane to react to. So either we keep the code as before, where you can see the other call, but cannot interact with it directly (except for intercept), or we don't show the information.

What is the issue with showing the remote call ? I mean it is a shared line, so you would expect it to show the state of the complete line, including remote calls going on.

Glasswire, 2016-12-11:
If I am on a shared line (ie. 210) and have an established call, any other phone with that shared extension just shows the red BLF. Even if you open a channel on the shared extenion, the other calls don't show. (See screenshots)
[image]
[image]

Glasswire, 2016-12-11:
After doing some further testing, when you try to pickup a call on hold on a shared line, it works if you press resume & the phone defaults to speaker phone, but if you pickup the handset & press the up arrow to select the call & press Resume, nothing happens.

Glasswire, 2016-12-11:
This also fixed issue #349

dkgroot, 2016-12-11:
@Glasswire 
So the patch fixes #354, #350 and #349, but introduces new issues
 - not being able to see callinfo and intercept the remote call (which is logical as we suppressed the callplane)
Correct ?

Is the speaker/handset issue the same before and after the patch ? Should we handle this as a new/seperate issue ?

Glasswire, 2016-12-11:
Correct

And yes that issue was present before

dkgroot, 2016-12-11:
Not sure how to proceed now:
- Intercepting is a nice feature, but does require the callplane to be visible.
- Not having the callplane visible, makes for a cleaner display with less distraction.

I don't understand how adding this one line of code, fixed issues #350 and #349. 
@marcelloceschia : Any thoughts ?

@Glasswire : Can you create a new issue for the handset/speaker call resume issue, please.

Glasswire, 2016-12-11:
For issue #349, the missed call issue works if it's the same extension (which it didn't before)  But if one phone in the ring group picks up, the other extensions show missed call.  Is this behavior normal/expected?
--
Sent from myMail for Android Sunday,

dkgroot, 2016-12-11:
Suppressing the callplane has a side effect in phonebook maintenance. We do not have direct control of what ends up in the phonebook (missed calls etc), we can only steer it's behaviour partially by 'showing' specific callid's at certain times.

Calls which are answered elsewhere should be suppressed from entering the phonebook. It would be helpfull to know from which revision you came, so we can use git bisect to figure out where the regression occured.

dkgroot, 2016-12-11:
@Glasswire: FYI: i cannot reproduce the speaker/handset resume issue on my 7970, with or without the patch.

Glasswire, 2016-12-11:
Skinny Client Control Protocol (SCCP). Release: 4.3.0 issues/354 - 931074bM (built by 'root' on '2016-12-11 17:02:03 UTC')

And the problem is that if a call is on hold on a shared extension, and you pickup the handset to retrieve (using the arrow keys to select the line) & press Resume, nothing happens.

[2016-12-11 17:34:09] WARNING[1876][C-00000006]: sccp_channel.c:1634 sccp_channel_resume: SCCP: swap_channels failed to put channel 62 on hold. exiting

Glasswire, 2016-12-12:
Would it be possible to just make the callplane minimized after it is answered on another extension? (IE. call comes in on shared line. Answered by ext 1. Second call comes in on shared ext & is answered on ext. 2. Then the call plane for the second incoming call that was answered should be minimized on ext 1) I hope that made sense.

dkgroot, 2016-12-12:
@Glasswire 
Regarding the Handset/Resume issue: Could you make a separate issue out of that. It does not seem directly related, and it's easier to deal with issue on their own. Please also include :
- type of phone 
- firmware version on that phone

@Glasswire 
In issues/354 patch i added a line to set the callplane to 'collapsed' which is just what you suggested. It does have different call behaviour though. We only have the choice of 'default', 'collapsed' and 'hidden'. It does not provide exactly what we are looking for though in this case.

Glasswire, 2016-12-13:
That didn't work as expected. Maybe it would be better to leave it as is which means

Cons
- You can not see who is on the shared line
- You can not use intercept

Pros
- Fixes missed call issue
- Fixes shared line hold issue
- Cleaner display

Maybe you could make Intercept an 'Optional Feature'. However I don't know how that would help seeing as it comes with all of the other issues. I'm just putting my 2 cents in.

Glasswire, 2016-12-13:
@dkgroot - Release: 4.3.0 origin/issue/351 - 3bf9a27M - Was the revision I came from before I posted this issue in reply to you previous comment about the phonebook maintenance.

dkgroot, 2016-12-13:
Adding even more options to the sccp.conf (which is pretty overcrowded as is), is not really an option.

- We know how to fix the missed call options (we can suppress that). 
- I would like to have intercept by default (if you don't want it, you can remove it from the softkeyset, so already configurable). 
- I personally don't mind the second call info box two much to be honest.

I need a little time to do some more research to check if something changed and what exactly changed in this regard.

dkgroot, 2016-12-14:
@Glasswire 
Updated issues/354 branch

Added the abiity to detect if the intercept softkey is available in the onhookstealable softkeyset. If not we suppress the presentation of the remote call when it is connected remotely (as per your request). If the intercept softkey is available, then we present the remote call, so it can be intercepted.

Re-added an old solution to suppress phonebook entries for missed calls, when they where actually answered remotely (ie: channel answered elsewhere) (related to issue #349).

Rebase this branch on top of changes made to the develop branch, so we do not get behind.

To test you can create two softkeysets one with and one without the intercept softkey, to make it a little easier to check the functionality.

Glasswire, 2016-12-15:
This did work perfectly except for the initial hold issue I was having happens when intercept is available.

dkgroot, 2016-12-16:
So this is a step forward (and it made the behaviour configurable :+1:  )
Now this combined with the 'issues/356' branch should fix the call resume remote call issue as well (I think).

If you like you can test the combination of the two branches (in a clean temp directory) by doing this:
```
git clone https://github.com/chan-sccp/chan-sccp.git chan-sccp_merge_354_356
cd chan-sccp_merge_354_356
git checkout issues/354
git pull origin issues/356
./configure [...your regular configure flags...]
make -j2 && make install && make reload
```
And do your tests.

Glasswire, 2016-12-17:
I did this & nothing changed :( I posted my debug log on issue #356

Glasswire, 2016-12-17:
This is what happens when I press Hold on a shared line & it doesn't pop up to Resume on the other phones.

```
SEP001FCAE812A7: Got Softkey
SEP001FCAE812A7: Got Softkey: Hold (3) line=2 callid=21
SEP001FCAE812A7: Getting the active channel on device.
SEP001FCAE812A7: Hold the channel SCCP/5210-00000015
SEP001FCAE812A7: Indicate SCCP new state:HOLD, current channel state:CONNECTED on call:SCCP/5210-00000015, lineInstance:2 (previous channelstate:HOLD)
 SEP001FCAE812A7: Remote Indicate state HOLD (32) with reason: NORMAL (0) on remote devices for channel SCCP/5210-00000015
SEP0024C4BED45F: Getting the active channel on device.
SEP0024C4BED45F: No active channel on device.
SEP001FCAE812A7: Finish Indicating state HOLD (32) with reason: NORMAL (0) on remote device SEP0024C4BED45F for channel SCCP/5210-00000015
SEP001FCAE812A7: Finish to indicate channel state:HOLD on call:SCCP/5210-00000015, lineInstance:2. New channel state:HOLD
SEP001FCAE812A7: Set the active channel 0 on device
```

dkgroot, 2016-12-18:
From the debug log you posted here, i would say it does:
```
SEP001FCAE812A7: Remote Indicate state HOLD (32) with reason: NORMAL (0) on remote devices for channel SCCP/5210-00000015
SEP0024C4BED45F: Getting the active channel on device.
SEP0024C4BED45F: No active channel on device.
```
It did go on to remote indicate, and then publish the call to SEP0024C4BED45F. But apparently you don't see it.

Would you be able to add a little debug output to __sccp_indicate_remote_device on your own, or would you like me to do that, so we can see what happens during remote indicate ?

dkgroot, 2016-12-27:
Fixed and Merged into the develop branch
You can remove the Intercept softkey from your onhookstealable softkeys. If intercept is not available, the call will only be displayed if it is on hold.

commit ed4537a2 (2016-12-27): Fix: This would show a remotely connected call in a collapsed callplane.
Fix: Suppress phonebook entry of a missed call when call is indicated remotely
Enh: Added sccp_softkey_isSoftkeyInSoftkeySet to check if a softkey is enabled in a particular softkeyset on a device
Enh: By removing intercept button from the onhookstealable keyset you can influence the way a remote call is displayed (collapsed or in plain view)

Fixes #354
References #349
Partially References #350
closed 2016-12-27 by dkgroot