chan-sccp/chan-sccp #356: Resume call issues

Glasswire, 2016-12-13:
On one phone we have 2 lines setup. The first (& default) line is ext 5211 (Ka'Loni Holt), the second is ext 5210 (Cody Campbell). If a call is received on ext 5210, then placed on hold. The call can not be resumed by picking up the handset and pressing the line key. The screen returns to idle & clears the 'Connected' softkey set & replaces it with the 'Onhook' softkey set.

**Reproduction Steps:**

1. Place call to line registered as secondary line on phone (ie. ext. 5210) & answer via handset.
2. Place call on hold and hang up handset
3. Pickup up handset & press the line key of the line that has the call on hold.

**Expected behavior:**

By pressing the BLF of the call that is on hold, it should open that line so you can select to resume call -or- make another outbound call on that line. ( See Below )

[image]

**Observed behavior:**

When BLF is pressed, screen returns to idle state (with the exception that you can see the call that is on hold, see Step 3 picture) & the softkey set changes from 'Connected' to 'Onhook'. Leaving you no way to pickup the on-hold call.

**Screenshots**

**Step 1 | Call is received (ringing)** 

[image]

**Step 2 | Call has been placed on hold, now handset is lifted and I press the line key to resume the call**

[image]

**Step 3 | This is what happens after I press the BLF**

[image]

**chan-sccp version:** (include revision if relevant)

Verison: 4.3.0 issues/354
Revision: 931074bM

**asterisk version:** (include revision if relevant)

Asterisk 11.24.1

**Addition / Optional Information:**
* Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No] Yes
* Problem can be reliably reproduced, doesn't happen randomly: [Yes/No] Yes
* Platform:
    * [CPU]
    * [32/64 Bit] 64 Bit
    * [Big/Small Endian]
    * [Operating System] FreePBX 11
    * [Operating System Version]
* Direct Contact information: [You can contact me here]
* Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]
* Comments: [Other Information]
*

dkgroot, 2016-12-13:
@Glasswire 
I do understand how to reproduce this now. Problem is this should work correctly using the develop branch.

I set 'sccp debug core action channel pbx indicate' to show you how this is handled on my phone.

line 1: 98041 is a single (non-shared) line
line 2: 98099 is a shared line

This is what happens when the initial call is on hold on the second line.
```
    --     -- SEP0023043403F9: Getting the active channel on device.
    --     -- SEP0023043403F9: No active channel on device.
    --     -- SEP0023043403F9: Taken Offhook
    --     -- SCCP: Looking for channel by state '21' on device: SEP0023043403F9
    --     -- SEP0023043403F9: line: '98041'
    --     -- SEP0023043403F9: line: '98099'
    --     -- SEP0023043403F9: Could not find active channel with state RINGING(21) on device
    --     -- SEP0023043403F9: Using line 98041
SEP0023043403F9: (getEmptyChannel) on line:98041, maybe_c:
SEP0023043403F9: (getEmptyChannel) getActiveChannel
    --     -- SEP0023043403F9: Getting the active channel on device.
    --     -- SEP0023043403F9: No active channel on device.
    --     -- SEP0023043403F9: Set the active channel 3 on device
```
I was not able to find an active channel on this line, so i creates a new one (with an empty callplane), which makes sense, it does not know yet that you picked up the receiver to actually resume a call on the second line
```
    --     -- 0001: New channel number: 3 on line 98041
    --     -- SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 98041, 
    --     -- SCCP:              cid_num: "98041"
    --     -- SCCP:             cid_name: "PHONE4"
    --     -- SCCP:          accountcode: "79005"
    --     -- SCCP:                exten: ""
    --     -- SCCP:              context: "internal"
    --     -- SCCP:             amaflags: "0"
    --     -- SCCP:            chan/call: "SCCP/98041-00000003"
...
    --     -- 0001: Allocated asterisk channel SCCP/98041-00000003
    --     -- SEP0023043403F9: Set asterisk state OffHook (2) for call 3
    --     -- SEP0023043403F9: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/98041-00000003, lineInstance:1 (previous channelstate:DOWN)
    --     -- SEP0023043403F9: Finish to indicate channel state:OFFHOOK on call:SCCP/98041-00000003, lineInstance:1. New channel state:OFFHOOK
    --     -- SCCP/98041-00000003: schedule digittimeout 6
```
The new channel was created and you have 6 seconds to start the first digit. Now you press the second line button to resume that call on hold.
FYI: please make sure your firstdigittimeout is long enough for you to have enough time to press the second line button.
```
    --     -- SEP0023043403F9: Got stimulus=Line (9) for instance=2, callreference=0, status=0
    --     -- SEP0023043403F9: Handle Line Button Stimulus
    --     -- SEP0023043403F9: Line Key press on line 98099
```
The button press (aka stimulus) was received. Because we had a new channel which was waiting for initial input on the first line, that needs to be cancelled first. Then we have to figure out why the line 2 button was pressed (if this is a new call, a resume action or answering of a ringing channel).
```
    --     -- SEP0023043403F9: Getting the active channel on device.
    --     -- SEP0023043403F9: got active channel SCCP/98041-00000003, line stimulus on line: 98099
    --     -- SEP0023043403F9: Call:SCCP/98041-00000003 not in progress. Ending Call
    --     -- SCCP/98041-00000003: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:4) (hangup_id:-1)
  ==   == SEP0023043403F9: Ending call SCCP/98041-00000003 (state:OFFHOOK)
    --     -- SEP0023043403F9: Sending hangupRequest to Call SCCP/98041-00000003 (state: OFFHOOK)
    --     -- SEP0023043403F9: Current channel SCCP/98041-00000003 state OFFHOOK(10)
    --     -- SEP0023043403F9: Indicate SCCP new state:ONHOOK, current channel state:OFFHOOK on call:SCCP/98041-00000003, lineInstance:1 (previous channelstate:DOWN)
    --     -- SEP0023043403F9: Finish to indicate channel state:ONHOOK on call:SCCP/98041-00000003, lineInstance:1. New channel state:DOWN
    --     -- SEP0023043403F9: Device is Requesting CallStatisticsAndClear
    --     -- SCCP: Cleaning channel SCCP/98041-00000003
    --     -- SEP0023043403F9: Set the active channel 0 on device
    --     -- SEP0023043403F9: Looking for selected channel (3)
    --     -- SEP0023043403F9: Active Call:SCCP/98041-00000003 and line:98099 pressed are different => shared line handling
    --     -- Destroying channel SCCP/98041-00000003
```
We have started destroying the channel for the first line and are trying to find out the state for line 2.
```
    --     -- SEP0023043403F9: Handle Line Button Stimulus on Inactive Line
[... 13 lines cut ...]
    --     -- SEP0023043403F9: Open receive channel with format G.711 A-law 64k[2], payload 2, echocancel: YES, passthrupartyid: 4294967293, callid: 2
    --     -- SEP0023043403F9: Remote Indicate state CONNECTED (30) with reason: NORMAL (0) on remote devices for channel SCCP/98099-00000002
    --     -- SEPE0D173E11D95: Getting the active channel on device.
    --     -- SEPE0D173E11D95: No active channel on device.
    --     -- SEPE0D173E11D95: Send callinfo(V20) for Inbound channel 2/1 on line instance 2
...
    --     -- SCCP: (pbx_indicate) start indicate 'AST_CONTROL_UNHOLD: Indicate call left hold'
[... cut ...]
The process described above could be influenced if you do not have:
```
<vendorConfig>
    <autoSelectLineEnable>0</autoSelectLineEnable>
    <autoCallSelect>0</autoCallSelect>
</vendorConfig>
 <autoCallPickupEnable>0</autoCallPickupEnable>
```
in your SEP-file. These settings can influence how the phone responds.

Can you check your cnf.xml file and produce a similar piece of logging, if it still does not work as you would expect.

Note that my second line was a shared line which might influence the call handling and logging slightly.

Glasswire, 2016-12-13:
Setting the new parameters in the SEP.CNF didn't work :/ I switched back to the develop branch before the patch & the same issue occurs. My output is not showing what your did. All I have is 
SEP001FCAE812A7: (allowPickup) on line:5211.

When I press the BLF

Glasswire, 2016-12-13:
The problem seems to lie in where you said _'Because we had a new channel which was waiting for initial input on the first line, that needs to be cancelled first.'_ - Reason being that when I pick up the handset and my main (first) line is selected, I can press the BLF of the first line to destroy that channel, then I can press the shared line and the call plane shows the call on hold & gives me the option to resume or make a new call.

See  https://youtu.be/W6lSt-gi-Mc

dkgroot, 2016-12-13:
Can you provide the debug logging with 'sccp debug core action channel pbx indicate' set for a single one of these calls. So i can have a look at what goes on, please.

FYI: If not grabbing from CLI, but using the log file, you should add filelinefunc to the debug parameters.

Glasswire, 2016-12-14:
Sorry about that, here is debug log

After I pick up that handset & the default line is selected

```
SEP001FCAE812A7: Getting the active channel on device.
SEP001FCAE812A7: No active channel on device.
SEP001FCAE812A7: Taken Offhook
SCCP: Looking for channel by state '21' on device: SEP001FCAE812A7
SEP001FCAE812A7: line: '5211' SEP001FCAE812A7: line: '5210'
SEP001FCAE812A7: Could not find active channel with state RINGING(21) on device
SEP001FCAE812A7: Using line 5211
SEP001FCAE812A7: (getEmptyChannel) on line:5211, maybe_c:
SEP001FCAE812A7: (getEmptyChannel) getActiveChannel
SEP001FCAE812A7: Getting the active channel on device.
SEP001FCAE812A7: No active channel on device.
SEP001FCAE812A7: Set the active channel 11 on device
0003: New channel number: 11 on line 5211
SCCP: (pbx_channel_allocate) try to allocate Outbound channel on line: 5211, 
SCCP:              cid_num: "5211"
SCCP:             cid_name: "KA'LONI HOLT"
SCCP:          accountcode: "79011"
SCCP:                exten: ""
SCCP:              context: "from-internal-xfer"
SCCP:             amaflags: "0"
SCCP:            chan/call: "SCCP/5211-0000000B"
SCCP:combined capabilities: "[slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a, rfc2833]"
SCCP:  reduced preferences: "[ulaw/64k]"
SCCP: (allocPBXChannel) Create New Channel with name: SCCP/5211-0000000B
SCCP/5211-0000000B: (recalculateReadformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a, rfc2833]
	preferences: [ulaw/64k]
	remote caps: []
	READ:[ulaw/64k]
SCCP/5211-0000000B: (recalculateWriteformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a, rfc2833]
	preferences: [ulaw/64k]
	remote caps: []
	Write:[ulaw/64k]
0003: Allocated asterisk channel SCCP/5211-0000000B
SEP001FCAE812A7: Set asterisk state OffHook (2) for call 11
SEP001FCAE812A7: Indicate SCCP new state:OFFHOOK, current channel state:DOWN on call:SCCP/5211-0000000B, lineInstance:1 (previous channelstate:DOWN)
SEP001FCAE812A7: Remote Indicate state OFFHOOK (10) with reason: NORMAL (0) on remote devices for channel SCCP/5211-0000000B
SEP0024C4BED45F: Getting the active channel on device.
SEP001FCAE812A7: Finish Indicating state OFFHOOK (10) with reason: NORMAL (0) on remote device SEP0024C4BED45F for channel SCCP/5211-0000000B
SEP001FCAE812A7: Finish to indicate channel state:OFFHOOK on call:SCCP/5211-0000000B, lineInstance:1. New channel state:OFFHOOK
SCCP/5211-0000000B: schedule digittimeout 16
```

After I Press the BLF of the line with the call on hold

```
SEP001FCAE812A7: Got stimulus=Line (9) for instance=2, callreference=0, status=0
SEP001FCAE812A7: Handle Line Button Stimulus
SEP001FCAE812A7: Line Key press on line 5210
SEP001FCAE812A7: Getting the active channel on device.
SEP001FCAE812A7: got active channel SCCP/5211-0000000B, line stimulus on line: 5210
SEP001FCAE812A7: Call:SCCP/5211-0000000B not in progress. Ending Call
SCCP/5211-0000000B: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:5) (hangup_id:-1)
SEP001FCAE812A7: Ending call SCCP/5211-0000000B (state:OFFHOOK)
SEP001FCAE812A7: Sending hangupRequest to Call SCCP/5211-0000000B (state: OFFHOOK)
SEP001FCAE812A7: Current channel SCCP/5211-0000000B state OFFHOOK(10)
SEP001FCAE812A7: Indicate SCCP new state:ONHOOK, current channel state:OFFHOOK on call:SCCP/5211-0000000B, lineInstance:1 (previous channelstate:DOWN)
SEP001FCAE812A7: Remote Indicate state ONHOOK (1) with reason: NORMAL (0) on remote devices for channel SCCP/5211-0000000B
SEP001FCAE812A7 -> SEP0024C4BED45F: indicate remote onhook (lineInstance: 2, callid: 11)
SEP001FCAE812A7: Finish Indicating state ONHOOK (1) with reason: NORMAL (0) on remote device SEP0024C4BED45F for channel SCCP/5211-0000000B
SEP001FCAE812A7: Finish to indicate channel state:ONHOOK on call:SCCP/5211-0000000B, lineInstance:1. New channel state:DOWN
SEP001FCAE812A7: Device is Requesting CallStatisticsAndClear
SCCP: Cleaning channel SCCP/5211-0000000B
SEP001FCAE812A7: Set the active channel 0 on device
SEP001FCAE812A7: Looking for selected channel (11)
SEP001FCAE812A7: Active Call:SCCP/5211-0000000B and line:5210 pressed are different => shared line handling
Destroying channel SCCP/5211-0000000B
SEP001FCAE812A7: Handle Line Button Stimulus on Inactive Line
SCCP: Looking for channel by state '21'
0001: Could not find active channel with state RINGING(21) on line
SCCP: Looking for channel by state '32'
SEP001FCAE812A7: Channel count on line 2 = 2[2016-12-13 19:22:31] NOTICE[59151]: sccp_actions.c:1802 SEP001FCAE812A7: Switch to line 2[2016-12-13 19:22:31] NOTICE[59151]: sccp_actions.c:3646 QualityStats: MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;ICR=0.0000;CCR=0.0000;ICRmx=0.0000;CS=0;SCS=0;MLQKvr=0.95
 SEP001FCAE812A7: Call Statistics:
       [
         Last Call        : CallID: 11 Packets sent: 62 rcvd: 57 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 3 Packets sent: 128 rcvd: 123 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
       ]

```

dkgroot, 2016-12-14:
@Glasswire 
The important part is in the last piece of logging, namely:
```
SEP001FCAE812A7: Active Call:SCCP/5211-0000000B and line:5210 pressed are different => shared line handling
Destroying channel SCCP/5211-0000000B
SEP001FCAE812A7: Handle Line Button Stimulus on Inactive Line
SCCP: Looking for channel by state '21'
0001: Could not find active channel with state RINGING(21) on line
SCCP: Looking for channel by state '32'
SEP001FCAE812A7: Channel count on line 2 = 2
SEP001FCAE812A7: Switch to line 2
```
If the channel count on line 2 would have been '1', then the hold/resume action would have been taken. Only if there is 1 single call we can find out what state it is an and make a definitive choice, what to do with that channel.

The question in this case why are there 2 calls on the second line ? Do you know if there are actually two calls on the second line ?

I am currently looking for a way to deal with 2+ calls on hold on a second line gracefully.

Glasswire, 2016-12-14:
I think it was because I called extention 5210 from 5210. So there were technically two calls? But I used 2 different phones.
--
Sent from myMail for Android Tuesday,

>@Glasswire
>The important part is in the last piece of logging, namely:
>SEP001FCAE812A7: Active Call:SCCP/5211-0000000B and line:5210 pressed are different => shared line handling
Destroying channel SCCP/5211-0000000B
SEP001FCAE812A7: Handle Line Button Stimulus on Inactive Line
SCCP: Looking for channel by state '21'
0001: Could not find active channel with state RINGING(21) on line
SCCP: Looking for channel by state '32'
SEP001FCAE812A7: Channel count on line 2 = 2
SEP001FCAE812A7: Switch to line 2

Glasswire, 2016-12-14:
UPDATE

I did use 2 different extensions this time & was able to resume the call. The only problem now is that the screen appears as it does in the *Step 3* picture above. I am speaking with the other party, but my options  revert back to the 'onhook softkey set', making it impossible to do anything else with the call except disconnect by going onhook. I'm working on getting the debug log.

dkgroot, 2016-12-14:
When a second line contains 1 channel, we receive a 'line stimulus event', but when it contains multiple channels, it just pops up, without sending an event. When you then press the resume button, we receive a 'softkey stimulus event' which gets handled in a completely different function. I added the same 'endcall' handling that was available in the 'line stimulus handler' to the 'softkey stimulus handler', which seems to remedy the issue.

Pushed a new branch 'issues/356'
Please check if that resolves this issue, and does not introduce new/other stuff.

If have not had time to check all possible softkey events that are handled here. So this needs a little further checking to make sure there are no negative side effects from this code change. Might be a good idea to set 'sccp debug core, channel, action, pbx' during testing.

dkgroot, 2016-12-14:
Thanks for the update. The 'line stimulus event handler' checks for the channels on the line, it does not check on which phone they are connected. So that explains why using 5210 from another phone would cause this effect. We ought to add that possibility into the equation as well, but this handler is already quite complex as is.

With the new code in 'issues/356' this is not fixed (yet), but it should handle it a little better, by showing the calls on hold and letting your choose, even if it's just one.

dkgroot, 2016-12-14:
Please first try / test
revision: [issues/356 47e11e]
Then switch to 
revision: [issues/356 831b1fe]
The second revision, changes the way we detect the number of calls on hold when dealing with 'line stimulus event'.

This new way partially reflects the shared line issue regarding using 5210 from two different devices. There is not 100% solution for this, because on a shared line with a channel on hold, there is no  connection between the device and the channel, it released that connection when putting the call on hold.

Glasswire, 2016-12-15:
How do you checkout specific revisions?

dkgroot, 2016-12-16:
Exactly they way you would think ;-)
```
git checkout 47e11ef
```
and/or
```
git checkout 831b1fe
```
afterwards to get back to the HEAD of develop you can do
```
git checkout develop
git pull
```

Glasswire, 2016-12-16:
I tired this and it didn't make difference? The same issue persists. I just went to my chan sccp folder and ran the following
Git fetch 
Git pull
Git checkout 'revision'
Make && make install && make reload
Then after everything was done I did fwconsole restart

dkgroot, 2016-12-16:
Can you produce the debug logging with 'sccp debug core action channel pbx indicate' so that i can see what happens on your system.

Glasswire, 2016-12-17:
```
SEP001FCAE812A7: Got stimulus=Line (9) for instance=2, callreference=0, status=0
SEP001FCAE812A7: Handle Line Button Stimulus
SEP001FCAE812A7: Line Key press on line 5210
SEP001FCAE812A7: Getting the active channel on device.
SEP001FCAE812A7: got active channel SCCP/5211-0000000E, line stimulus on line: 5210
SEP001FCAE812A7: Call:SCCP/5211-0000000E not in progress. Ending Call
SCCP/5211-0000000E: Disabling scheduler / Removing Scheduled tasks (digittimeout_id:2) (hangup_id:-1)
SEP001FCAE812A7: Ending call SCCP/5211-0000000E (state:OFFHOOK)
SEP001FCAE812A7: Sending hangupRequest to Call SCCP/5211-0000000E (state: OFFHOOK)
SEP001FCAE812A7: Current channel SCCP/5211-0000000E state OFFHOOK(10)
SEP001FCAE812A7: Indicate SCCP new state:ONHOOK, current channel state:OFFHOOK on call:SCCP/5211-0000000E, lineInstance:1 (previous channelstate:DOWN)
SEP001FCAE812A7: Remote Indicate state ONHOOK (1) with reason: NORMAL (0) on remote devices for channel SCCP/5211-0000000E
SEP001FCAE812A7 -> SEP0024C4BED45F: indicate remote onhook (lineInstance: 2, callid: 14)
SEP001FCAE812A7: Finish Indicating state ONHOOK (1) with reason: NORMAL (0) on remote device SEP0024C4BED45F for channel SCCP/5211-0000000E
SEP001FCAE812A7: Finish to indicate channel state:ONHOOK on call:SCCP/5211-0000000E, lineInstance:1. New channel state:DOWN
SEP001FCAE812A7: Device is Requesting CallStatisticsAndClear
SCCP: Cleaning channel SCCP/5211-0000000E
SEP001FCAE812A7: Set the active channel 0 on device
SEP001FCAE812A7: Looking for selected channel (14)
SEP001FCAE812A7: Active Call:SCCP/5211-0000000E and line:5210 pressed are different => shared line handling
Destroying channel SCCP/5211-0000000E
SEP001FCAE812A7: Handle Line Button Stimulus on Inactive Line
SCCP: Looking for channel by state '21'
0001: Could not find active channel with state RINGING(21) on line
SCCP: Looking for channel by state '32'
SEP001FCAE812A7: Channel count on line 2 = 1
SEP001FCAE812A7: Resume channel 13 on line 2
SEP001FCAE812A7: Getting the active channel on device.
SEP001FCAE812A7: No active channel on device.
SEP001FCAE812A7: Resume the channel SCCP/5210-0000000D
SEP001FCAE812A7: Set the active channel 13 on device
SCCP/5210-0000000D: (recalculateReadformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a, rfc2833]
	preferences: [ulaw/64k, ulaw/56k]
	remote caps: [ulaw/64k]
	READ:[ulaw/64k]
SCCP/5210-0000000D: (recalculateWriteformat) 
	capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a, rfc2833]
	preferences: [ulaw/64k, ulaw/56k]
	remote caps: [ulaw/64k]
	Write:[ulaw/64k]
SEP001FCAE812A7: Indicate SCCP new state:CONNECTED, current channel state:HOLD on call:SCCP/5210-0000000D, lineInstance:2 (previous channelstate:CONNECTED)
SEP001FCAE812A7: Open receive channel with format G.711 u-law 64k[4], payload 4, echocancel: YES, passthrupartyid: 4294967282, callid: 13
SEP001FCAE812A7: Remote Indicate state CONNECTED (30) with reason: NORMAL (0) on remote devices for channel SCCP/5210-0000000D
SEP0024C4BED45F: Getting the active channel on device.
SEP001FCAE812A7: (indicate_remote_device) Already Own Part of the Call: Skipped
SEP001FCAE812A7: Finish to indicate channel state:CONNECTED on call:SCCP/5210-0000000D, lineInstance:2. New channel state:CONNECTED
SEP001FCAE812A7: Set calledPartyNumber '5210' calledPartyName 'CODY CAMPBELL'
SCCP: do connected line for line 'SCCP/5210-0000000D', name: CODY CAMPBELL ,num: 5210
SEP0024C4BED45F: (pbx_indicate) start indicate 'AST_CONTROL_UNHOLD: Indicate call left hold' (17) condition on channel SCCP/5211-0000000C (readStat:2, receiveChannelState:2, rtp:yes, default res:inband signaling (-1))
SEP0024C4BED45F: (pbx_indicate) finish: send indication 'outofband signaling' (0)
SEP0024C4BED45F: (pbx_indicate) start indicate 'AST_CONTROL_CONNECTED_LINE: Indicate connected line has changed' (22) condition on channel SCCP/5211-0000000C (readStat:2, receiveChannelState:2, rtp:yes, default res:inband signaling (-1))
SCCP/5211-0000000C: Got connected line update, connected.id.number=5210, connected.id.name=CODY CAMPBELL, source=answer
  SCCP/5211-0000000C: 0x7fb9a002dea0: (getCallInfoStr):
 - calledParty: CODY CAMPBELL <5210>, valid
 - callingParty: KA'LONI HOLT <5211>, valid
 - originalCalledParty: Cody Campbell <5210>, valid, reason: 0
 - originalCallingParty: KA'LONI HOLT <5211>, valid
 - lastRedirectingParty: KA'LONI HOLT <5211>, valid, reason: 0
 - presentation: CallerId Presentation Allowed

SEP0024C4BED45F: send callInfo on SCCP/5211-0000000C with lineInstance: 2
SEP0024C4BED45F: (pbx_indicate) finish: send indication 'outofband signaling' (0)
 QualityStats: MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;ICR=0.0000;CCR=0.0000;ICRmx=0.0000;CS=0;SCS=0;MLQKvr=0.95
[2016-12-16 20:35:44] NOTICE[95688]: sccp_actions.c:3718 handle_ConnectionStatistics:     -- SEP001FCAE812A7: Call Statistics:
       [
         Last Call        : CallID: 14 Packets sent: 76 rcvd: 1 lost: 0 jitter: 0 latency: 0
         Last Quality     : MLQK=0.0000;MLQKav=0.0000;MLQKmn=0.0000;MLQKmx=0.0000;MLQKvr=0.95|ICR=0.0000;CCR=0.0000;ICRmx=0.0000|CS=0;SCS=0
         Mean Statistics  : #Calls: 7 Packets sent: 181 rcvd: 94 lost: 613565412 jitter: 0 latency: 0
         Mean Quality     : MLQK=0.5516;MLQKav=0.5798;MLQKmn=0.5516;MLQKmx=4.2556;MLQKvr=0.95|ICR=0.0076;CCR=0.0043;ICRmx=0.0434|CS=0;SCS=0
       ]
SEP001FCAE812A7: Getting the active channel on device.
SEP001FCAE812A7: Set the active channel 13 on device
SEP001FCAE812A7: send callInfo on SCCP/5210-0000000D with lineInstance: 2
SEP001FCAE812A7: Getting the active channel on device.
SEP001FCAE812A7: Set asterisk state Up (6) for call 13
```

dkgroot, 2016-12-18:
@Glasswire 
This look like the debug report from revision 47e11ef (correct ?). Reading the debug log, it looks like it hungup the temp channel, then calculated the number of channels on hold (=1) , and resumed that call, as far as i can tell. Which part of this is not working ? 

I think i might be missing a specific detail. When posting a debug log, please mention which revision (or combination of revisions) you where running at the time. Also tell me exactly what you tried and what you did / did not see at the time.

Glasswire, 2016-12-19:
I re-did this using revision 831b1fe by doing the follow (I apologize, I am not experienced with github)

```
git checkout 831b1fe
./configure
make -j2 && make install && make reload
```

and when I ran 'sccp show version', it displayed 

> Skinny Client Control Protocol (SCCP). Release: 4.3.0 (no - 831b1feM 

And the debug log is the same.

The call does resume, but it shows as below

[image]

You can't put the call back on hold or do anything but disconnect.

dkgroot, 2016-12-27:
@Glasswire 
I can't reproduce this using the current develop branch (without any other issue branches present). Can you confirm that.

commit 8b11ac08 (2016-12-14): Fix: when handling softkey stimulus with callid + lineInstance
     and we have a device->ActiveChannel in OFFHOOK state then hang it up and switch to the callid/linestate requested
     before handling the softkey request
Fix: when handling channel_resume and we have a device->ActiveChannel in OFFHOOK state, hang it up
Enh: cleanup line stimulus debug output by adding some returns at the end
Enh: in line-stimulus, if call on hold but multiple line channels then switch to the choosen line, and update display

resolves #356
PR #359 (2016-12-27): Issues/354
Fixes Issue: #354

functionality: Including Intercept softkey in your onhookstealable softkeyset, will make the remotely connected call visable in the callplane. If the intercept softkey is not available, then the call is collapsed.

Note: bit of cleanup of stale unused/testing code in sccp_indicate.c also happened.

Related issues: #356,  #350, #349

Completely fixed by this Fix: #349, #350 
Partially fixed by this Fix: #356
commit 47e11eff (2017-01-06): Fix: when handling softkey stimulus with callid + lineInstance
     and we have a device->ActiveChannel in OFFHOOK state then hang it up and switch to the callid/linestate requested
     before handling the softkey request
Fix: when handling channel_resume and we have a device->ActiveChannel in OFFHOOK state, hang it up
Enh: cleanup line stimulus debug output by adding some returns at the end
Enh: in line-stimulus, if call on hold but multiple line channels then switch to the choosen line, and update display

resolves #356
closed 2017-01-06 by dkgroot