chan-sccp/chan-sccp #357: Device Restarts upon receiving a faulty 'CallInfoDynamic' message

abradfordcms, 2016-12-14:
Device Restarts upon receiving a faulty 'CallInfoDynamic' message
Phone generates IllegalArgumentException and complains about Invalid Call Id=201

**Reproduction Steps:**

1. Asterisk needs to be up for a consistent number of weeks (sometimes a month) for this to occur - at least in the cases exhibited across two environments, so far
2. Place inbound call to phone
3. Upon answering (or sometimes upon call arriving at endpoint) the endpoint will un-register and then cycle into Registering 

**Expected behavior:**

Normal operation - no phone recycles

**Observed behavior:**

Regardless of if the devices lose registration (i.e intermittent network issues, power cycle, etc) prior to the condition being hit - the problem will still occur. End users, when faced with this issue, have tried to power cycle the phone only for it to occur again. Restarting the Asterisk service appears to be the only fix. 

Asterisk:
- version: 11.16.0 
chan-sccp: 
- branch: 4.3.0 dev
- revison: 1e00778M
Phone:
- type: 7975
- name: SEP2C3ECF76C81E
- firmware: 9.4(2SR1.1S)

- Log Files:
https://gist.github.com/abradfordcms/14f06ebf4840f528d432b69207eb4c7d

dkgroot, 2016-12-14:
Hi Andy,
Please update in a couple of steps through these revisions
11eb22f5, 049869, f53c1ad and edb0385

Your revision is very close to the first 3 revisions, so that should not be a problem. The last one is a little further ahead (and could contain regressions compared to your current enviroment).

I would like to make sure this issue is not already fixed, before investigating the 201 callreference / callid further.

abradfordcms, 2016-12-14:
Hi Diederik,

Sure thing - will be giving this a try.

Andy

dkgroot, 2016-12-14:
Already checked if callInstance should have been callId, but the current source code is correct in that regard. It might have been faulty in your revision, causing the issues. Marcello must have already fixed that. Once you confirm the newer revisions, we can close this issue in this regard.

dkgroot, 2016-12-22:
@abradfordcms 
Closing this issue. Did not get any further update about it not working. If the issue is still there using the latest 'develop' branch, then please reopen the issue.

closed 2016-12-22 by dkgroot