chan-sccp/chan-sccp #358: Asterisk restarts once a day

voipAsteriskGuy, 2016-12-15:
The asterisk server restarts after an error with the chan_sccp 
**Reproduction Steps:**
i dont have reproduction steps the only thing i start watching is that a phone starts to unregister and start to have a warning message

[2016-12-15 13:39:38] VERBOSE[15266] sccp_actions.c:  SEPXXXXXXXXXXXX: is registering, Instance: 0, UserId: 0, Type: Cisco 7911 (307), Version: 22 (loadinfo 'SCCP11.9-3-1SR4-1S')
[2016-12-15 13:39:38] NOTICE[15266] sccp_actions.c: SEPXXXXXXXXXXXX: Cleaning previous session, come back later, state:OK
[2016-12-15 13:39:38] VERBOSE[15266] sccp_device.c:  SCCP: Clean Device SEPXXXXXXXXXXXX
[2016-12-15 13:39:38] VERBOSE[15266] sccp_device.c:     -- SEPXXXXXXXXXXXX: (sccp_dev_set_registered) Setting Registered Status for Device from OK to Cleaning
[2016-12-15 13:39:38] VERBOSE[15266] sccp_device.c:     -- SEPXXXXXXXXXXXX: Registration State is 'Cleaning'
[2016-12-15 13:39:38] VERBOSE[15266] sccp_device.c:     -- Removed device 'SEPXXXXXXXXXXXX' from Glob(devices)
[2016-12-15 13:39:38] VERBOSE[15266] sccp_device.c:   == SCCP: Remove Line 54499 from device SEPXXXXXXXXXXXX
[2016-12-15 13:39:38] VERBOSE[15266] sccp_line.c:     -- Removed line '54499' from Glob(lines)
[2016-12-15 13:39:38] VERBOSE[15266] sccp_device.c:   == SCCP: Unregister Device SEPXXXXXXXXXXXX
[2016-12-15 13:39:38] VERBOSE[4955] sccp_devstate.c:     -- SEPXXXXXXXXXXXX: (sccp_devstate_deviceRegisterListener) device unregistered
[2016-12-15 13:39:38] VERBOSE[5532] sccp_device.c:     -- SEPXXXXXXXXXXXX: Registration State is 'Cleaning'
[2016-12-15 13:39:38] VERBOSE[5532] sccp_device.c:  SCCP: Clean Device SEPXXXXXXXXXXXX
[2016-12-15 13:39:38] VERBOSE[5532] sccp_device.c:     -- SEPXXXXXXXXXXXX: (sccp_dev_set_registered) Setting Registered Status for Device from Cleaning to Cleaning
[2016-12-15 13:39:38] VERBOSE[5532] sccp_device.c:     -- SEPXXXXXXXXXXXX: Registration State is 'Cleaning'
[2016-12-15 13:39:38] VERBOSE[5532] sccp_device.c:     -- Removed device 'SEPXXXXXXXXXXXX' from Glob(devices)
[2016-12-15 13:39:38] VERBOSE[5532] sccp_device.c:   == SCCP: Unregister Device SEPXXXXXXXXXXXX
[2016-12-15 13:39:38] VERBOSE[4955] sccp_devstate.c:     -- SEPXXXXXXXXXXXX: (sccp_devstate_deviceRegisterListener) device unregistered
[2016-12-15 13:39:38] VERBOSE[15266] sccp_device.c:     -- SEPXXXXXXXXXXXX: Registration State is 'None'
[2016-12-15 13:39:38] VERBOSE[5532] sccp_refcount.c: SCCP (sccp_line.c    :1121 (sccp_line_deleteLineButtonsArray   )) ALARM !! trying to release a (nil) with invalid memory reference! this should never happen !
[2016-12-15 13:39:38] ERROR[5532] sccp_refcount.c: SCCP: (release) Refcount Object 0xa94e8e0 could not be found (Major Logic Error). Please report to developers
[2016-12-15 13:39:38] WARNING[5532] sccp_utils.c: SCCP: (backtrace)
================================================================================
OPERATING SYSTEM: Linux, ARCHITECTURE: x86_64, KERNEL: 2.6.18-348.1.1.el5
ASTERISK: 11.17.1
CHAN-SCCP-b: Skinny Client Control Protocol (SCCP). Release: 4.3.0 TRUNK - unknown (built by 'root' on '2016-12-15 02:08:17 UTC')

To get a better backtrace you would need to install libbfd (package binutils devel package)
--------------------------------------------------------------------------(bt)--
 (bt) > /usr/lib64/asterisk/modules/chan_sccp.so [0x2b9128e4b2f6]
 (bt) > /usr/lib64/asterisk/modules/chan_sccp.so [0x2b9128dc605c]
 (bt) > /usr/lib64/asterisk/modules/chan_sccp.so [0x2b9128d9daa3]
 (bt) > /usr/lib64/asterisk/modules/chan_sccp.so [0x2b9128dafc42]
 (bt) > /usr/lib64/asterisk/modules/chan_sccp.so [0x2b9128db010f]
 (bt) > /usr/lib64/asterisk/modules/chan_sccp.so [0x2b9128db0d39]
 (bt) > /usr/sbin/asterisk [0x546e7c]
 (bt) > /lib64/libpthread.so.0 [0x3a0660683d]
 (bt) > /lib64/libc.so.6(clone+0x6d) [0x3a05ad503d]
==============
[2016-12-15 13:39:38] VERBOSE[5532] sccp_device.c:     -- SEPXXXXXXXXXXXX: (sccp_dev_set_registered) Setting Registered Status for Device from None to None
[2016-12-15 13:39:38] VERBOSE[5532] sccp_device.c:     -- SEPXXXXXXXXXXXX: Registration State is 'None'
[2016-12-15 13:39:38] VERBOSE[15266] sccp_line.c:  SEPXXXXXXXXXXXX: LineDevice FREE 0xa94e8e0
[2016-12-15 13:39:42] Asterisk 11.17.1 built by root on a x86_64 running Linux 

**Expected behavior:**
The service as to be running 

**Observed behavior:**
The asterisk service restarts after this error appens i had 3 o 4 of the previous message before the service restarts

**chan-sccp version:** (include revision and/or branch if relevant)
4.3.0 master and 4.3.0 develop

**asterisk version:** (include revision and/or branch if relevant)
11.21

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [No]
- Problem can be reliably reproduced, doesn't happen randomly: [No]
- Platform:
  - [CPU] Intel(R) Xeon(R) CPU E5-2640 @ 2.50GHz
  - [64 Bit]
  - [Big/Small Endian]
  - [Operating System]Linux Centos 
  - [Operating System Version] 5.9
- Direct Contact information: 
- Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]
- Comments: I have running chan_sccp using realtime and have almost 900 phones running first i tought it could be a problem with the keepalive time and i increased it but the problem still appens, it is not related to one extension, and it can append at any time of the day 
If you need more information or more logging verbose let me know

voipAsteriskGuy, 2016-12-15:
I have added the log i showed in the last message attached on this message
[sccp_error.txt](https://github.com/chan-sccp/chan-sccp/files/655698/sccp_error.txt)

dkgroot, 2016-12-16:
Related Issue Reports: #333 and #247 

We have had this report before and have not been able to make a lot of progress fixing this issue. In most other reports it was of a sporadic nature, meaning it only happened once every 15+ days. Increasing the number of phone and calls apparently makes this issue occur faster and more often. 

We already assumed the issue must be 'refcount' related, and your report confirms that. It would help a lot if you would install the 'binutils-devel' package and recompile chan-sccp. That way the backtrace will provide much better information. After installing the new version, we would have to wait until the issue happens again, for the better report.

We really need to figure out what is causing this refcount issue. If your server is not CPU/Disk bound, there are a couple of extra debug parameters that would help. Beware though, we would be gathering a whole lot of data, that needs to be processed and stored somewhere until the issue happens.

FYI: With this many phones connected to one server, the impact of such an issue must be great. If you have a chance of cloning your server instance and splitting your users over for example three asterisk instances, you would greatly reduce the impact to your users. chan-sccp does support rapid failover in a cluster environment.

voipAsteriskGuy, 2016-12-16:
I have already installed the binutils-devel package but I don't see any extra info on the backtrace, do i have to recompile chan-sccp with an extra option to get the extra debug? 
I don't believe the server is CPU/Disk bounded, could you please provide the extra debug parameters so we can get more info that could help us solve this problem?

Thanks.

dkgroot, 2016-12-16:
Normally when you have libbfd (aka binutils-devel) installed you only need to re-run ./configure. During configure it will detect bfd during 'Checking build environment'. It should show something like:
```
Checking build environment:
...
checking for execinfo.h... (cached) yes
checking for backtrace_symbols in -lexecinfo... (cached) no
checking for dlfcn.h... (cached) yes
checking for library containing bfd_openr... (cached) -lbfd
checking for bfd.h... (cached) yes
```
Meaning that it found the bfd library. Centos 5.9 installation is pretty ancient, so that might influence the outcome.

###Instructions on refcount debugging
You can add the '--eenable-refcount-debug' flag to you configure process. Which will create a new file called "/tmp/sccp_refs". Be aware that file will grow rapidly (about 40-50 lines per call) so you really need to make sure you have enough drive space. If your tmp drive is not a real disk but in memory, you need to make sure that you create an empty file somewhere where there is enough space, and create a link to it from /tmp/sccp_refs.

You need to activate the 'full' log in /etc/asterisk/logger.conf. I would also activate log rotation and compression in /etc/asterisk/logger.conf.

You should set the sccp debug level to 'core, device, line, channel, socket, refcount, filelinefunc'. And create a two cron entries:
```
0 60 * * * *  asterisk -rx 'logger rotate'
15 0 * * * * asterisk -rx 'sccp show refcount show'
```
Once the issue does occur you can run
```
asterisk -rx 'sccp debug all'
```
and wait long enough for the device which causes trouble do one more complete cycle of register attempts.

All of this will generate heaps and heaps of debug logging. So please make sure you have the space. If necessary you can move rotated / compressed log file to another location, to create some more space. Once we gather all this logging we will have to figure out a way to gather all the data together and process it (which will be quite a job as well).

BTW: I am currently working on a patch to make the debugging of this particular issue a little easier, so you might want to wait 1-2 more days, before doing this.

dkgroot, 2016-12-17:
@voipAsteriskGuy 

Created a new branch called 'issues/358' to help testing/checking this issue. It should generate a little more information during one of these events
```
git clone https://github.com/chan-sccp/chan-sccp.git chan-sccp_358
cd chan-sccp_358
git checkout issues/358
./configure --enable-refcount-debug --with-hash-size=5087 --enable-conference
make -j2 && make install && make reload
```
Note: i did add --enable-refcount-debug, so be prepared for a large /tmp/sccp_refs file.
Please run this on your test environment first to make sure everything still works.

### new refcount report
This new version will generate a 'refcount' report when ever a device fails to register, and whenever a device is being cleaned up. This should give us some insight into what is going wrong here. If everything is fine it would look like:
```
SEP0023043403F9 (handle_register) (realtime: yes)
refcount_report:

== refcount report =======================================================================
 device            SEP0023043403F9           ( 0x7fcafc032930), refcount:0001, alive:yes  
== related objects =======================================================================
==========================================================================================
```
it has no related/connected objects pointing at it and the refcount has gone all the way down to 1, so everything is nice a clean, ready for destruction (once we leave this function, the refcount goes to 0 and the device-object will be destroyed) . 

But when things are really wrong it would look like:
```
SEP0023043403F9 (handle_register) (realtime: yes)
refcount_report:

== refcount report =======================================================================
 device            SEP0023043403F9           (      0x31ef7f0), refcount:0011, alive:yes  
== related objects =======================================================================
 line              98041                     (      0x31f8250), refcount:0005, alive:yes  
 linedevice        SEP0023043403F9/98041     ( 0x7f1608001f80), refcount:0002, alive:yes  
 linedevice        SEP0023043403F9/98099     ( 0x7f160800cbc0), refcount:0002, alive:yes  
 line              98099                     (      0x31f8eb0), refcount:0006, alive:yes  
 linedevice        SEP0023043403F9/98041     ( 0x7f1618005070), refcount:0003, alive:yes  
==========================================================================================
```

### regarding '--with-hash-size':
Note: you said you have 900 phones connected to the server so i added the hash-size parameter to increase the size of the refcount hash table. It is advisable to keep the refcount fill factor below 0.70 (70%), you can check this with 'sccp show refcount'. If it is higher then 0.70 then please increase the number to a higher prime number. The default value is 536 which is way to low in your case.

dkgroot, 2016-12-18:
I added a very simple /tmp/sccp_refs rotation which will move the old file to /tmp/sccp_refs.? every 10 Mb. This should make it a little easier to deal with the file growth. You need to take care to move the /tmp/sccp_refs.? files to some other location. This will help prevent you running out of memory (because on most systems the /tmp drive is a memdisk).

Note: we need all of the /tmp/sccp_refs files, so we can later stitch them back together.

voipAsteriskGuy, 2016-12-19:
Diedrick, 
Thanks for following this up. I've upgraded the server to centos  7 so we can get the extra info on the stack trace related to binutils-devel now. I'm running the code from the new branch you created.
Also i have been using hash-size value 7993 for about a month now and the command sccp show refcount shows a value of 0.13. I've enabled the cronjobs  and refcount debug  as you suggested, also dedicated 300GB for logging on the sccp_refs  temp files.

dkgroot, 2016-12-19:
@voipAsteriskGuy 

As of the latest revision of this branch the /tmp/sccp_refs file does automaitcally rotate every 10Mb, So only the /tmp/sccp_refs file needs to remain in /tmp, the other files you can move to a safe location. No need for the 300Mb anymore. Just add a cronjob to move /tmp/sccp_refs.? somewhere else :-)

I made a couple of small changes to device destruction while working on this, which might have changed the frequency of the issue happening. Did everything go well during the last 11 hours ?

voipAsteriskGuy, 2016-12-19:
Yes it is working fine, but yesterday after the changes i did a sccp reload and at the middle of the reload the asterisk server crashed i noticed it in the logs but i cant find the trace like the previous time ¿could it be for the sccp loggin level?

dkgroot, 2016-12-19:
Let's not bother with a 'reload' issue for now. That is a different issue and we can look into that later.

Good to see that you found the hash-size flag on your own. 0.13 sound absolutely perfect.

BTW: The refcount values shown by 'sccp show refcount suppress' should remain between every run. This report only shows the refcounts for phones not doing anything at this moment. When a phone is on hook, it's refcount should return to the 'stable' state. What i have not prepared yet is a tool to compare these reports. The nice thing would be that using this report, we could find the /var/log/asterisk/full.xxxx debug logs, that are completely fine (so we do not have to look at them any more). Would you be able to figure out a way to process these automatically ?

dkgroot, 2016-12-19:
@voipAsteriskGuy 
Can you register on
https://gitter.im/chan-sccp/chan-sccp?utm_source=badge&utm_medium=badge&utm_campaign=pr-badge
So we can chat

dkgroot, 2016-12-20:
I finally found an easy way to reproduce one of these cross over session.
Wait until the device registers and when the keepalive time is between 0-10, and you pull the UTP cable from the phone and reconnect it 3 seconds later. The device will re-register within the time the server will wait before declaring the session timed out. In this case you get a cross over session causing all of the trouble.

Because i can now reproduce the issue, i was also able to find the reason for all of the re-registration trouble and refcount issues. We are cleaning up the device from the new thread, but if the server then starts to cleanup the device from the old thread as well, everything goes south.

Instead of cleaning up the device from the new thread, we should just pthread_kill the old thread and join it, that will cause both threads to run as one, and cleanup will be performed as a unified process, fixing all the refcount issues i could see.

Working on a clean fix for this, that we should run in your test environment for a while. Maybe you can sign over some of the devices during the next weekend, so we can run a full test with multiple devices over the weekend.

dkgroot, 2016-12-20:
Comment about ebe7837: after finding the cause and fixing the re-register issue using a pthread_cancel + pthread_join solution, there is some more work cleaning up all the old thread_stopping stuff in sccp_session.c. This can be done later at a slow pace.

dkgroot, 2016-12-26:
@voipAsteriskGuy Is reporting good initial results using this branch. So far no more segfaults or refcount errors, over the last couple of days. When it holds up for the next week, we should move forward and integrate these changes into the develop branch.

PR #364 (2016-12-28): Fix: device cross session cleanup (Issues/358)
Fix: device cross session cleanup issues (cause by two different session thread running sccp_dev_clean)
Fix: device re-registration
Fix: device timeout
Fix: module unload/load time
Enh: Extended refcount debugging when --enable-refcount-debug is enabled
Enh: sccp_session source cleanup

Fixes #358
References #309
closed 2016-12-28 by dkgroot