chan-sccp/chan-sccp #361: Pickup / GPickup phonebook records not set

dkgroot, 2016-12-26:
Issue: Pickup / GPickup phonebook are not updated

Cause: 
- connected_line update does not re-indicate last indication but only set's new callinfo
- this is good for normal calls (so we want to keep this)
- for pickup features this means some of the indications are missing, and callstate is not set and phone skips the phonebook update.

Fix:
We need to perform the required callstates while processing the pickup, so the entry is made

Note:
When using sendCallInfoV16 we need to set HUNT_NUMBER so that we get a phonebook entry:
```
To: picker
From: orig caller
For: orig called 
```

Current develop branch
Reported by Miguel

dkgroot, 2016-12-26:
@miguelsanzpardo
Can you please test/check the 'issues/361' branch, using
```
git fetch -a
git branch -a
git checkout issues/361
make -j2 && make install && make reload
```

Fixes:
 - The phonebook 'Received Calls' entry should now be added
 - When trying to pickup when there is nothing to pickup, you will here and error sound and a message will be displayed that there is nothing to be picked up.

miguelsanzpardo, 2017-01-09:
The problem still appears, at least when 2 or more phones try to do a gpickup when there is no calls to one of the phones that belongs to the call pickup group.

- If one phone calls to another (which belongs to the pickup group) and 2 or more phones try to pickup the call at the same time the problem doesnt appears. For example:
1008 --> Toni
1103 --> Ricard (Belongs to call pickup group)
1104 --> Albert (Belongs to call pickup group)
1102 --> Miguel (Belongs to call pickup group)
1º) 1103 calls to 1008
2º) 1104 and 1102 press GPickup button at the same time
3º) 1104 takes the call and see on callplane "From Ricard To 1008". When takes the phone see on callplane: "From Ricard(1103)". 1102 try to takes the call and cant do because 1104 has taken the call. 1102 see on the status bar "No call to pickup"
On the phonebook of 1104 it appeas this entry: From Ricard (1103) To 1104-Miguel For 1008, which is fine.

It appears another problem related with BLF update and gpickup:
1º) 1104 calls to 1103 and 1102 make a gpickup.
2º) 1102 see on callplane "From Albert (1104)" which is fine
     1104 see on callplane "To Miguel(1102)" which is fine
     1102 has a BLF which is mapping 1104. When 1102 make a pickup we can see on the screen that:
      Ricard (1103) <==> 1104-Albert instead of Miguel (1102) <==> 1104-Albert 

- If nobody is calling to one phone which belongs to the pickup group and 2 or more phones try to pickup the call at the same time the problem appears. For example:
1104 --> Albert (Belongs to call pickup group)
1102 --> Miguel (Belongs to call pickup group)
1º) 1102 and 1104 press at the same time Gpickup button
2º) On 1102 phone callplane appears: "From Unknown number To 1104". (I understand that it shouldnt appear anything or it should show "No call to pickup")
3º) When the user answer from 1102 phone it appears on the call plane "From Unknown number"
4º) On 1104 phone I have a BLF which maps 1102. When this problem appears, the blink light stay red until you reboot the phone
If you execute: CLI> sccp show channels

+--- Channels -----------------------------------------------------------------------------------------+
| ID    Name                      LineName   DeviceName       NumCalled  PBX State  SCCP State ReadCodec  WriteCodec RTPPeer                Direct DTMFmode |
===== ========================= ========== ================ ========== ========== ========== ========== ========== ====================== ====== ======== +
| 2     SCCP/1102-00000002        1102       SEP0026CBC03B37             Up         DOWN       alaw/64k   alaw/64k      192.168.1.127:23842 no     RFC2833  |
| 8     SCCP/1102-00000008        1102       SEP0026CBC03B37             Up         DOWN       alaw/64k   alaw/64k      192.168.1.127:26384 no     RFC2833  |
| 11    SCCP/1102-0000000B        1102       SEP0026CBC03B37             Up         DOWN       alaw/64k   alaw/64k      192.168.1.127:26702 no     RFC2833  |
| 18    SCCP/1102-00000012        1102       SEP0026CBC03B37             OffHook    DOWN       alaw/64k   alaw/64k                          no     RFC2833  |
+-------------------------------------------------------------------------------------------------------+

it looks like there is channels which doesnt be closed

miguelsanzpardo, 2017-01-09:
[full.txt](https://github.com/chan-sccp/chan-sccp/files/693936/full.txt)

dkgroot, 2017-01-10:
Asterisk-11
[Full Log](https://files.gitter.im/miguelsanzpardo/dYW9/full)
[Video](http://www.mediafire.com/file/e8nad37sp8iy1jj/VID_20170109_162352.mp4)

commit f758abfe (2016-12-27): Fix: phonebook entries during pickup and gpickup
Fix: when there is nothing to pickup
 - error sound(BEEPBONK instead of ZIPZIP)
 - displayprinotify instead of displayprompt

Fixes #361
commit d69508d5 (2016-12-27): Fix: phonebook entries during pickup and gpickup
Fix: when there is nothing to pickup
 - error sound(BEEPBONK instead of ZIPZIP)
 - displayprinotify instead of displayprompt

Fixes #361
commit 2759af96 (2017-01-11): Fix: pickup/gpickup race condition, where we actually try to pickup another sccp channel which was already trying to pickup

Cause: asterisk does not only allow RINGING/RINGIN channels to be picked up, but also channels with STATE_DOWN
Solution: Remove the callgroup / named callgroup from the newly created channel, which is doing a pickup. That way they don't
participate in the list of potential pickup candidates.

Fixes #361
closed 2017-01-11 by dkgroot