chan-sccp/chan-sccp #365: "Denied Receipt of Transferee" error on some transfers

abradfordcms, 2016-12-29:
Users processing transferring inbound calls will occasionally have called party experience no audio and call drop treatment on transfer

**Reproduction Steps:**

1. Asterisk uptime seems to need to be ~15 days
2. Place inbound call to a phone (7975 in experienced system) 
3. Transfer caller to another SCCP phone
4. Called party is notified of call and answers, but receives no audio 

**Expected behavior:**

Called party should receive successful transfer (either blind or attended) with full, two-way audio 

**Observed behavior:**

Called party experiences no-way audio and the transferred call leg drops. Logs show "(sccp_channel_transfer_cancel) Denied Receipt of Transferee XXXX [extension] by the Receiving Party. Cancelling Transfer and Putting transferee channel on Hold." 

**Screenshots**

N/A

SCCP version 4.3 dev commit f53c1adM 
Asterisk 11.10.0 

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes]
- Problem can be reliably reproduced, doesn't happen randomly: [Kind of - seems to occur after several days of uptime and call processing ]
- Platform:

Linode cloud PBX, CentOS 6.5

- Direct Contact information: andy.bradford.cms@gmail.com

dkgroot, 2016-12-29:
First Step in analysing this: do a hard reset on one of these devices, that seems to be stuck. If that clear up the issue for that device, we know that we are dealing with unclosed rtp ports for devices. If it does not clear up the issue, we need to look further.

dkgroot, 2016-12-29:
Based on the debug log you sent:
```
rtp_engine.c:     -- Locally bridging SCCP/XXX-0000XXX and SCCP/XXX-0000XXX
sccp_netsock.c:     -- SCCP: No externip/externhost set in sccp.conf.
sccp_actions.c:     -- SEPF09EXXXXXXXX: is Onhook (buttonIndex: 0, callid: 0)
sccp_channel.c:   == SEPF09EXXXXXXXX: Ending call SCCP/404-00001895 (state:CONNECTED)
sccp_channel.c:     -- SEPF09EXXXXXXXX: (sccp_channel_transfer_cancel) Denied Receipt of Transferee 6282 404 by the Receiving Party. Cancelling Transfer and Putting transferee channel on Hold.
sccp_channel.c:     -- SEPF09EXXXXXXXX: Sending hangupRequest to Call SCCP/404-00001895 (state: CONNECTED)
```
If the receiving party did not got Onhook, before the transfer was completed, based on your 'reproduction steps', then the 'is Onhook (buttonIndex: 0, callid: 0)' message should not have been received at this moment. That would indicate that the message before this got misinterpreted.

Marcello has recently committed a patch to faulty message handling. So please upgrade your chan-sccp version and test again.

I also noticed some devices timeing out, which also got fixed, so an update is a good idea.

If the issue persists after updating chan-sccp develop branch, please report back and do send a pcap / tcpdump for one of these calls, so that we can analyse the skinny messages received.

dkgroot, 2017-02-25:
Related Revision develop: e6fd157
Please pull the latest changes, an recheck

Also started a little project, to cleanly rewrite the whole transfer process in a seperate branch called feature/transfer. Would you mind helping out to test/check this (test environment only at this moment).
```
git checkout feature/transfer
./configure [... your configure flags...]
make -j2 && make reload
```
Mind you, i am still working on getting the callinfo correct, so please do not pay attention to that to much.

abradfordcms, 2017-03-03:
Hi Diederik,

Would be happy to. Will get it onto my test system this weekend.

Thanks,

Andy

dkgroot, 2017-03-03:
It still needs some more work. You can give it a go, but it's not completely ready yet 😂
Still having trouble with callinfo but transfer seem to be OK. Have concentrated on:
- simpler code
- error handling
- transfer softkey press via either channel (autoCallSelect in Sep file)
- transfer to dialplan functions
- transfer using speeddial
- real blind transfer (press transfer button before entering the digits of the destination).
- cancel real blind transfer (press transfer again (toggle).

These all seem to work, but have not been debugged completely. Call info and phonebook are the last hurdles.

Create a little project [project]https://github.com/chan-sccp/chan-sccp/projects/2#card-1860323) page to keep track.

Have fun checking it out. When it breaks just give me a shout
Don't put it in production yet !!