chan-sccp/chan-sccp #366: Incomplete information on phonebook directory (Placed and Missed calls)

miguelsanzpardo, 2017-01-09:
With plain dialplan, 2x7942 phones with firmware 9.4.2-2SR1 and Asterisk 11.21.2 I have these sccp extensions:
Ext.1102-Miguel --> Is monitoring on one line button the ext. 1104
Ext.1104-Albert --> Is monitoring on one line button the ext. 1102

If 1104-Albert calls to 1102-Miguel:

If 1102 take the call:
On the phonebook directory of 1104 on Placed Calls you see:  "To 1102" instead of seeing "To Miguel 1102" : Bad Behaviour
On the phonebook directory of 1102 on Received Calls you see:  "From Albert 1104"  : Good Behaviour

If 1102 doesnt tke the call:
On the phonebook directory of 1104 on Placed Calls you see:  "To 1102" instead of seeing "To Miguel 1102" : Bad Behaviour
On the phonebook directory of 1102 on Missed Calls you see:  "From Albert 1104"  : Good Behaviour

dkgroot, 2017-01-10:
When the call is not completed and no connection exists between both endpoints, then from the caller perspective only the dialed extension is known, not the dialed endpoint. So this is the correct behaviour. 

Example: 1104 calls extension 1111 which executes DIAL(SCCP/1102&SCCP/1103) and none of these endpoints answers the call, then the only information we have is:
- 1104 (cid:1104, cidname Albert) called extension 1111
- 1102 incoming call from 1104 (cid:1104, cidname Albert)
- 1103 incoming call from 1104 (cid:1104, cidname Albert)

If you would set callerid/name information before performing the DIAL, then we would have some caller id information to display, and we would set the information you wanted. But before the call is completed there is no information for us to set.

dkgroot, 2017-01-10:
Placed call update is incorrect and needs to be fixed :-((

dkgroot, 2017-01-12:
9.4 is more panicky about the ordering of events regarding callinfo sending.
Doing it correctly for 9.4 means it also works for 9.3.

The exact events need to be:
- Set DialedNumber
- callstate: Proceed
- sendCallInfo (callsecurity == 0) (with called, calling, orignally called, last redirecting)
- callstate: Ringout
- sendCallInfo (callsecurity == 0) (with called, calling, orignally called, last redirecting)
- callstate: Connected
- sendCallInfo (callsecurity == 1) (with called, calling, orignally called, last redirecting)

When sending calledNumber we also need to send calledName, and it needs to be correct on the first try.

This means we need to postpone sending the first CallInfo until we have gotten: AST_CONTROL_CONNECTED_LINE. This happens immediately after AST_CONTROL_RINGING. So code has been adapted to follow the ordering above:
- AST_CONTROL_RINGING 
   - set callstate: Proceed
- AST_CONTROL_CONNECTED_LINE 
   - sendCallInfo
   - callstate: RINGOUT
   - sendCallInfo
- sccp_pbx_answered
    - indicate(connected)
    - callstate: Connected
    - sendCallInfo (callsecurity == 1) (with called, calling, orignally called, last redirecting)

This seems to work correctly, further testing is need
We should try to reduce the impact/size of this patch if possible

Pushed branch 'issues/366' for initial testing

dkgroot, 2017-01-12:
Fixed ringing sound

miguelsanzpardo, 2017-01-12:
Tested and working fine now, thanks Diederik.

dkgroot, 2017-01-17:
Thanks for the update. This branch will need a little more work, to figure out al the corner cases.

dkgroot, 2017-03-12:
Revision: 5a1d9a7 fixes the "no callplane number shown during dialing stage" issue

Can be reproduced using
```
555 => {
    Wait(30);
};
```
reported by @drtmk

drtmk, 2017-03-12:
I've tested this on 9.3 and 9.4 firmware and looks good.

dkgroot, 2017-03-13:
@drtmk thanks for rechecking. Will merge this branch into develop and close this issue

closed 2017-03-13 by dkgroot