chan-sccp/chan-sccp #367: Cisco Phones Freezes

gitpropitia, 2017-01-09:
We are using SCCP 4.2 stable with asterisk 11.15. If we change anything on sccp.conf cisco phones freezes and stuck in re-registering. Issue resolved only after the amportal restart

chan-sccp version:Chan_SCCP-4.2.3_STABLE_r6728, Chan_SCCP-v4.2_STABLE_r6172

asterisk version:11.15.0

Addition / Optional Information:
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - [CPU]
  - [64 Bit]
  - [Medium]
  - [Operating System-CentOS]
  - [Operating System Version-6.X]
- Comments: We tried installing the same version in three different servers all are VMware ESXi, same problem for all the three servers.

I tried sccp 4.1 as well it crashes asterisk and same behavior if we change anything on sccp.conf file

I tried 4.3 development release as well same issue.

Cisco phone Model: 7970, 7975, 7942, 7962 and 7960/40

Thanks

miguelsanzpardo, 2017-01-10:
I have this problem with chan_sccp 4.2.0.6168 on several times when I add/delete/modify an sccp extension and execute a "sccp reload" or a "core reload", but with the last develop branch I should say that this problem doesnt exist,

dkgroot, 2017-01-10:
@gitpropitia 
It's  time to upgrade to the current 'develop' branch on git. V4.2.0 issue are not going to be fixed any more. 

The 'develop' branch should be running stable and not incur this reload issue. If you have a somewhat larger collection of phones, it is advised to use a realtime / database setup, so that reloads are not necessary, a device restart would suffice to pick up any changes to the config. See: [Using Realtime](https://github.com/chan-sccp/chan-sccp/wiki/using%20realtime%20database)

The annoying behaviour of freepbx to reload all modules for each and every little change is not something we can solve. This should never have been implemented this way, in my opinion. Most of the other asterisk modules are mostly stateless and don't really care about being reloaded, the sccp protocol however is statefull, which makes this more difficult. Switching to using realtime/database does eleviate most of the issues though.

closed 2017-01-10 by dkgroot

gitpropitia, 2017-01-10:
Thank you for your quick response. I already tried 4.3 master node let me install development node and update you the status.

Thank you miguelsanzpardo and dkgroot

dkgroot, 2017-01-10:
V4.3 develop is working and stable, with a couple of minor issues which we are working on. V4.3 master is still mostly the same as V4.2. As soon as the issues reported for 'develop' have been fixed, it will be merged into V4.3 master and we will issue a new release.